espressif/esp-hosted-mcu #17: ESP32 as HOST MCU

alberk8, 2024-11-05:
Is it possible to have an example implementation where ESP32 is use as the HOST and ESP32-C3 as the client for both Wifi and BLE with SPI and/or Uart.  Thank you.

michaesc, 2025-01-05:
It looks like Espressif tried to make a host driver for ST MCUs, failed at it and decided to give up and not try to make it work with any other MCUs. The state of compatability is so poor, that no ESP MCUs at all (even those lacking radio) can be found in the host subdirectory.

I guess this is why @mantriyogesh renamed the stm32 subdirectory to microcontrollers and make a limited attempt at correcting the failed Espressif engineering there.

If it's not true that Espressif failed to create an Espressif (for example ESP32-C3) host project to connect with a MCU running the Network Adapter firmware, then please show the location so that we can download and build the project.

mantriyogesh, 2025-01-05:
Hello @alberk8 ,

There are numerous people happily using the driver.
No worries. Please do let us know exact MCU chipset you use as host.

There is pure stm32 support being added to ESP-Hosted-MCU branch.

In fact we are focussed for MCU only solution, which is the reason , we created different separate repo for MCU case.
Let us know the places you faced issues with proper logs both host and slave to assist you further.

We are here to help technically. May be we can change your perception, may be not. In fact, because of your requirement, we started to support ESP as MCU host. We get it working fine with any ESP as host, which has supported communication bus.

Currently any ESP which has wifi and/or bluetooth can be used. Current chipsets as slave supported:

- QSPI transport supported slaves : ESP32-C2/C3/S2/S3/C6/C61

- SPI transport supported slaves: above all and ESP32

- SDIO transport supported Slaves: ESP32, ESP32-C6

- Uart transport supported: same set as SPI

closed 2025-01-23 by mantriyogesh