espressif/esp-hosted-mcu #18: Use common prefix for Kconfig option names

igrr, 2024-11-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Currently at least several different prefixes for Kconfig option names are used ( `CONFIG_ESP_HOSTED_`, `CONFIG_ESP_SDIO_`, `CONFIG_ESP_SPI_0`, `CONFIG_SLAVE_`, `CONFIG_IDF_SLAVE_`).

For most components, we have a rule that there should be a common prefix which is related to the component name. We could rename Kconfig options and use sdkconfig.rename file to achieve backwards compatibility.

mantriyogesh, 2025-01-20:
@SohKamYung-Espressif PTAL, ASAP

SohKamYung-Espressif, 2025-01-22:
@igrr Just to check: if I update the `Kconfig` options (to use a common prefix) and update ESP-Hosted code to use the new options, I do not need to provide a `sdkconfig.rename`?