espressif/esp-hosted-mcu #2: ESP32-P4 can not connect to ESP32-C6 on the Function_EV_Board v1.2 and v1.4

me-no-dev, 2024-11-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect to be able to connect to the C6 over SDIO

### Actual behavior (suspected bug)

Instead I get `E (226409) transport: Not able to connect with ESP-Hosted slave device`

### Error logs or terminal output

```shell
E (226409) transport: Not able to connect with ESP-Hosted slave device
```

### Steps to reproduce the behavior

Follow your steps in https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.0.1

### Shell

Bash

### Additional context

I made sure that the versions of both the P4 host and C6 slave are the same. Logs on both devices do not show any errors, other than the one above. C6 shows zero packets and is properly reset by the P4 host. Tested on my two EV boards v1.2 and v1.4. This is for adding support for P4 to the Arduino project

closed 2024-11-05 by espressif-bot

mantriyogesh, 2024-11-05:
@me-no-dev, please test and let us know.

mantriyogesh, 2025-03-06:
Added missing functionality in [daf33f5](https://github.com/espressif/esp-hosted-mcu/commit/daf33f54f16fab786ef49d2979d3ffa37ef58b9c)

Limited yet for SDIO. will need to propagate for other transports..

me-no-dev, 2025-03-06:
@mantriyogesh still missing the option to set the slave reset pin in config

mantriyogesh, 2025-03-11:
0a92d91053286a8523a78d39d29a50c9241908c9

me-no-dev, 2025-03-11:
Awesome! Will give it a shot when we get the new version