espressif/esp-hosted-mcu #3: WiFi connection example for ESP32-P4-Function-EV-Board

dmartauz, 2024-11-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Please provide minimal example for the P4 host using C6 connected over SDIO to make a connection to an access point, possibly without any unnecessary dependencies such as esp-bsp.

### Use cases

Basic testing of esp-hosted compatibility with selected version of esp-idf.

iothub01, 2025-06-19:
how to use esp AP mode in esp32p4 ? need to create a hotspot from esp32p4 dev board.

mantriyogesh, 2025-06-19:
You can even try using the above example, https://github.com/espressif/esp-hosted-mcu/issues/3#issuecomment-2457116901.

It already has integrated shell commands to start SoftAP. Try running help in the CLI and follow.
The APIs underlying you can use. https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html APIs we aim to extend to. current API set we support are in https://github.com/espressif/esp-hosted-mcu/blob/main/host/api/priv/esp_hosted_api_priv.h