espressif/esp-hosted-mcu #6: [esp32-p4] hosted bluetooth with nimble only

chegewara, 2024-11-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Hi,
is there any reason that fg hosted for esp32 mcu, esp32 P4 in this case, has enabled bluetooth stack for nimBLE only?
Why we cant use bluedroid?

Thanks

### Use cases

Classic BT, bluedroid BLE with ble mesh.

vap1977, 2025-01-03:
Are there any changes in hosted Bluedroid status?
Seems that it is the most straight way to get BT Classic support on ESP32P4.

Seems that esp-idf/components/bt/host/bluedroid/api/include/api/esp_bluedroid_hci.h interface is somewhat related, and it was added more than a year ago, but I don't see any usage of it.

SohKamYung-Espressif, 2025-01-06:
@vap1977 You can try this patch on top of current main.

[hosted-bluedroid-support.zip](https://github.com/user-attachments/files/18315352/hosted-bluedroid-support.zip)

It adds a BlueDroid based VHCI interface to ESP-Hosted, updates the documentation, and adds an example BlueDroid project based on the ESP-IDF [BlueDroid Host-only example using UART](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/bluedroid_host_only/bluedroid_host_only_uart), replacing the UART part with the ESP-Hosted VHCI interface.

To test it, you need to connect a ESP32 to the ESP32-P4 Dev Board via the GPIO header on the P4 (details in the updated Bluetooth documentation). Set the ESP-Hosted transport as SPI FD (full duplex) and configure the GPIOs used.

This patch (plus other changes) is targeted to be merged into main. Once it has been merged, this patch should not be needed.

vap1977, 2025-01-06:
Thank you very much!

SohKamYung-Espressif, 2025-01-21:
@chegewara @vap1977 The patch has been merged as https://github.com/espressif/esp-hosted-mcu/commit/3bbbf733787db6187f0609c46a76c9b3feaa2ee7 . You should be able to use either NimBLE or Bluedroid.

There are other BT related changes merged after that commit, so please pull all changes or get the latest version from the ESP Component Manager.

chegewara, 2025-01-21:
Thanks, i will test it.

mantriyogesh, 2025-01-21:
Its just one example slave and host chipset pins we tried while documenting.
Any other supported slave and any esp or non esp host would work @chegewara .

mantriyogesh, 2025-01-21:
All the slave chipset that you can use : ESP32, ESP32-C2/C3/S3/C5/C6.
Host can be practically anything.

Regarding your question, P4-C6 board, it should just work with existing pins (don't worry about hardware connections).
Just try as is, should just work.

chegewara, 2025-01-21:
Hi @mantriyogesh,
should i update firmware on C6 with new esp_hosted? (im guessing i should)

I can build esp mesh code on P4 now. I didnt test all options yet, but there is one that is not working (issue with building firmware) due to many
`undefined reference to bt_mesh_xxx` errors.

Thanks for hard work and updates

mantriyogesh, 2025-01-21:
We have not tried the mesh example ourselves. But in general, the bluedroid and bluetooth over vhci is available and it should be seamless, unless there are very specific apis that are dependent upon controller. In general all the APIs at host should just function fine.
Can you push your current code changes, that we can evaluate and possibly correct?

chegewara, 2025-01-21:
In general `esp_bt.h` is not available on P4 thats why API from it is also not available, including TX power.

Im guessing that esp-mesh v1.1 is not available because it is using prebuild binary, but that is only educated guessing.

All is just a brief observation, because most of the time i spend on figuring out why i cant build ble-mesh (i had some misconfig) on esp32 and then P4. It will take few weeks or maybe months before i test ble-mesh on P4, but i will try to report all i can find.

Thanks

mantriyogesh, 2025-01-21:
We will also give a thought, how easily this could be done.
Possibly through example. We as hosted team have not worked on mesh as such, so we also would need to add some efforts if something is missing at application.

In general, our approach we limit only till connectivity.
In any case, if you can push or send you part of code that you are trying and we able to reproduce the issue, we can get it ahead.

chegewara, 2025-01-21:
Hi,
i have some very small things to test by team:
- `esp_hosted_bt.h` is not C++ friendly,
- `BT_BLE_50_FEATURES_SUPPORTED` - im not sure if its general issue or just with ble mesh - when this option is enabled then after calling `esp_ble_mesh_node_prov_enable` i get this error
```
E (3878) BT_HCI: CC evt: op=0x200b, status=0xc
E (3888) BLE_MESH: start_le_scan, Invalid status 12
E (3888) BLE_MESH: starting scan failed (err -1)
```
- when `BT_BLE_50_FEATURES_SUPPORTED` mesh should works and i can detect node with nrfConnect, but cant find it with provisioners on android (neither nrf mesh or espBLEmesh)

I am using esp-idf master, updated today; esp_hosted 1.0.5 on P4 and C6 as slave
```
Name: 
Type: SDIO
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 0MB
CSD: ver=1, sector_size=0, capacity=0 read_bl_len=0
SCR: sd_spec=0, bus_width=0
TUPLE: DEVICE, size: 3: D9 01 FF 
TUPLE: MANFID, size: 4
  MANF: 0092, CARD: 6666
TUPLE: FUNCID, size: 2: 0C 00 
TUPLE: FUNCE, size: 4: 00 00 02 32 
TUPLE: CONFIG, size: 5: 01 01 00 02 07 
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
I (2982) sdio_wrapper: Function 0 Blocksize: 512
I (2992) sdio_wrapper: Function 1 Blocksize: 512
I (2992) H_SDIO_DRV: SDIO Host operating in STREAMING MODE
I (2992) H_SDIO_DRV: generate slave intr
I (3002) transport: Received INIT event from ESP32 peripheral
I (3002) transport: EVENT: 12
I (3012) transport: EVENT: 11
I (3012) transport: capabilities: 0xd
I (3012) transport: Features supported are:
I (3022) transport:      * WLAN
I (3022) transport:        - HCI over SDIO
I (3022) transport:        - BLE only
I (3032) transport: EVENT: 13
I (3032) transport: ESP board type is : 13 

I (3032) transport: Base transport is set-up

I (3042) transport: Slave chip Id[12]
I (3042) vhci_drv: Host BT Support: Enabled
I (3042) vhci_drv:      BT Transport Type: VHCI
I (3052) H_SDIO_DRV: Received INIT event
```

Thanks

PS it may be hard to post my code, because i was trying to make C++ ble-mesh library, which is on hold for few months now
since i can use ble bluedroid on P4 and my library should work, i can try to build mesh provisioner with lvgl GUI, but at first i have to make it work with basic features

in general i think there may be missing some API, just not sure where (here or esp-idf) to support ble mesh v1.1 and ble 5.x

chegewara, 2025-01-22:
For peace of my mind i tried ble mesh with espressif example (generic on/off server). It seems to work with couple exceptions:
- BLE 5.0 features are not supported, as described above
- cant build with `undefined reference` exceptions as described above, but this may be esp-idf issue, not the hosted library

Thanks for support and patience

chegewara, 2025-02-11:
I think i can close this issue.
Most of the ble mesh code is working and more will be implemented later.

I didnt test with normal ble code (non ble mesh), but i think it should also works.

Thanks

closed 2025-02-11 by chegewara

mantriyogesh, 2025-02-11:
Please let us know if you still face any issues or otherwise as well.

Thanks for letting us know..

chegewara, 2025-02-11:
As for now i am working with ble mesh v1.0, but i would like to see v1.1 on P4 working.
I like to use P4 for ble mesh for couple reasons:
- with 32MB PSRAM, big displays and powerful CPU it is very good choice to build provisioner/controller pretty easy
- with mesh v1.1 and blob model it should be easy to add OTA over mesh or more complex configurations
- P4 with mesh and ethernet is good combination, and sd card is a bonus

Thanks for support

mantriyogesh, 2025-02-12:
BTW, P4 has two sd buses. One sdio is default used (configurable to QSPI as well) for hosted. Other bus used on sd card.

Thread safe simultaneous access is available on sd card, from developer perspective.

OTA of?? P4 or slave?

chegewara, 2025-02-12:
OTA of P4 and/or slave is easy, there is API for that.
What i mean ble mesh nodes OTA, even non espressif one, like nordic chips.