espressif/esp-hosted-mcu #9: ESP-Hosted-FG:about Bluetooth stack

1653323560, 2024-11-05:
I saw in the ESP-Hosted FG document that the Bluetooth stack for MCU will be released in the future, when will it be released?
I wanted to adapt the ESP-Hosted FG to the STM32H750, but I couldn't find a suitable Bluetooth stack for MCU.
[image]

mantriyogesh, 2025-01-06:
@1653323560 ,

Just heads-up!

We recently tested bluedroid support on ESP32-P4 (+ ESP32). 
Reference: https://github.com/espressif/esp-hosted-mcu/issues/6#issuecomment-2572506990
We will shortly add this on master.

Both BLE and Classic-BT was tested. But the host chipset used is ESP.
I completely understand your query being support on STM32.

This could serve as reference implementation of esp-bluedroid over esp as host. Potentially, any host could add bluedroid component in build as component. We also have stm32 as host supported. we will check feasibility to add this port to work on stm32 as host in future.

mantriyogesh, 2025-01-20:
Any updates @gavin-hy ?

closed 2025-03-01 by mantriyogesh