espressif/esp-hosted-mcu #19: Support for ESP Now

203Null, 2024-11-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Is it possible to enable ESP Now via fg firmware?

I’m interested to build a P4 powered display hub to communicate with ESP Now devices but it seems to be impossible at the moment.

### Use cases

Allow ESP32 P4 to communicate with ESP Now devices 

### Alternatives

ESP-Extcomm shows it’s also not supported at the moment

NonaSuomy, 2025-11-28:
We use it in ESPHome for various things https://esphome.io/components/espnow/ I was shocked when I went to use it, that it wasn't supported with the esp32-p4 hosted platform. https://api-docs.esphome.io/classesphome_1_1espnow_1_1_e_s_p_now_component.html

mantriyogesh, 2025-12-31:
@203Null , @NonaSuomy , @HarryVienna @hyzgit,
Thanks for tagging the problem you faced here. It really helped us to prioritise the ESP-Now integration.
And Yes, adding ESP-NOW support is definitely on our roadmap.

At the moment, the focus is on understanding the minimal and correct set of ESP-NOW APIs and data structures that need to be supported over RPC, so that the implementation is both complete and maintainable. We are currently reviewing the ESP-NOW control and data paths to identify the appropriate integration points.

If anyone has prior experience or insights on which specific ESP-NOW APIs or use cases are most critical (for example, peer management, send/receive callbacks, or security-related APIs), feel free to share them. That input would help us prioritize and scope the initial implementation more effectively.

NonaSuomy, 2026-01-11:
That link above seems to be exactly what you are asking for https://api-docs.esphome.io/classesphome_1_1espnow_1_1_e_s_p_now_component.html

dpeart, 2026-02-12:
I came here for the same issue.  I have 5 esp32 S3 displays and one P4.  I can receive all my ESP_NOW data except for the P4.  I was surprised that the "best" esp32 didn't support this.  

I agree the post from @NonaSuomy is sufficient.