espressif/esp-hosted-mcu #20: Does one manually have to flash esp-hosted

ctag-fh-kiel, 2024-12-25:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Is it necessary to manually flash esp-hosted on an esp target when used a co-processor?
I am using the p4 (Espressif devboard) and a C6 as co-processor. The P4 connects thru SDIO with the C6 and esp-hosted as well as esp_wifi_remote allow the C6 to be used transparently. I am just not sure if it was initially flashed with any firmware, or if the C6 boots thru the SDIO from the p4? 

So if you does a board design, do the UART and Bootselect pins need to be accessible to flash the C6 with the esp-hosted firmware?

Thanks for clarification.

mantriyogesh, 2024-12-25:
Hello @ctag-fh-kiel,

The ESP32-C6 on the P4-Function-EV-Board comes/expected to come pre-flashed with an older version of the ESP-Hosted MCU "slave" binary.

You can easily flash the latest slave C6 binary using the [Quick Demo Guide.
](https://github.com/espressif/esp-hosted-mcu?tab=readme-ov-file#4-quick-demo-with-esp32-p4-function-ev-board)

For updates, OTA is supported. Host the network_adapter.bin (slave application binary) on your HTTP server, and the host can seamlessly perform OTA for the slave via SDIO just by using api call without the serial pins.

Note: OTA from the host works only if any hosted slave binary is already flashed on the C6

Till date, all new host side firmwares are downward compatible to the pre-flashed C6 binaries.

Is it really needed to flash C6, if it comes pre flashed on factory?

Upto you. The default C6 should still working for normal throughout.
We advice to flash slave and keep it in sync with host  firmware version for new features, bug fixes and very high throughput rates.

ctag-fh-kiel, 2024-12-25:
Hi @mantriyogesh thanks a lot for the clarifications.

ctag-fh-kiel, 2024-12-25:
For better understanding, is the C6 generally to be expected pre-flashed with esp-hosted or just on the P4 eval board?
Thanks :)

mantriyogesh, 2024-12-26:
C6 as slave on P4 eval board is expected to be pre flashed.
While, Individual retail C6 without P4 eval board, is **not** expected to be.

However, if your order is such then factory may accepts your expected bin to be as per your requirement, depending upon the order size.

mpolitowicz, 2025-02-19:
Hi @mantriyogesh , 

I have one additional question in context of that issue:

The esp_p4_function_ev_board description says that in order to do OTA one should

```
** On the ESP32-P4 host, add the following code to your application to initiate the OTA update:

#include "esp_hosted.h"

esp_err_t esp_hosted_ota_start(const char *url);

** Call the esp_hosted_ota_start function with the URL of the firmware binary:

esp_err_t err = esp_hosted_ota_start("http://example.com/path/to/network_adapter.bin");
if (err != ESP_OK) {
    ESP_LOGE(TAG, "Failed to start OTA update: %s", esp_err_to_name(err));
}
```
I could not find `esp_hosted_ota_start` but there is `esp_hosted_ota` which I assume should be used. Am I correct ? 

But when I include `esp_hosted.h` only and try to use `esp_hosted_ota`  I get following compilation error:
```
managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.h:107:33: error: unknown type name 'interface_buffer_handle_t'; did you mean 'MessageBufferHandle_t'?
  107 | void process_priv_communication(interface_buffer_handle_t *buf_handle);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 MessageBufferHandle_t

```
What am I missing ?

mantriyogesh, 2025-02-19:
> esp_hosted_ota_start

Yes, you are correct, the doc should be changed to esp_hosted_ota()

Let we try at our side and fix. Can you please share the snippet or small example, which would reproduce the issue?

mpolitowicz, 2025-02-19:
Thank you !

The example is quite simple as it is just wrapper for esp_hosted_ota:
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"

#include "esp_wifi.h"
#include "esp_hosted.h"

static const char *TAG = "http_hosted_ota";

static void http_hosted_ota_task(void *pvParameter)
{
    esp_err_t err = esp_hosted_ota(pvParameter);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to start OTA update: %s", esp_err_to_name(err));
    }
}

void start_http_hosted_ota(char *url_buf){    
    esp_wifi_set_ps(WIFI_PS_NONE);    
    xTaskCreate(&http_hosted_ota_task, "http_hosted_ota_task", 8192, url_buf, 5, NULL);
}
```

I tried esp_hosted version 1.2.0 and earlier 1.1.4 (the same issue).

mantriyogesh, 2025-02-20:
Hello @mpolitowicz ,

I added:

```diff
 ~/c/i/esp-idf >> examples/wifi/iperf > git diff 
diff --git a/examples/wifi/iperf/main/iperf_example_main.c b/examples/wifi/iperf/main/iperf_example_main.c
index 478fab79a4..cd8c865b44 100644
--- a/examples/wifi/iperf/main/iperf_example_main.c
+++ b/examples/wifi/iperf/main/iperf_example_main.c
@@ -21,6 +21,7 @@
 #include "wifi_cmd.h"
 #include "iperf_cmd.h"
 #include "ping_cmd.h"
+#include <esp_hosted.h>

 #if CONFIG_ESP_WIFI_ENABLE_WIFI_TX_STATS || CONFIG_ESP_WIFI_ENABLE_WIFI_RX_STATS
@@ -135,6 +136,8 @@ void app_main(void)
     printf(" |                                                |\n");
     printf(" =================================================\n\n");

+       esp_hosted_ota("http://1.2.3.4/network_adapter.bin");
+
     // start console REPL
     ESP_ERROR_CHECK(esp_console_start_repl(repl));
 }
```

I have used the idf as -
```git
 ~/c/i/esp-idf >> examples/wifi/iperf > git log --oneline |head -3                                                                                                    961ms  Thu Feb 20 13:04:56 2025
c71d74e2f8 Merge branch 'feat/upload_status_to_s3' into 'master'
bc1965a109 ci(all): upload support status when pushed go github
328665135c Merge branch 'fix/i2c_scl_freq_s2' into 'master'
```

Just tested the build on top of host as p4.

Procedure I followed:
```bash
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh
. ./export.sh
cd examples/wifi/iperf
idf.py fullclean
idf.py set-target esp32p4
idf.py build # default build is success first
<code changes above>
idf.py build
idf.py -p <host_port> flash monitor # I did not do this step, but should just work
```

mpolitowicz, 2025-02-20:
Ok, that is really odd. 

I used iperf example and removed everything except `esp_ota_hosted` (including sdkconfig.defaults and components except esp_hosted and esp_remote) and it compiles without issues.

On the other hand my project works without problems except the part when I try to compile and use `esp_hosted.h`. 

I thought that it is because of project configuration but I have not found differences that could impact it.  

So I made a guess and changed name of files `common.h` and `common.c` in component included to my project to 'common_aa.h' and `common_aa.c` (and subsequently in includes in module and CMakeLists.txt) and it worked.... 

So the workaround that worked was to add `#include <host/utils/common.h>` prior to `#include <esp_hosted.h>`

mantriyogesh, 2025-02-20:
I think common.h name itself is not good. But I was used internally so, we did not bother till now. But the files should use esp_hosted prefix.

I will add such fix.

mantriyogesh, 2025-02-25:
Hi @mpolitowicz ,  one large commit went in, to rectify your issue, as well as additionally do:
1. properly organise the files structure
2. correct the file naming for public exposed files
3. hide unneeded file to the public header inclusions
4. better way to share the headers in between slave and host
5. unclutter the large headers into smaller manageable and readable headers

commit: https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted_mcu/-/commit/c5fcc0e0cb0431a6eb33ed1d2435439f8005b340

closed 2025-02-25 by mantriyogesh