espressif/esp-hosted-mcu #21: Discussion of the feature/esp_as_mcu_host branch

Evlers, 2024-12-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

哈喽，时隔一年不到，我又回来了。
目前我已经基于feature/esp_as_mcu_host branch分支移植了wrapper层并通过了`esp_wifi_init`的无感调用。
但我对rpc_core.c的set_sync_resp_sem函数有点疑问！
在737行处创建了一个信号量：
https://github.com/espressif/esp-hosted/blob/83efce638beb0d59b641399df862b20e3aa12f36/host/drivers/rpc/core/rpc_core.c#L737
但是在738行的位置马上就使用阻塞方式获取信号量导致线程的阻塞：
https://github.com/espressif/esp-hosted/blob/83efce638beb0d59b641399df862b20e3aa12f36/host/drivers/rpc/core/rpc_core.c#L738
请问这里是有什么特殊的意义吗？

mantriyogesh, 2024-12-26:
Please check the semaphore create function:
https://github.com/espressif/esp-hosted/blob/83efce638beb0d59b641399df862b20e3aa12f36/host/port/os_wrapper.c#L486-L510

We are releasing semaphore immediately after creation.
This is the reason, it is acquired after creation.

Anyway, To make it easily portable for any host port, this call needs to be replaced by,
```c
g_h.funcs->_h_get_semaphore(app_req->rx_sem, 0);
```
this way, non-blocking way it would just cleanup the semaphore count, irrespective of port used

closed 2024-12-26 by espressif-bot

Evlers, 2024-12-26:
ok, thank you !
I also gave a signal when I created it.
used to take after rpc creation.

mantriyogesh, 2024-12-26:
Slave side api returned error. That same error was sent back to host as return value.

Worth to check and compare return value with
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv428esp_wifi_scan_get_ap_recordsP8uint16_tP16wifi_ap_record_t

Evlers, 2024-12-27:
Found the problem! Because I provided an invalid wifi_scan_config_t.
[image]
Now can connect to the Internet, hey hey

mantriyogesh, 2024-12-27:
That's indeed great news. May we know which platform you have ported onto?
If it is fine for you, Someone else in future may also refer to you for same platform.

Evlers, 2024-12-27:
I ported it to RT-Thread.
The spi device driver of rtthread is connected to the transport layer.
The network pair is connected to the wlan framework.
If you will allow me to PR, I will push a change on portability later.

mantriyogesh, 2024-12-27:
That would be a great help, @Evlers .

I am not sure exactly right now, how or if we can take the PR over main branch, or how can we support it.

But still having PR would make it usable for RT-Thread port for someone else.

Evlers, 2024-12-27:
My current porting mode is outside the esp-hosted directory,
that is, I do not modify the esp-hosted code(Modified some freertos apis and log print to the wrapper layer)
I think the main thing is to ensure the universal portability of esp-hosted, and then I will make a software package and add this esp-hosted as a submodule to perfectly support RT-Thread

mantriyogesh, 2024-12-27:
Actually, the port layer in the current repo (esp-hosted**-mcu**) is simplified.

We also have added recently STM32 support (not yet on master), which we tried to make simplified as non-ESP-IDF port.
But not yet on the master. check 

[esp-hosted-stm32_non_IDF-port.tar.bzip2.zip](https://github.com/user-attachments/files/18257958/esp-hosted-stm32_non_IDF-port.tar.bzip2.zip)

```sh
unzip ./esp-hosted-stm32_non_IDF-port.tar.bzip2.zip
tar xvf ./esp-hosted-stm32_non_IDF-port.tar.bzip2
```

Evlers, 2024-12-27:
I wrap the port at the level above the esp-hosted directory.
Instead of compiling freertos port directory in the esp-hosted.
[image]

Good, I see ota functionality has been ported to this port layer, I can use it as a reference.

mantriyogesh, 2024-12-27:
Great.

Can you share this in PR?
or some fork?

Evlers, 2024-12-27:
It will be open sourced after the port is completed,
Let me start modify some portability issues and PR

Evlers, 2024-12-27:
I just submitted a PR:https://github.com/espressif/esp-hosted-mcu/pull/22
Can you see if there is any problem

mantriyogesh, 2024-12-27:
Checked your PR. I didn't see any issues as such.

Also please have a look at https://github.com/espressif/esp-hosted-mcu/issues/21#issuecomment-2563365949
attached file.
I know its bit late for you now, but would restructuring the files like these under `extra_components`, would have helped earlier to you?

Any comments would be helpful from your angle.

Evlers, 2024-12-27:
Yes I will define the WiFi type using your way as a component.
In my opinion, the type definition of WiFi should be within the range supported by esp-hosted, otherwise you need to use the esp-idf to obtain it.

I'm currently working on a throughput issue.
Now iperf tx is unstable, while iperf rx is stable at 11mbps.
[image]
[image]

mantriyogesh, 2024-12-27:
For throughput, better to first check raw throughput test in rx and tx individually.

Evlers, 2024-12-27:
I think it makes sense to log to let the user implement the interface, Instead of using the esp-idf set of logs
Since each platform has its own log output format, is it not better to let users interface with the appropriate log format?

For example, I implemented it this way:
[image]
[image]

Evlers, 2024-12-28:
**memory leak problem was detected**
As long as the rpc call is executed, 336 bytes of memory are lost..
Have you found any?

mantriyogesh, 2024-12-28:
I have to check on this. 
Any specific api?

Evlers, 2024-12-28:
The actual test has the following two processes:
- Calling esp_wifi_get_mode and esp_wifi_scan_start(NULL, true) and esp_wifi_scan_get_ap_records lost 336 bytes
- Calling esp_wifi_get_mode and esp_wifi_sta_get_ap_info lost 336 bytes

mantriyogesh, 2024-12-28:
Got it. Will test this and let you know.

@SohKamYung-Espressif , ptal

Evlers, 2024-12-29:
OK, thank you.

Evlers, 2024-12-30:
The last PR(https://github.com/espressif/esp-hosted-mcu/pull/22) has been updated
Fixed a memory leak issue：https://github.com/espressif/esp-hosted-mcu/pull/22/commits/851acfa3e5f18c59d2b44225ce957e0c427d9866

SohKamYung-Espressif, 2024-12-30:
@Evlers Thank you for finding and fixing the memory leak issue. When I apply a patch using https://github.com/espressif/esp-hosted-mcu/commit/851acfa3e5f18c59d2b44225ce957e0c427d9866 , the leak is gone.