espressif/esp-hosted-mcu #25: esp32p4  spi 4 line mode  host function too slowly

skylin008, 2025-01-16:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

esp32p4 spi 4 line function, used 40M clock, max speed only 1 MiB, does it can be 5 MiB reachable.

### Use cases

esp32p4 connect esp32c5 wifi module via  SPI 4 Line mode, used FTP protocol download file from FTP server via esp32c5 wifi，it only 1 MiB download speed , how to solve this issue. Thanks!  

### Alternatives

None

### Additional context

None

skylin008, 2025-01-16:
@mantriyogesh thanks for your support,  I will  be try!

skylin008, 2025-01-17:
@mantriyogesh  I had try to QSPI communication between esp32c5 and esp32p4 , esp32p4  io port define as follow:

[image]

esp32c5 io port define as follow:

[image]

Run the examples/wifi/getting-started/station example, any issue shows, attach the logs:

[P4.LOG](https://github.com/user-attachments/files/18448013/P4.LOG)

[C5.LOG](https://github.com/user-attachments/files/18448015/C5.LOG)

the sdkconfig as bellow:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.5.0 Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_ANA_CMPR_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_AXI_GDMA_SUPPORTED=y
CONFIG_SOC_DW_GDMA_SUPPORTED=y
CONFIG_SOC_DMA2D_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_LCDCAM_CAM_SUPPORTED=y
CONFIG_SOC_LCDCAM_I80_LCD_SUPPORTED=y
CONFIG_SOC_LCDCAM_RGB_LCD_SUPPORTED=y
CONFIG_SOC_MIPI_CSI_SUPPORTED=y
CONFIG_SOC_MIPI_DSI_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_ETM_SUPPORTED=y
CONFIG_SOC_PARLIO_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_WIRELESS_HOST_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_LP_CORE_SUPPORTED=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_ISP_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_ECC_SUPPORTED=y
CONFIG_SOC_ECC_EXTENDED_MODES_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
[... 2231 lines cut ...]
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1 is not set
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y
CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE=y
CONFIG_ESP32_WIFI_ENABLE_WPA3_OWE_STA=y
CONFIG_WPA_MBEDTLS_CRYPTO=y
CONFIG_WPA_MBEDTLS_TLS_CLIENT=y
# CONFIG_WPA_WAPI_PSK is not set
# CONFIG_WPA_SUITE_B_192 is not set
# CONFIG_WPA_11KV_SUPPORT is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_WPA_DPP_SUPPORT is not set
# CONFIG_WPA_11R_SUPPORT is not set
# CONFIG_WPA_WPS_STRICT is not set
# CONFIG_WPA_DEBUG_PRINT is not set
# CONFIG_WPA_TESTING_OPTIONS is not set
# End of deprecated options

Thanks for your support!

mantriyogesh, 2025-01-17:
Can you please click a photo how your connections look like?

skylin008, 2025-01-17:
thanks @mantriyogesh

ESP32P4                     ESP32C5              
gpio1                              gpio7            
gpio2                             gpio0
gpio3                             gpio5
gpio4                            gpio4
gpio5                            gpio3
gpio6                           gpio2
gpio17                         reset pin

attach the shematic

[MA_1.0.pdf](https://github.com/user-attachments/files/18450674/MA_1.0.pdf)

mantriyogesh, 2025-01-17:
I wanted to see the photo, not schematic..

How are you connecting the QSPI? PCB or jumper cables?

skylin008, 2025-01-17:
From Ours had designed a PCB .

ESP32P4 GPIO PIN              <------->     ESP32C5 GPIO PIN
          GPIO1                           <------->       GPIO7
          GPIO2                          <------->       GPIO0
          GPIO3                          <------->       GPIO5
          GPIO4                          <------->        GPIO4
          GPIO5                          <------->         GPIO3
          GPIO6                          <------->        GPIO2
          GPIO17                        <------->      C5 reset pin

SohKamYung-Espressif, 2025-01-17:
@skylin008 I tested the QSPI between the ESP32-P4 and ESP32-C5. Here are my findings:

My HW setup:

[image]

This is the ESP32-P4 Function EV Board. Connected to the GPIO header on the P4 board is a board with a ESP32-C5-WROOM-1 module. The board on the left is a ESP-Prog used to flash firmware to the ESP32-C5.

My QSPI GPIO settings on the C5 slave:

[image]

My QSPI GPIO settings on the P4 host:

[image]

The SPI CLK is set to 40 MHz.

I ran the iperf example program and these are the results I got

P4 as iperf TCP server:
```
iperf> iperf -s -i 1 -t 10
I (185754) IPERF: mode=tcp-server sip=localhost:5001, dip=0.0.0.0:5001, interval=1, time=10
I (185755) iperf: Socket created
iperf> I (186648) iperf: accept: 10.10.2.162,53339

Interval       Bandwidth
 0.0- 1.0 sec  24.70 Mbits/sec
 1.0- 2.0 sec  24.60 Mbits/sec
 2.0- 3.0 sec  24.68 Mbits/sec
 3.0- 4.0 sec  24.63 Mbits/sec
 4.0- 5.0 sec  24.72 Mbits/sec
 5.0- 6.0 sec  24.68 Mbits/sec
 6.0- 7.0 sec  24.67 Mbits/sec
 7.0- 8.0 sec  24.69 Mbits/sec
 8.0- 9.0 sec  24.60 Mbits/sec
 9.0-10.0 sec  24.69 Mbits/sec
 0.0-10.0 sec  24.67 Mbits/sec
I (196653) iperf: TCP Socket server is closed.
I (196654) iperf: iperf exit
```

P4 as iperf UDP server:
```
iperf> iperf -s -i 1 -t 10 -u
I (235651) IPERF: mode=udp-server sip=localhost:5001, dip=0.0.0.0:5001, interval=1, time=10
I (235652) iperf: Socket created
I (235652) iperf: Socket bound, port 35091
iperf>
Interval       Bandwidth
 0.0- 1.0 sec  35.71 Mbits/sec
 1.0- 2.0 sec  35.72 Mbits/sec
 2.0- 3.0 sec  35.71 Mbits/sec
 3.0- 4.0 sec  35.64 Mbits/sec
 4.0- 5.0 sec  35.77 Mbits/sec
 5.0- 6.0 sec  35.71 Mbits/sec
 6.0- 7.0 sec  35.70 Mbits/sec
 7.0- 8.0 sec  35.75 Mbits/sec
 8.0- 9.0 sec  35.79 Mbits/sec
 9.0-10.0 sec  35.83 Mbits/sec
 0.0-10.0 sec  35.73 Mbits/sec
I (246480) iperf: Udp socket server is closed.
I (246480) iperf: iperf exit
```

P4 as iperf TCP client:
```
iperf> iperf -c 10.10.2.162 -i 1 -t 10
I (274300) IPERF: mode=tcp-client sip=localhost:5001, dip=10.10.2.162:5001, interval=1, time=10
I (274304) iperf: Successfully connected

Interval       Bandwidth
iperf>  0.0- 1.0 sec  29.38 Mbits/sec
 1.0- 2.0 sec  29.50 Mbits/sec
 2.0- 3.0 sec  29.75 Mbits/sec
 3.0- 4.0 sec  28.88 Mbits/sec
 4.0- 5.0 sec  29.62 Mbits/sec
 5.0- 6.0 sec  24.12 Mbits/sec
 6.0- 7.0 sec  7.50 Mbits/sec
 7.0- 8.0 sec  7.88 Mbits/sec
 8.0- 9.0 sec  13.88 Mbits/sec
 9.0-10.0 sec  29.50 Mbits/sec
 0.0-10.0 sec  23.00 Mbits/sec
I (284310) iperf: TCP Socket client is closed.
I (284311) iperf: iperf exit
```

P4 as iperf UDP client:
```
iperf> iperf -c 10.10.2.162 -i 1 -t 10 -u -b 50
I (291956) IPERF: mode=udp-client sip=localhost:5001, dip=10.10.2.162:5001, interval=1, time=10
I (291957) iperf: Socket created, sending to -1576924662:5001

Interval       Bandwidth
iperf>  0.0- 1.0 sec  42.47 Mbits/sec
 1.0- 2.0 sec  38.18 Mbits/sec
 2.0- 3.0 sec  31.55 Mbits/sec
 3.0- 4.0 sec  30.60 Mbits/sec
 4.0- 5.0 sec  42.21 Mbits/sec
 5.0- 6.0 sec  42.25 Mbits/sec
 6.0- 7.0 sec  42.30 Mbits/sec
 7.0- 8.0 sec  42.20 Mbits/sec
 8.0- 9.0 sec  42.20 Mbits/sec
 9.0-10.0 sec  42.34 Mbits/sec
 0.0-10.0 sec  39.63 Mbits/sec
I (301971) iperf: UDP Socket client is closed
I (301971) iperf: iperf exit
```

For your reference, here are my sdkconfig files:

[skdconfig-for-slave.txt](https://github.com/user-attachments/files/18452449/skdconfig-for-slave.txt)

[sdkconfig-for-host.txt](https://github.com/user-attachments/files/18452455/sdkconfig-for-host.txt)

mantriyogesh, 2025-01-17:
We can't exactly test your used GPIOs, but we verified with the board with us, the QSPI is working without issues.

To add up, we also faced the issue of the message not receiving in our QSPI, when we used the jumper wires setup. although, the same issue was not observed when we used wires and 'dualSPI'.

With QSPI over PCB, however such issue was not discovered.

Suggestions:
1. Try dual SPI first
2. with QSPI, lower the freq and try
3. gradually increase the freq and try in iterations, where or when the drivers are getting saturated.

If you need, we can share out schematic over mail. Please find my mail id on the profile.

skylin008, 2025-01-17:
@mantriyogesh SPI test is ok, the clock is 40Mhz, none error information shows.

mantriyogesh, 2025-01-20:
@skylin008 Any updates?

skylin008, 2025-01-22:
@mantriyogesh  Thanks for your kindly support. The same issue shows, it can't work correctly, it crash and reboot!

skylin008, 2025-01-22:
Attach the ESP32P4 logs

[esp32p4 debug info.txt](https://github.com/user-attachments/files/18498688/esp32p4.debug.info.txt)

[image]

mantriyogesh, 2025-01-22:
Check if you have connected the pins correctly. Also ensure the slave 'EN' is connected to your reset pin in the P4 and the C5 gets resets every time the P4 is reset.

There are some suggestions mentioned in here: https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2597855512

As you say, the SPI test is okay, you can use same pins to test dual SPI first.
I also have shared the schematic with you, with which our board performs just fine. although, there is small issue in the schematic we sent and hence we need the jumper cables flying as you can see the diagram.

But anyway, these jumpers are only for getting C5 into download mode, nothing to do with QSPI. the QSPI pins are through PCB.

At the lease, did you try dual SPI or QSPI with lower frequency like 5MHz?

Apart from this, In prior sdkconfig of the **Full-Duplex SPI**, I could see 

**ESP32-P4**
```
CONFIG_ESP_SPI_GPIO_MOSI=4
CONFIG_ESP_SPI_GPIO_MISO=3
CONFIG_ESP_SPI_GPIO_CLK=5
CONFIG_ESP_SPI_GPIO_CS=6
CONFIG_ESP_SPI_GPIO_HANDSHAKE=1
CONFIG_ESP_SPI_GPIO_DATA_READY=2
CONFIG_ESP_SPI_GPIO_RESET_SLAVE=17
..
..
CONFIG_ESP_SPI_CLK_FREQ=40
```

mapped to:

**ESP32-C5**
```
CONFIG_ESP_SPI_GPIO_MOSI=4
CONFIG_ESP_SPI_GPIO_MISO=5
CONFIG_ESP_SPI_GPIO_CLK=3
CONFIG_ESP_SPI_GPIO_CS=2
CONFIG_ESP_SPI_GPIO_HANDSHAKE=1
CONFIG_ESP_SPI_GPIO_DATA_READY=0
CONFIG_ESP_SPI_GPIO_RESET=-1
```

These pins map to all your pins mentioned in https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2597712251

except, `CONFIG_ESP_SPI_GPIO_HANDSHAKE=1` in P4 was mapped to `CONFIG_ESP_SPI_GPIO_HANDSHAKE=1` in C5.

In your recent pins, you mentioned GPIO1 in P4 connected to GPIO7 in C5.

I am confused, which one is correct and real mapping in your schematic.
Can you please review as per your schematic for this pin?

skylin008, 2025-01-22:
Thanks @mantriyogesh ,  esp32c5 GPIO1 pin  to connected to esp32p4  GPIO1 pin in my shematic.  I had test the same pin in Full-Duplex SPI, it work ok. For Half-Duplex SPI mode, esp32p4  only add  GPIO15 pin to connect esp32c5 GPIO7 pin.

mantriyogesh, 2025-01-22:
1. Try QSPI with correct gpios with low freq
2. If 1 fails, try using dual spi and confirm
3. If 2 fails, Can you please send the c5 and P4 sdkconfig and their logs from the start?

skylin008, 2025-01-23:
Try QSPI with correct gpios with low freq： **test fail**
If 1 fails, try using dual spi and confirm: **test fail**
If 2 fails, Can you please send the c5 and P4 sdkconfig and their logs from the start?

[esp32c5 debug info_dual spi.txt](https://github.com/user-attachments/files/18519870/esp32c5.debug.info_dual.spi.txt)
[esp32c5 debug info_qspi.txt](https://github.com/user-attachments/files/18519871/esp32c5.debug.info_qspi.txt)
[esp32p4 debug info_dual spi.txt](https://github.com/user-attachments/files/18519869/esp32p4.debug.info_dual.spi.txt)
[esp32p4 debug info_qspi.txt](https://github.com/user-attachments/files/18519868/esp32p4.debug.info_qspi.txt)

sdkconfig file follow below:

[sdkconfig_for_host_dualspi.txt](https://github.com/user-attachments/files/18520032/sdkconfig_for_host_dualspi.txt)
[sdkconfig_for_host_qspi.txt](https://github.com/user-attachments/files/18520031/sdkconfig_for_host_qspi.txt)
[sdkconfig_for_slave_dualspi.txt](https://github.com/user-attachments/files/18520034/sdkconfig_for_slave_dualspi.txt)
[sdkconfig_for_slave_qspi.txt](https://github.com/user-attachments/files/18520033/sdkconfig_for_slave_qspi.txt)

mantriyogesh, 2025-01-23:
did the standard SPI with these pins worked for you any time?

skylin008, 2025-01-23:
did the standard SPI with these pins worked for you any time?  **Yes**

"""The extconn and esp-hosted cannot work together.

Please disable extconn as described in the documentation:

https://github.com/espressif/esp-hosted-mcu/blob/main/docs/spi_full_duplex.md#3-remove-conflicting-configuration
"""
**I had disable the extconn, but error information shows, attach the debug file**

[esp32p4 debug info_qspi.txt](https://github.com/user-attachments/files/18521583/esp32p4.debug.info_qspi.txt)

[esp32c5 debug info_qspi.txt](https://github.com/user-attachments/files/18521585/esp32c5.debug.info_qspi.txt)

mantriyogesh, 2025-01-24:
Currently in the log, It look likes the message exchange for some commands was already successful 
wifi init                      # fine
wifi set country        # fine
wifi set ps                 # not fine at slave

So the QSPI at hardware level seems to be okay. The crash in slave times out the host for the command.

Kindly wait from us to check on this and get back. We will test again and confirm.

By the time, Can you please confirm:

1. idf exact git commit
2. git status outout under idf directory
3. If any changes that you made in iperf example or anywhere apart from GPIO pins?

SohKamYung-Espressif, 2025-01-24:
I just tested with the ESP32C5-board on the ESP32-P4 Dev Kit, and the logs look ok. On the ESP32C5 side, I do not see a crash after the `wifi set ps` line.

See attached debug log output

[esp32c5-20250124-debug-qspi.txt](https://github.com/user-attachments/files/18530667/esp32c5-20250124-debug-qspi.txt)

[esp32p4-20250124-debug-qspi.txt](https://github.com/user-attachments/files/18530668/esp32p4-20250124-debug-qspi.txt)

mantriyogesh, 2025-01-24:
Still something is mismatched between us and @skylin008 's either hardware, or software.

As you state full duplex spi works fine, but QSPI doesn't. 

Let me compare the schematic in https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2597629541 with ours.

mantriyogesh, 2025-01-24:
@skylin008 , As you have core dump shared already, can you please send us the c5 binaries and related files to debug the core dump at c5?

Alternative 1) use `idf.py monitor` for esp logging, instead of minicom or other terminal emulator programs.

Are you using `idf.py monitor` to see logs at esp32-c5? If so it would **auto** decode the core dump, where the crash happened.

Alternative 2) collect files manually and send to use for finding the line where it crashed

cd <slave_c5_proj>/build
tar cvzf c5_bins.tgz 

# c5 bin files
1. build/network_adapter.bin
2. build/ota_data_initial.bin
3. build/network_adapter.bin
4. build/partition_table/partition-table.bin

# elf files
5. build/bootloader/bootloader.elf
6. build/network_adapter.elf

# map files
7. build/bootloader/bootloader.map
8. build/network_adapter.map

# latest logs
9. c5 log with core dump

You can also run below script and get us c5_core_bundle.tgz

```
#c5 core bundle
tar cvzf c5_core_bundle.tgz build/flash_args ./partitions.esp32*.csv build/network_adapter.bin build/ota_data_initial.bin build/network_adapter.bin build/partition_table/partition-table.bin build/bootloader/bootloader.elf build/network_adapter.elf build/bootloader/bootloader.map build/network_adapter.map sdkconfig*
```

Please note:
1. We would see the crash happened at slave through this.
2. If you have built the c5 again, please take fresh logs and above set of files.
3. We would not be able to test unless the pins are adjusted to our pcb GPIOs. Let's go step by step. First we will analyse above logs.

mantriyogesh, 2025-01-25:
@skylin008 , can you please attach the necessary files for crash dump analysis as mentioned above or use IDF.py monitor to get crash log?

skylin008, 2025-01-27:
Thanks @mantriyogesh , attach the p4 logs and c5 logs from idf.py monitor

[logs.zip](https://github.com/user-attachments/files/18558792/logs.zip)

mantriyogesh, 2025-01-27:
I am quite surprised to see that C5 despite being slave, shows the log similar to 'host'.

For reference successful logs, please refer: https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2611613450

Are you flashing host on C5 and P4 both?
It would not work this way.

I also compared your old logs. In the old logs, I can see that you were flashing C5 from slave project.
reference (your c5 log from https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2609900054) : 

[image]
is correct.

The recent log you attached for c5 is not correct.

Possibly, you are either flashing host on both C5 & P4 (likely), or you are accessing wrong uart port for logs (unlikely).

Can you please create a new project using:
https://components.espressif.com/components/espressif/esp_hosted/versions/1.1.1/examples/slave?language=en

```
idf.py create-project-from-example "espressif/esp_hosted=<x.y.z>:slave"
```
where, x.y.z is same to one displayed in your ESP32-P4 log while flashing.

Ensure that you create slave project **outside** of the esp-idf directory tree?

skylin008, 2025-02-09:
Thanks @mantriyogesh, I'm back form my vacation. Attatch the latest c5 logs and p4 logs from idf.py monitor.

mantriyogesh, 2025-02-09:
Great. Thanks for the logs with backtrace in slave side.

```
(2155) slave_ctrl: Received Req [0x118]
I (2159) phy_init: phy_version 102,3b01d78a,Nov 27 2024,17:24:11
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4000152c  RA      : 0x42083402  SP      : 0x40832800  GP      : 0x40814584
--- Stack dump detected
--- 0x4000152c: __call_phy_bt_tx_gain_init in ROM
0x42083402: register_chipv7_phy at ??:?

TP      : 0x408329b0  T0      : 0x40035ce2  T1      : 0x0000000f  T2      : 0xffffffff
--- 0x40035ce2: memset in ROM

S0/FP   : 0x40815000  S1      : 0x4083be3c  A0      : 0x0000006d  A1      : 0x00000000
A2      : 0x00000006  A3      : 0x00000004  A4      : 0x00000000  A5      : 0x0000001c
A6      : 0x00000000  A7      : 0x408161e8  S2      : 0x40815778  S3      : 0x00000000
S4      : 0x420a3be8  S5      : 0x0038e7ff  S6      : 0x4085ff28  S7      : 0x4004b000
S8      : 0x40860000  S9      : 0x40860000  S10     : 0x4085ff40  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x42027280  T5      : 0x420273d6  T6      : 0x42027172
--- 0x42027280: aes_128_cbc_encrypt at D:/Espressif/frameworks/esp-idf-master/components/wpa_supplicant/esp_supplicant/src/crypto/crypto_mbedtls.c:445
0x420273d6: pbkdf2_sha1 at D:/Espressif/frameworks/esp-idf-master/components/wpa_supplicant/esp_supplicant/src/crypto/crypto_mbedtls.c:752
0x42027172: hmac_sha256_vector at D:/Espressif/frameworks/esp-idf-master/components/wpa_supplicant/esp_supplicant/src/crypto/crypto_mbedtls.c:341
```

In the slave log confirms that the slave has crashed clearly and host awaited for the request, which it got failure and was marked as Fatal, resulting abort at host.

Now, as I see the crash is null pointer access and in some bluetooth function and embedtls.

My first and very clear guess is that you are using c5 with 0.0 version.

The c5 with eco1(1.0+) are only supported on latest IDF and older c5 get crash with latest IDF. Considering the c5 is not MP yet, these chip revisions are expected. Please note these are hardware revisions, so in worst case you might go back to older IDF commit (unsure which right now), but basically without IDF software support (till some commits it would work, no later), as this chip hardware is deprecated (anyway C5 is not in MP yet).

I also checked/confirmed from your log of eFuse that the chip is 0.0. you can alternatively run esptool.py with chip_id argument.

```
esptool.py --port <slave-port> chip_id
```

Please let us know if this c5 is eco1 or later.

mantriyogesh, 2025-02-09:
idf commit, as I found out, https://github.com/espressif/esp-idf/commit/d092c1ba55d3ff4932572ba4e69c21d963e422e2

Any C5 eco0 (0.0) c5 would not be able to work ahead this.
At least eco1 is needed (>0.0)

alternatively, you can also see the idf flashing log, which would contain string `eco1` or later
```
[11/26/24 3:18 PM] Lv Haiyu
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB1 115200

--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

der: 0xf ESP-ROM:esp32c5-eco1-20240726

Build:Jul 26 2024

rst:0x1 (POWERON),boot:0x10 (SPI_FAST_FLASH_BOOT) 
```

The above trace only holds true for c5 eco1 devices.

mantriyogesh, 2025-02-10:
@skylin008 please let us know further, once you have some observation..

skylin008, 2025-02-10:
Thanks@mantriyogesh, I will be try! If eco0 firmware can be used eco1 devices?

mantriyogesh, 2025-02-10:
yes.

If you can wait a little (unsure how much), you can also try to use eco2 directly.
No guarantees, but eco2 should also have sdio on c5.

Other way around, you can actually downgrade to specific commit in idf and try on existing device, if you are going to wait for eco2. Also, it would rule out the issue is indeed only because of the hardware incompatibility.

Anyway, eco1 is supported on IDF master.

skylin008, 2025-02-12:
@mantriyogesh Thanks for your support! when esp-hosted update to 1.1.2：slave, used eco0 chip, the esp32C5 module keeps restarting approximately every 1 second when used SPI_FULL_DUPLEX. But when used  esp-hosted 0.0.22:slave, it can be worked.

mantriyogesh, 2025-02-12:
It would be helpful if you attach both side for quicker resolution on lastest slave. Please also let us know the IDF version in use and all submodules are updated in the esp-idf directory.

Please be informed, if the issue happens in IDF driver(s), there is no way we can support the c5 eco. However if the issue is in hosted, we can certainly fix.

skylin008, 2025-02-12:
The esp-hosted version is: 1.1.2 ，esp-idf version is: ESP-IDF v5.5-dev-847-gcb3ac7429c-dirty, two versions used to eco0 esp32c5 chip can be work well, but them used to the eco1 esp32c5 chip can not be work, esp32c5 can not  communication with esp32p4. The above work with SPI_FULL_DUPLEX.

mantriyogesh, 2025-02-12:
https://github.com/espressif/esp-idf/commit/cb3ac7429c > https://github.com/espressif/esp-idf/commit/d092c1ba55d3ff4932572ba4e69c21d963e422e2 . should not be used on eco0. if it works, mere co-incidence.

**Irrespective of transport**, please stick with commits. The commits are software, but they are tied to specific hardware.

commit <  https://github.com/espressif/esp-idf/commit/d092c1ba55d3ff4932572ba4e69c21d963e422e2 => eco0
commit >=  https://github.com/espressif/esp-idf/commit/d092c1ba55d3ff4932572ba4e69c21d963e422e2 => eco1

skylin008, 2025-02-12:
what about  issue about eco1 can not be communication with each other?

mantriyogesh, 2025-02-12:
Sorry did not understand what exactly you meant  can you please detail a bit?

mantriyogesh, 2025-02-12:
1. There are too many variables. Let us rule out some lower priority scenarios.

2. If you already have eco1, I would strongly suggest **not** use eco0, as it is deprecated hardware.

3. How is eco1 now connected to P4? Is it similar PCB hardware or jumper cables? Can **click a camera ** photo and share?

4. Please set up **Latest master** IDF. Ensure all submodules are updated.

5. Clean build  (rm -rf sdkconfig, build/) **host** with latest IDF. (idf.py fullclean)
6. Flash **host** P4 firmware to  latest hosted registry component version

7. Clean build (rm -rf sdkconfig, build/) slave **eco1** with latest IDF (idf.py fullclean)
8. Flash **C5 eco1**  with new project using 'slave' 'example' inside registry component.

9. Ensure the spi connections are expected as per your GPIO mapping

10. Ensure the slave EN is connected to P4's 'Reset' GPIO

11. RESET P4. This should also reset slave.
12. Share your observations with the **textual logs attached**. Without logs we cannot understand.

mantriyogesh, 2025-02-15:
Sorry, but without the necessary logs and specific information, we cannot proceed to even understand the issue.

https://github.com/espressif/esp-hosted-mcu/issues/25#issuecomment-2652961156

skylin008, 2025-02-18:
[c5.txt](https://github.com/user-attachments/files/18846436/c5.txt)
[p4.txt](https://github.com/user-attachments/files/18846435/p4.txt)
Attatch esp32-c5 eco1 chip version and esp32p4 running logs.

SohKamYung-Espressif, 2025-02-19:
@skylin008 From the `p4.txt`:

```
I (11562) transport: Not able to connect with ESP-Hosted slave device
I (11562) transport: Reset slave using GPIO[17]
I (11562) os_wrapper_esp: GPIO [17] configured
```

The P4 is unable to reset the C5. I do not see the C5 rebooting in the `c5.txt`. This is required to start SPI transactions between the P4 and C5. 

Please check that the P4 GPIO used (GPIO 17) is properly connected to the pin used to reset the C5.

Also, the logs indicate that you are running with SPI FD (full duplex) not SPI 4-line mode (SPI half duplex). This is the correct SPI mode? From `p4.txt`:

```
I (542) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
spi_mempool_create free:33702128 min-free:33702128 lfb-def:33030144 lfb-8bit:33030144

I (546) spi_wrapper: Transport: SPI, Mode:3 Freq:40MHz TxQ:20 RxQ:20
 GPIOs: MOSI:4 MISO:3 CLK:5 CS:6 HS:1 DR:2 SlaveReset:17
```

From `c5.txt`:

```
I (462) fg_mcu_slave: *********************************************************************
I (471) fg_mcu_slave:                 ESP-Hosted-MCU Slave FW version :: 0.0.6                        
I (481) fg_mcu_slave:                 Transport used :: SPI only                      
I (490) fg_mcu_slave: *********************************************************************
```

skylin008, 2025-02-19:
@SohKamYung-Espressif  Thanks! Yes, I used stand SPI full duplex to run with esp32C5 ECO1 chip.

SohKamYung-Espressif, 2025-02-20:
@skylin008 Do you have any updates on the eps32C5 ECO1 chip?

mantriyogesh, 2025-03-04:
@skylin008 any updates further?

mantriyogesh, 2025-07-08:
Have you tested on the latest IDF with C5 Eco2?

Is it possible to create a small example to reproduce the issue? Also, please add the SDKconfig generated?