espressif/esp-hosted-mcu #27: Using esp_bt.h with esp-hosted-mcu

Daniel-Abdi, 2025-01-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Given how esp_wifi_remote is able to wrap the standard wifi function calls and transport them over SDIO to the on-board ESP32C6 on the ESP32P4 development board, is this possible to implement with BLE (specifically NimBLE) as well? I'm trying to port some code that makes use of esp_bt.h and I am having trouble including it or otherwise using it in the program. What is best practice for this?

Thank you

### Use cases

The use case would be to wrap the standard esp_bt calls similar to how esp_wifi_remote wraps the wifi calls. This would allow for much simpler BT development for something like the ESP32P4 EV board.

### Alternatives

Maybe I can port all the calls in the library I'm using to use the functions provided by this library, but I would like to avoid that if possible.

mantriyogesh, 2025-01-20:
@SohKamYung-Espressif , PTAL

SohKamYung-Espressif, 2025-01-20:
@Daniel-Abdi BT usage over ESP-Hosted is different from Wi-FI.

For Wi-Fi, ESP-Hosted is transporting the wifi commands from the host to coprocessor for execution. esp_wifi_remote is wrapping the Wi-Fi commands for ESP-Hosted.

For BT, ESP-Hosted is transporting the HCI packets between the BT-Host on the host and the BT-Controller on the co-processor. An equivalent to esp_wifi_remote is not needed as no BT APIs (in `esp_bt.h`) are transported: only HCI packets.

See the [BT implementation document](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md) for more information.

See the BT examples [from the ESP-Registry for ESP-Hosted](https://components.espressif.com/components/espressif/esp_hosted/) for how to use NimBLE/Bluedroid for BLE with ESP-Hosted.

Daniel-Abdi, 2025-01-20:
I see. Thank you for your reply. Apologies for my ignorance on this topic. I've managed to get the vendor-provided SDK working with a bit of hacking, which is nice. I just have a couple of questions. I've managed to remove the dependency on esp_bt.h in the SDK. However, there is one call to esp_bt_mem_release that I have commented out for the time being. I don't need this particular function at the moment, but I would like to know. Is there a way to call this or get the same effect without esp_bt.h? I also notice that in esp-hosted, ble_transport_ll_init() is defined, but ble_transport_ll_deinit() does not seem to be. Is there any particular reason for this? Apologies if this is off topic in any way. Thank you for your assistance.

SohKamYung-Espressif, 2025-01-20:
> Is there a way to call this or get the same effect without esp_bt.h

Are you using NimBLE or Bluedroid?

If you are using NimBLE, then it should not be necessary to call `esp_bt_mem_release()` as it is meant for Bluedroid memory management.

According to the [documentation on `esp_bt_mem_release()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv418esp_bt_mem_release13esp_bt_mode_t):

> Release the Controller memory, BSS and data section of the Classic Bluetooth/BLE Host stack as per the mode. 

Since NimBLE is only for BLE, there is no BT Classic memory to release.

> ble_transport_ll_deinit() does not seem to be. Is there any particular reason for this? 

In the current architecture, once ESP-Hosted transport has been init, it remains running, so there is no deinit. If this changes in the future, a deinit will be added.

If you need a deinit, you can add a stub function for now.

Daniel-Abdi, 2025-01-22:
This has been very informative. Thank you for the information!

closed 2025-01-22 by Daniel-Abdi