espressif/esp-hosted-mcu #29: compile error

e-sr, 2025-02-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Following the documentation for [spi full duplex transport](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/spi_full_duplex.md) Chapter 8 i expect to be able to compile host  firmware starting from iperf example.
I did all steps like in Chapter 8

  

### Actual behavior (suspected bug)

I tried with ESP32 and ESP32c6 as host, in both cases there are compile problem related  to esp_wifi_remote
compiling `managed_components/espressif__esp_wifi_remote/idf_v5.4/esp_wifi_remote_weak.c`
missing definition of  `wifi_tx_rate_config_t`

### Error logs or terminal output

```shell
Executing action: all (aliases: build)
Running ninja in directory /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build
Executing "ninja all"...
[1/301] Performing build step for 'bootloader'
[1/1] cd /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build/bootloader/esp-idf/esptool_py && /home/enzo/.espressif/python_env/idf5.4_py3.11_env/bin/python /home/enzo/esp/v5.4/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build/bootloader/bootloader.bin
Bootloader binary size 0x6be0 bytes. 0x420 bytes (4%) free.
[2/301] No install step for 'bootloader'
[3/301] Completed 'bootloader'
[4/301] Building C object esp-idf/espressif__esp_wifi_remote/CMakeFiles/__idf_espressif__esp_wifi_remote.dir/idf_v5.4/esp_wifi_remote_weak.c.obj
FAILED: esp-idf/espressif__esp_wifi_remote/CMakeFiles/__idf_espressif__esp_wifi_remote.dir/idf_v5.4/esp_wifi_remote_weak.c.obj 
/home/enzo/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.4-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build/config -I/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/include -I/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/idf_v5.4/include -I/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/eppp -I/home/enzo/esp/v5.4/esp-idf/components/newlib/platform_include -I/home/enzo/esp/v5.4/esp-idf/components/freertos/config/include -I/home/enzo/esp/v5.4/esp-idf/components/freertos/config/include/freertos -I/home/enzo/esp/v5.4/esp-idf/components/freertos/config/xtensa/include -I/home/enzo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/enzo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/enzo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/enzo/esp/v5.4/esp-idf/components/freertos/esp_additions/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/include/soc -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/dma/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/ldo/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/debug_probe/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/port/esp32/. -I/home/enzo/esp/v5.4/esp-idf/components/esp_hw_support/port/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/heap/include -I/home/enzo/esp/v5.4/esp-idf/components/heap/tlsf -I/home/enzo/esp/v5.4/esp-idf/components/log/include -I/home/enzo/esp/v5.4/esp-idf/components/soc/include -I/home/enzo/esp/v5.4/esp-idf/components/soc/esp32 -I/home/enzo/esp/v5.4/esp-idf/components/soc/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/soc/esp32/register -I/home/enzo/esp/v5.4/esp-idf/components/hal/platform_port/include -I/home/enzo/esp/v5.4/esp-idf/components/hal/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/hal/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_rom/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_rom/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_rom/esp32/include/esp32 -I/home/enzo/esp/v5.4/esp-idf/components/esp_rom/esp32 -I/home/enzo/esp/v5.4/esp-idf/components/esp_common/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_system/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_system/port/soc -I/home/enzo/esp/v5.4/esp-idf/components/esp_system/port/include/private -I/home/enzo/esp/v5.4/esp-idf/components/xtensa/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/xtensa/include -I/home/enzo/esp/v5.4/esp-idf/components/xtensa/deprecated_include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/include/apps -I/home/enzo/esp/v5.4/esp-idf/components/lwip/include/apps/sntp -I/home/enzo/esp/v5.4/esp-idf/components/lwip/lwip/src/include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/port/include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/port/freertos/include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/port/esp32xx/include -I/home/enzo/esp/v5.4/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/enzo/esp/v5.4/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/enzo/esp/v5.4/esp-idf/components/esp_event/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_netif/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_wifi/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_wifi/include/local -I/home/enzo/esp/v5.4/esp-idf/components/esp_wifi/wifi_apps/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_wifi/wifi_apps/nan_app/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_phy/include -I/home/enzo/esp/v5.4/esp-idf/components/esp_phy/esp32/include -I/home/enzo/esp/v5.4/esp-idf/components/esp-tls -I/home/enzo/esp/v5.4/esp-idf/components/esp-tls/esp-tls-crypto -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/port/include -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/mbedtls/include -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/enzo/esp/v5.4/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/home/enzo/esp/v5.4/esp-idf/components/vfs/include -I/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__eppp_link/include -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=/home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf=. -fmacro-prefix-map=/home/enzo/esp/v5.4/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/espressif__esp_wifi_remote/CMakeFiles/__idf_espressif__esp_wifi_remote.dir/idf_v5.4/esp_wifi_remote_weak.c.obj -MF esp-idf/espressif__esp_wifi_remote/CMakeFiles/__idf_espressif__esp_wifi_remote.dir/idf_v5.4/esp_wifi_remote_weak.c.obj.d -o esp-idf/espressif__esp_wifi_remote/CMakeFiles/__idf_espressif__esp_wifi_remote.dir/idf_v5.4/esp_wifi_remote_weak.c.obj -c /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/idf_v5.4/esp_wifi_remote_weak.c
In file included from /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/include/esp_wifi_remote.h:8,
                 from /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/managed_components/espressif__esp_wifi_remote/idf_v5.4/esp_wifi_remote_weak.c:7:
[... cut ...]
ninja failed with exit code 1, output of the command is in the /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build/log/idf_py_stderr_output_26988 and /home/enzo/Documents/embedded/esp32/iperfhostesp32/iperf/build/log/idf_py_stdout_output_26988
```

### Steps to reproduce the behavior

idf.py build

### Project release version

idf 5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Fish

mantriyogesh, 2025-02-04:
Looks very much related to : https://github.com/espressif/esp-dev-kits/issues/120, please let us know ahead.

lukilukeskywalker, 2025-02-04:
It is the same problem, as in my ticket/issue

mantriyogesh, 2025-02-04:
Are you using 'ESP32' and/or 'ESP32C6' as **host**?

If so, did you disable the **native** Wi-Fi first as mentioned in:

https://github.com/espressif/esp-hosted-mcu/blob/main/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi ?

Which chip are you using as slave?

e-sr, 2025-02-04:
i have tried different combinations, ESP32C6 host and slave, ESP32C6 host ESP32 slave and ESP32 slave and ESP32C6 host,
i cannot compile host firmware. 
also  setting
```
  espressif/esp_wifi_remote: '0.5.0'
  espressif/esp_hosted: '0.0.6'
```
compile and link

mantriyogesh, 2025-02-04:
It's better to use latest hosted component.

I understand, It is little heavy to grasp, what I am trying to describe here, so ⚠ please go extremely slow ! 

Also, please note, if you use any esp chipset which have 'native' 'wifi', you need to disable it.
afterall, the intention is to get it working on some other mcu, which doesn't have wifi.

It could be esp chip itself like esp32-p4 or esp32-h2, which has non native wifi.
or some other non esp as host.

for esp with native wifi, need to be disabled through the 'soc' only. once you disable through soc, the native wifi would go away.

Also as a more special case, if you have two chips of same variant, for example, c6 as host and c6 as slave also,  or c3 as host and c3 as slave. in these cases, one acting as host need to be built and flashed 'with soc wifi' `disabled` and the other same chip now acting as slave, obviously need to enable 'soc' for wifi and built.

e-sr, 2025-02-04:
still compile error, does we speak from the same? can you please be more precise then ''with soc wifi' ' 
which fields?
i modified like this
[esp/v5.4/esp-idf/components/soc/esp32/include/soc/Kconfig.soc_caps.in](https://github.com/user-attachments/files/18659564/Kconfig.soc_caps.txt)
what i am missing?
also see dependencies
[dependencies.lock.txt](https://github.com/user-attachments/files/18659615/dependencies.lock.txt)

mantriyogesh, 2025-02-04:
Can you give your final try on the idf latest master?

If still doesn't work, will figure out with actual code and environment

hathorr-hub, 2025-02-04:
@mantriyogesh Yes I am able to build it with the master branch. When we can expect that this will be available in release branch?

e-sr, 2025-02-04:
> [@mantriyogesh](https://github.com/mantriyogesh) Yes I am able to build it with the master branch. When we can expect that this will be available in release branch?
can confirm build of host work
sadly slave build process(like in the docs) exit with link errors...

e-sr, 2025-02-04:
link error
```
esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_hostap.c.obj): in function `hostap_init':
/esp/v5.4/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_hostap.c:161:(.text.hostap_init+0x188): undefined reference to `esp_wifi_ap_get_transition_disable_internal'
```
disabling soft-ap capabilities from menuconfig let compile and link

host<->slave complete setup work  with above restriction.

SohKamYung-Espressif, 2025-02-05:
@e-sr Can you do a `git status` on your ESP-IDF code and paste the output here?

e-sr, 2025-02-05:
I  checked out and some submodules was not as should be. Now i can compile both slave(esp32c6) and host(esp32) like stated in the docs without any changes.
thanks a lot

as you asked `git status` 
```
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   components/soc/esp32/include/soc/Kconfig.soc_caps.in
```

closed 2025-02-05 by e-sr