espressif/esp-hosted-mcu #30: ESP32P4+ESP32C2      C2 malloc error

SSEHX, 2025-02-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

can connect wifi or create softap

### Actual behavior (suspected bug)

mem_dump free:83684 min-free:83652 lfb-dma:73728 lfb-def:73728 lfb-8bit:73728
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x17d8
load:0x403acb70,len:0xbd8
load:0x403aeb70,len:0x2904
entry 0x403acb7a
I (32) boot: ESP-IDF v5.3.2-500-gfb25eb02eb-dirty 2nd stage bootloader
I (32) boot: compile time Feb  5 2025 18:06:28
I (34) boot: chip revision: v1.2
I (40) boot: efuse block revision: v0.1
I (47) boot.esp32c2: MMU Page Size  : 32K
I (54) boot.esp32c2: SPI Speed      : 60MHz
I (61) boot.esp32c2: SPI Mode       : DIO
I (69) boot.esp32c2: SPI Flash Size : 2MB
I (76) boot: Enabling RNG early entropy source...
I (84) boot: Partition Table:
I (90) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (112) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (124) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (136) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (147) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (159) boot: End of partition table
I (166) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=2476ch (149356) map
I (212) esp_image: segment 1: paddr=00034794 vaddr=3fcb0e90 size=029c4h ( 10692) load
I (216) esp_image: segment 2: paddr=00037160 vaddr=40380000 size=00eb8h (  3768) load
I (222) esp_image: segment 3: paddr=00038020 vaddr=42000020 size=706bch (460476) map
I (339) esp_image: segment 4: paddr=000a86e4 vaddr=40380eb8 size=0ffd0h ( 65488) load
I (360) boot: Loaded app from partition at offset 0x10000
I (361) boot: Disabling RNG early entropy source...
I (379) cpu_start: Unicore app
I (387) cpu_start: Pro cpu start user code
I (387) cpu_start: cpu freq: 120000000 Hz
I (388) app_init: Application information:
I (392) app_init: Project name:     network_adapter
I (400) app_init: App version:      1
I (407) app_init: Compile time:     Feb  5 2025 18:06:24
I (416) app_init: ELF file SHA256:  bf2ed6a42...
I (424) app_init: ESP-IDF:          v5.3.2-500-gfb25eb02eb-dirty
I (435) efuse_init: Min chip rev:     v1.0
I (442) efuse_init: Max chip rev:     v1.99 
I (450) efuse_init: Chip rev:         v1.2
I (457) heap_init: Initializing. RAM available for dynamic allocation:
I (468) heap_init: At 3FCB8050 len 00024B20 (146 KiB): RAM
I (478) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
I (488) spi_flash: detected chip: generic
I (494) spi_flash: flash io: dio
I (501) sleep: Configure to isolate all GPIO pins in sleep state
I (511) sleep: Enable automatic switching of GPIO sleep configuration
I (522) main_task: Started on CPU0
[... 143 lines cut ...]
change main xtal to 26Mhz 
save 
download to esp32c2

esp32p4 use esp-remote and esp-hosted connect wifi, (the software is run to esp32c6 good, now change to esp32c2 cannot run)

### Project release version

slavel v1.1.2

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

macos

### Shell

ZSH

mantriyogesh, 2025-02-05:
1. You have changed the xtal to 26Mhz. So I'm assuming that you have 26MHz xtal chip and not the 40 MHz  C2 chip.
2. This is because of the very high buffers configured at P4 (default config from documentation holds true for most other chips except C2).

There are two parts to handle this:
1. Reduce the wifi buffer at P4
2. Lower the C2 memory
As C2 has lower on-board memory, some constraints apply to lower self memory usage as well.
  - 2.1 correct and optimise mempool
  - 2.2 disable bluetooth if or when not required (either at the start or later stage, when it is no more needed, depending upon application use)

I will correct this on latest code and point you to the solution.
Please note above discussion application only for special esp32-c2 case only, due to lower on board memory.. other chips would not required these adjustments.

SSEHX, 2025-02-05:
thank you !
1： The C2 chip defaults to using a 26MHz chip, which is the official recommendation
1.1 ：the C2 chip is good ，we use ESP-IDF SOFTAP example can create softap ，the hardware is good

2: this is C2 slave menuconfig from change last 

[image]

this is p4 menuconfig from change last 

[image]

the software no error log ,However, the softap has not been successfully created yet

the softap is cant create !!!!!!!!!!!!!!!, my phone and computer cant scan this softap

`mem_dump free:83272 min-free:83240 lfb-dma:73728 lfb-def:73728 lfb-8bit:73728
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x1948
load:0x403acb70,len:0xd90
load:0x403aeb70,len:0x29f8
entry 0x403acb7a
I (32) boot: ESP-IDF v5.3.2-500-gfb25eb02eb-dirty 2nd stage bootloader
I (32) boot: compile time Feb  5 2025 21:27:52
I (34) boot: chip revision: v1.2
I (40) boot: efuse block revision: v0.1
I (47) qio_mode: Enabling default flash chip QIO
I (55) boot.esp32c2: MMU Page Size  : 32K
I (62) boot.esp32c2: SPI Speed      : 60MHz
I (69) boot.esp32c2: SPI Mode       : QIO
I (77) boot.esp32c2: SPI Flash Size : 2MB
I (84) boot: Enabling RNG early entropy source...
I (92) boot: Partition Table:
I (98) boot: ## Label            Usage          Type ST Offset   Length
I (109) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (121) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (132) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (144) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (155) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (167) boot: End of partition table
I (174) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=2476ch (149356) map
I (214) esp_image: segment 1: paddr=00034794 vaddr=3fcb0e90 size=029c4h ( 10692) load
I (217) esp_image: segment 2: paddr=00037160 vaddr=40380000 size=00eb8h (  3768) load
I (224) esp_image: segment 3: paddr=00038020 vaddr=42000020 size=706b8h (460472) map
I (320) esp_image: segment 4: paddr=000a86e0 vaddr=40380eb8 size=0ffd0h ( 65488) load
I (339) boot: Loaded app from partition at offset 0x10000
I (339) boot: Disabling RNG early entropy source...
I (357) cpu_start: Unicore app
I (366) cpu_start: Pro cpu start user code
I (366) cpu_start: cpu freq: 120000000 Hz
I (366) app_init: Application information:
I (370) app_init: Project name:     network_adapter
I (379) app_init: App version:      1
I (386) app_init: Compile time:     Feb  5 2025 21:27:55
I (395) app_init: ELF file SHA256:  9a9205730...
I (403) app_init: ESP-IDF:          v5.3.2-500-gfb25eb02eb-dirty
I (413) efuse_init: Min chip rev:     v1.0
I (420) efuse_init: Max chip rev:     v1.99 
I (428) efuse_init: Chip rev:         v1.2
I (436) heap_init: Initializing. RAM available for dynamic allocation:
I (447) heap_init: At 3FCB8050 len 00024B20 (146 KiB): RAM
I (456) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
I (466) spi_flash: detected chip: generic
I (473) spi_flash: flash io: qio
I (479) sleep: Configure to isolate all GPIO pins in sleep state
I (489) sleep: Enable automatic switching of GPIO sleep configuration
[... 56 lines cut ...]
I (2170) slave_ctrl: Resp_MSGId for req[0x104] is [0x204], uid 2
I (2177) slave_ctrl: Received Req [0x104]
I (2187) slave_ctrl: Resp_MSGId for req[0x11c] is [0x21c], uid 3
I (2195) slave_ctrl: Received Req [0x11c]
I (2207) slave_ctrl: Resp_MSGId for req[0x118] is [0x218], uid 4
I (2212) slave_ctrl: Received Req [0x118]
I (2220) phy_init: phy_version 340,e255ce0,Jun  4 2024,16:44:11
I (2258) wifi:mode : sta (c0:5d:89:50:27:64) + softAP (c0:5d:89:50:27:65)
I (2258) wifi:enable tsf
I (2441) wifi:Total power save buffer number: 3
I (2441) wifi:Init max length of beacon: 752/752
I (2441) wifi:Init max length of beacon: 752/752
I (2445) slave_ctrl: Sending Wi-Fi event [43]
I (2452) slave_ctrl: Sending Wi-Fi event [2]
I (2459) slave_ctrl: softap started
I (2461) slave_ctrl: Resp_MSGId for req[0x101] is [0x201], uid 5
I (2476) slave_ctrl: Received Req [0x101]
I (2483) slave_ctrl: mac [c0:5d:89:50:27:64]
I (2491) slave_ctrl: Sending Wi-Fi event [12]
I (2500) slave_ctrl: Resp_MSGId for req[0x101] is [0x201], uid 6
I (2508) slave_ctrl: Received Req [0x101]
I (2516) slave_ctrl: mac [c0:5d:89:50:27:65]
mem_dump free:21500 min-free:21376 lfb-dma:11264 lfb-def:11264 lfb-8bit:11264
mem_dump free:21500 min-free:21376 lfb-dma:11264 lfb-def:11264 lfb-8bit:11264
mem_dump free:21500 min-free:21376 lfb-dma:11264 lfb-def:11264 lfb-8bit:11264`

SSEHX, 2025-02-05:
### **This is the log of P4 attempting to connect to other wifi using C2, even though the wifi router is nearby and the device cannot connect**

mem_dump free:83272 min-free:83240 lfb-dma:73728 lfb-def:73728 lfb-8bit:73728
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x1948
load:0x403acb70,len:0xd90
load:0x403aeb70,len:0x29f8
entry 0x403acb7a
I (32) boot: ESP-IDF v5.3.2-500-gfb25eb02eb-dirty 2nd stage bootloader
I (32) boot: compile time Feb  5 2025 21:27:52
I (34) boot: chip revision: v1.2
I (40) boot: efuse block revision: v0.1
I (47) qio_mode: Enabling default flash chip QIO
I (55) boot.esp32c2: MMU Page Size  : 32K
I (62) boot.esp32c2: SPI Speed      : 60MHz
I (69) boot.esp32c2: SPI Mode       : QIO
I (77) boot.esp32c2: SPI Flash Size : 2MB
I (84) boot: Enabling RNG early entropy source...
I (92) boot: Partition Table:
I (98) boot: ## Label            Usage          Type ST Offset   Length
I (109) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (121) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (132) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (144) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (155) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (167) boot: End of partition table
I (174) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=2476ch (149356) map
I (214) esp_image: segment 1: paddr=00034794 vaddr=3fcb0e90 size=029c4h ( 10692) load
I (217) esp_image: segment 2: paddr=00037160 vaddr=40380000 size=00eb8h (  3768) load
I (224) esp_image: segment 3: paddr=00038020 vaddr=42000020 size=706b8h (460472) map
I (320) esp_image: segment 4: paddr=000a86e0 vaddr=40380eb8 size=0ffd0h ( 65488) load
I (339) boot: Loaded app from partition at offset 0x10000
I (339) boot: Disabling RNG early entropy source...
I (357) cpu_start: Unicore app
I (366) cpu_start: Pro cpu start user code
I (366) cpu_start: cpu freq: 120000000 Hz
I (366) app_init: Application information:
I (370) app_init: Project name:     network_adapter
I (379) app_init: App version:      1
I (386) app_init: Compile time:     Feb  5 2025 21:27:55
I (395) app_init: ELF file SHA256:  9a9205730...
I (403) app_init: ESP-IDF:          v5.3.2-500-gfb25eb02eb-dirty
I (413) efuse_init: Min chip rev:     v1.0
I (420) efuse_init: Max chip rev:     v1.99 
I (428) efuse_init: Chip rev:         v1.2
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (447) heap_init: At 3FCB8050 len 00024B20 (146 KiB): RAM
I (456) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
I (466) spi_flash: detected chip: generic
I (473) spi_flash: flash io: qio
I (479) sleep: Configure to isolate all GPIO pins in sleep state
I (489) sleep: Enable automatic switching of GPIO sleep configuration
I (500) main_task: Started on CPU0
I (506) main_task: Calling app_main()
I (512) fg_mcu_slave: *********************************************************************
I (526) fg_mcu_slave:                 ESP-Hosted-MCU Slave FW version :: 1.1.2                        
I (541) fg_mcu_slave:                 Transport used :: SPI only                      
I (555) fg_mcu_slave: *********************************************************************
I (569) fg_mcu_slave: Supported features are:
I (576) fg_mcu_slave: - WLAN over SPI
I (583) fg_mcu_slave: capabilities: 0xa0
I (590) fg_mcu_slave: Supported extended features are:
I (599) fg_mcu_slave: extended capabilities: 0x0
I (620) SPI_DRIVER: Using SPI interface
I (620) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (628) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 130 lines cut ...]
I (14492) slave_ctrl: Received Req [0x11a]
I (14492) slave_ctrl: ************ connect ****************
W (14502) wifi:m f probe req l=0

W (14627) wifi:m f probe req l=0

W (14748) wifi:m f probe req l=0

W (14868) wifi:m f probe req l=0

W (14988) wifi:m f probe req l=0

W (15109) wifi:m f probe req l=0

W (15229) wifi:m f probe req l=0

W (15349) wifi:m f probe req l=0

W (15470) wifi:m f probe req l=0

W (15590) wifi:m f probe req l=0

W (15711) wifi:m f probe req l=0

I (16912) slave_ctrl: Sta mode disconnect

SSEHX, 2025-02-05:
Can you provide a usable version or configuration? We are prepared to delay for the customer, performance is not important 

### Thank YOU!!!

SSEHX, 2025-02-05:
[image]
this is 26Mhz my hardware is 26Mhz

SSEHX, 2025-02-05:
i change to 40Mhz Still unable to work

mantriyogesh, 2025-02-05:
If it is 26 Mhz, please don't use xtal as 40.

mantriyogesh, 2025-02-05:
P4 sdkconfig.defaults should have wifi comfigs not more than that of:

https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/examples/wifi/iperf/sdkconfig.defaults.esp32c2#L4-L10

Basically these wifi Params are going to be sent to C2 for wifi init using RPC calls from P4->C2.
Apart from wifi parameters in above link would not be needed to be added at P4.
Basically other Params would be local to P4.

SSEHX, 2025-02-05:
[image]
I understand this mechanism,this is My p4 sdkconfig

mantriyogesh, 2025-02-05:
⚠ I have not tested above code on latest  commit.

There may also be merge conflicts (although I expect not to)

My point is I just tried building with above patch. ** NOT TESTED **

just provided for quick response, in case useful.

SSEHX, 2025-02-05:
Sorry, this patch still doesn't work, buddy. I'm going to break it

mantriyogesh, 2025-02-05:
In history, I have observed that C2 would not connect to wifi, if the xtal freq is not set to what it should be.
But in this case you already would be using 26 xal as the flash says it is 26.

SSEHX, 2025-02-05:
There was no memory crash or error, but I couldn't connect to WiFi or create a softap. I tried the ESP-IDF WiFi connection program, and the chip was working fine,

mantriyogesh, 2025-02-05:
Even the station ? I need a favour, Can you read the chip and tell exact name ? Or box if you have with you?

It's very suprising, that chip works on IDF default program and not with hosted(??)!
In general hosted uses spi transport and spi communication is just fine within two chips, as I understand.

Can you please attach the sdkconfig both sides and logs both sides?
Also what is the memory used ?

Also, can you change P4 wifi config side,
https://github.com/espressif/esp-hosted-mcu/issues/30#issuecomment-2636970043

To the IDF iperf default for c2, https://github.com/espressif/esp-hosted-mcu/issues/30#issuecomment-2636963290?

Ensure whenever you change sdkconfig, ensure to run

idf.py fullclean
Or
rm -rf sdkconfig build/
idf.py set-target <chip-type>

Both sides

SSEHX, 2025-02-05:
[esp-hosted-c2.zip](https://github.com/user-attachments/files/18675196/esp-hosted-c2.zip)
this is commit was https://github.com/espressif/esp-hosted-mcu/commit/9bc802b72be72aa83b9df3a5113ae320fa63e446

[sdkconfig-P4.txt](https://github.com/user-attachments/files/18675230/sdkconfig-P4.txt)

this is esp32p4 sdkconfig

Later on, I will provide a complete screen recording for troubleshooting. I am a professional developer and the chip model is completely correct. We have conducted similar development many times, and this is the first time we have connected P4 and C2

mantriyogesh, 2025-02-05:
I know and realise you are deep into the issue already.
It was more of surprise and a puzzle for me, what could have gone..

Anyway, I can check the logs tomorrow morning and get you back on this.

SSEHX, 2025-02-05:
https://github.com/user-attachments/assets/e2d0cc13-df18-4bd2-a4ff-3215dd8d09db

SSEHX, 2025-02-05:
https://github.com/user-attachments/assets/a5685952-6348-4966-a1f8-7a5809db8a6a

mantriyogesh, 2025-02-05:
Very basic questions, just want to rule out, before trying this tomorrow:

1. `git status` in IDF directory? All submodules clean?
2. All the connections made in p4-c2?
3. Can you take a photo how you connected gpios (to understand the jumpers if there any)? Also any schematic? All connections including the slave reset?

SSEHX, 2025-02-05:
1： yes , rm -rf build/
2:  gpio is all connect good 
3 :  no jumpers

[image]

all gpio is connect include chip_en 

spi is can translate data you can see in the log

SohKamYung-Espressif, 2025-02-06:
@SSEHX In your `sdkconfig-P4.txt` I see this:

```
# CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0 is not set
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y
```

This will cause the ESP32-C2 to try to run the wifi task on Core 1, which does not exist on the C2.

Reset it to

```
CONFIG_ESP_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP_WIFI_TASK_PINNED_TO_CORE_1 is not set
```

To fix the issue. Wi-Fi should work as expected.

SSEHX, 2025-02-06:
**### Wow, it's working. Thank you very much for everyone's support ！！！！**

closed 2025-02-06 by mantriyogesh