espressif/esp-hosted-mcu #31: Lack of support for esp_wifi_scan_get_ap_record

mpolitowicz, 2025-02-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Add support for 'esp_wifi_scan_get_ap_record' in esp-hosted / esp-slave

### Use cases

The code for esp-hosted / esp-slave has support for 'esp_wifi_scan_get_ap_records' (plural) but lacks of support for 'esp_wifi_scan_get_ap_record' (singular) 

Calling 'esp_wifi_scan_get_ap_record' results message as below followed by hosted crash:
```
W (54854) esp_wifi_remote_weak: esp_wifi_remote_scan_get_ap_record unsupported
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x4801d2ae
```

### Alternatives

None. 

### Additional context

'esp_wifi_scan_get_ap_record' is part of standard esp_wifi interface.

mantriyogesh, 2025-02-10:
That's right , @mpolitowicz .
Thank you for bringing this up.

@SohKamYung-Espressif , Can we add such API?

SohKamYung-Espressif, 2025-02-10:
Can be added. But will need to add `esp_wifi_clear_ap_list()` also to prevent memory leaks in scan list.

SohKamYung-Espressif, 2025-02-11:
> esp_wifi_clear_ap_list() is already implemented

You are correct. That will be tested along with `esp_wifi_scan_get_ap_record` when added.

closed 2025-02-12 by espressif-bot

SohKamYung-Espressif, 2025-02-13:
@mpolitowicz  `esp_wifi_scan_get_ap_record` command added as part of https://github.com/espressif/esp-hosted-mcu/commit/2ea607357d676db41f8ba3210549533ba8cbcde2

mpolitowicz, 2025-02-13:
Thank you for super fast action ! We already tested new version 1.1.3 in our code and it works like a charm.