espressif/esp-hosted-mcu #32: esp32p4 ble client with esp_hosted

sign25893, 2025-02-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi people, recently bought myself ESP32P4 evolution board kit, figured out the Wi-Fi, LVGL, but I can not understand how to deal with BLE, I use the stack NimBle. In the Internet did not find normal examples for creating a client, because my task is to connect to a device (server) and periodically receive data. I tried to use GATT, but due to, probably, misunderstanding, I didn't understand where to move further. Unfortunately, I can't attach the code, because I don't have it. The task is to create a ble client on esp32p4 with esp32c6 via SDIO connected. Help who can, I will be very grateful!!! :( 
I'll provide the logs below.

### Actual behavior (suspected bug)

It seems to me that there is no support for connectivity in the nimble stack, because it hasn't been implemented yet.
below is the code under test:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "host/util/util.h"
#include "services/gap/ble_svc_gap.h"
#include "services/gatt/ble_svc_gatt.h"
#include "esp_heap_caps.h"

// Целевой UUID для поиска (128-bit UUID)
#define TARGET_SERVICE_UUID { 0x4F, 0xAF, 0xC2, 0x01, 0x1F, 0xB5, 0x45, 0x9E, 0x8F, 0xCC, 0xC5, 0xC9, 0xC3, 0x31, 0x91, 0x4B }

static const char *TAG = "BLE-Client";
static uint8_t ble_addr_type;
static uint16_t conn_handle = BLE_HS_CONN_HANDLE_NONE; // Дескриптор соединения

// Объявление функции обработки GAP событий
static int ble_gap_event(struct ble_gap_event *event, void *arg);

// Функция сканирования BLE-устройств
static void ble_app_scan(void) {
    struct ble_gap_disc_params disc_params;
    memset(&disc_params, 0, sizeof(disc_params));
    disc_params.itvl = 80;  // Интервал сканирования
    disc_params.window = 40; // Время сканирования
    disc_params.filter_policy = BLE_HCI_SCAN_FILT_NO_WL;
    disc_params.passive = 0; 

    ESP_LOGI(TAG, "Starting BLE scan...");
    ble_gap_disc(ble_addr_type, BLE_HS_FOREVER, &disc_params, ble_gap_event, NULL);
}

// Функция инвертирования UUID
static void reverse_uuid(uint8_t *uuid) {
    for (int i = 0; i < 8; i++) {
        uint8_t tmp = uuid[i];
        uuid[i] = uuid[15 - i];
        uuid[15 - i] = tmp;
    }
}

// Функция обработки GAP событий
static int ble_gap_event(struct ble_gap_event *event, void *arg) {
    switch (event->type) {
    case BLE_GAP_EVENT_DISC: {
[... 188 lines cut ...]
4. Running the code

### Project release version

esp-idf v5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

Использую esp32p4 evolution board с заводской прошивкой для esp32c6, esp_hosted - v1.1.2, esp_wifi_remote - v0.6.2

volca, 2025-02-21:
I am also encountering the same issue

volca, 2025-02-21:
> Does the ESP-Hosted [BLE Host-Only Peripheral VHCI Example](https://github.com/espressif/esp-hosted-mcu/tree/main/examples/host_nimble_bleprph_host_only_vhci) work? It uses esp-NimBLE to set up a BLE Client on the ESP32-P4 that you can test with a BLE app on mobile like LightBlue. You can use it as a reference.

The example works well.

The issue can be reproduced when you make a ble-central project. I believe the reason is that the event `BLE_GAP_EVENT_DISC` is triggered too quickly with the line

```
    ble_gap_disc(ble_addr_type, BLE_HS_FOREVER, &disc_params, ble_gap_event, NULL);
```

mantriyogesh, 2025-02-24:
@volca , 

Can you please send a small code to reproduce the issue you face?
Also, please share exact expectation (what fails exactly)?

If it is related to esp-hosted, we would be happy to reproduce and fix it asap.

mantriyogesh, 2025-03-04:
@volca , any updates?

volca, 2025-03-04:
Sorry. I'm a little busy recently. I will upload the code to reproduce the issue this afternoon.

volca, 2025-03-04:
> [@volca](https://github.com/volca) , any updates?

Please check the attachment. You can reproduce the issue with the code

* The code is copied from the example `host_nimble_bleprph_host_only_vhci`. I modified the code to do BLE scan only.
* esp-idf release/v5.4
* The log for esp32p4
```
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x10f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x162c
load:0x4ff2abd0,len:0xd70
load:0x4ff2cbd0,len:0x32fc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4-552-ge37d33cc1c-dirty 2nd stage bootloader
I (26) boot: compile time Mar  4 2025 14:43:33
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (30) boot: efuse block revision: v0.0
I (34) boot.esp32p4: SPI Speed      : 80MHz
I (38) boot.esp32p4: SPI Mode       : DIO
I (41) boot.esp32p4: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=40060020 size=23cc0h (146624) map
I (114) esp_image: segment 1: paddr=00033ce8 vaddr=30100000 size=0000ch (    12) load
I (116) esp_image: segment 2: paddr=00033cfc vaddr=3010000c size=00038h (    56) load
I (120) esp_image: segment 3: paddr=00033d3c vaddr=4ff00000 size=0c2dch ( 49884) load
I (137) esp_image: segment 4: paddr=00040020 vaddr=40000020 size=5de9ch (384668) map
I (202) esp_image: segment 5: paddr=0009dec4 vaddr=4ff0c2dc size=03d20h ( 15648) load
I (207) esp_image: segment 6: paddr=000a1bec vaddr=4ff10000 size=024cch (  9420) load
I (214) boot: Loaded app from partition at offset 0x10000
I (214) boot: Disabling RNG early entropy source...
I (227) cpu_start: Multicore app
I (237) cpu_start: Pro cpu start user code
I (237) cpu_start: cpu freq: 360000000 Hz
I (238) app_init: Application information:
I (238) app_init: Project name:     bleprph_host_only_vhci
I (243) app_init: App version:      1
I (246) app_init: Compile time:     Mar  4 2025 14:43:30
I (251) app_init: ELF file SHA256:  5349a2f4a...
I (255) app_init: ESP-IDF:          v5.4-552-ge37d33cc1c-dirty
I (261) efuse_init: Min chip rev:     v0.1
I (265) efuse_init: Max chip rev:     v1.99
I (269) efuse_init: Chip rev:         v0.1
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (279) heap_init: At 4FF157C0 len 00025800 (150 KiB): RAM
I (284) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (289) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (295) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (300) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (306) spi_flash: detected chip: generic
I (308) spi_flash: flash io: dio
W (311) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (324) host_init: ESP Hosted : Host chip_ip[18]
I (355) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:183836 min-free:183836 lfb-def:139264 lfb-8bit:139264

I (359) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (367) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (376) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (384) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (392) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[... 86 lines cut ...]
E (5639) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5639) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5649) NimBLE_CENT: adv found
E (5649) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5659) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5659) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5669) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5669) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5679) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5679) NimBLE_CENT: adv found
E (5679) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5689) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5689) NimBLE_CENT: adv found
E (5689) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5699) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5699) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5709) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5709) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5719) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5719) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5729) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5729) vhci_drv: Rx: failed transport_alloc_evt(1)
```

[host_nimble_blescan_host_only_vhci.zip](https://github.com/user-attachments/files/19065575/host_nimble_blescan_host_only_vhci.zip)

mantriyogesh, 2025-03-04:
Are you using latest registry component for esp-hosted-mcu?

If unsure can you do 
idf.py fullclean
idf.py -p <host-port> flash monitor 

Attach full log from build..

volca, 2025-03-04:
Yes. I just tried rebuild with your steps. But no lucky. See the full build log

[build.log](https://github.com/user-attachments/files/19066145/build.log)

SohKamYung-Espressif, 2025-03-04:
@volca I just built and flash your example, and it seems to be ok for me.

```
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x162c
load:0x4ff2abd0,len:0xd70
load:0x4ff2cbd0,len:0x32fc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4-553-gcaaab2c1a0 2nd stage bootloader
I (26) boot: compile time Mar  4 2025 16:21:35
I (26) boot: Multicore bootloader
I (28) boot: chip revision: v1.0
I (30) boot: efuse block revision: v0.1
I (33) boot.esp32p4: SPI Speed      : 80MHz
I (37) boot.esp32p4: SPI Mode       : DIO
I (41) boot.esp32p4: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=40060020 size=23cd0h (146640) map
I (114) esp_image: segment 1: paddr=00033cf8 vaddr=30100000 size=0000ch (    12) load
I (116) esp_image: segment 2: paddr=00033d0c vaddr=3010000c size=00038h (    56) load
I (119) esp_image: segment 3: paddr=00033d4c vaddr=4ff00000 size=0c2cch ( 49868) load
I (136) esp_image: segment 4: paddr=00040020 vaddr=40000020 size=5de9ch (384668) map
I (202) esp_image: segment 5: paddr=0009dec4 vaddr=4ff0c2cc size=03d30h ( 15664) load
I (206) esp_image: segment 6: paddr=000a1bfc vaddr=4ff10000 size=024cch (  9420) load
I (213) boot: Loaded app from partition at offset 0x10000
I (213) boot: Disabling RNG early entropy source...
I (227) cpu_start: Multicore app
I (238) cpu_start: Pro cpu start user code
I (238) cpu_start: cpu freq: 360000000 Hz
I (238) app_init: Application information:
I (238) app_init: Project name:     bleprph_host_only_vhci
I (243) app_init: App version:      1
I (246) app_init: Compile time:     Mar  4 2025 16:21:28
I (251) app_init: ELF file SHA256:  a2d1f0432...
I (256) app_init: ESP-IDF:          v5.4-553-gcaaab2c1a0
I (261) efuse_init: Min chip rev:     v0.1
I (265) efuse_init: Max chip rev:     v1.99
I (269) efuse_init: Chip rev:         v1.0
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (279) heap_init: At 4FF157C0 len 00025800 (150 KiB): RAM
I (284) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (289) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (295) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (300) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (306) spi_flash: detected chip: generic
I (308) spi_flash: flash io: dio
W (311) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (324) host_init: ESP Hosted : Host chip_ip[18]
I (355) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:183836 min-free:183836 lfb-def:139264 lfb-8bit:139264

I (359) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (367) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (376) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (384) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (392) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (401) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (409) H_API: ** add_esp_wifi_remote_channels **
I (413) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000cc68] Rx[0x4001b5fa]
--- 0x4000cc68: transport_drv_sta_tx at /home/kysoh/projects/test/blescan/host_nimble_blescan_host_only_vhci/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:209
0x4001b5fa: esp_wifi_remote_channel_rx at /home/kysoh/projects/test/blescan/host_nimble_blescan_host_only_vhci/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
[... 143 lines cut ...]
I (5999) NimBLE_CENT: adv found
I (5999) NimBLE_CENT: adv found
```

The list of components I used are the same as in your `build.log`.

```
NOTICE: Updating lock file at /home/kysoh/projects/test/blescan/host_nimble_blescan_host_only_vhci/dependencies.lock
NOTICE: Processing 5 dependencies:
NOTICE: [1/5] espressif/eppp_link (0.2.0)
NOTICE: [2/5] espressif/esp_hosted (1.3.2)
NOTICE: [3/5] espressif/esp_serial_slave_link (1.1.0)
NOTICE: [4/5] espressif/esp_wifi_remote (0.5.5)
NOTICE: [5/5] idf (5.4.0)
```

For reference, this is my idf version
```shell
> idf.py --version
ESP-IDF v5.4-553-gcaaab2c1a0
```

Can you try my `bleprph_host_only_vhci.bin` which I have attached.

[host_nimble_blescan_20250304.zip](https://github.com/user-attachments/files/19066963/host_nimble_blescan_20250304.zip)

volca, 2025-03-05:
> Can you try my `bleprph_host_only_vhci.bin` which I have attached.

I flashed the `host_nimble_blescan_20250304.zip` but it still goto the error.

* I did `idf.py erase-flash` first and then flash the firmware
* The chip revision of esp32p4 is v0.1 and yours is v1.0
* Here're many BLE devices in my office.

this is my idf version

```
> idf.py --version
ESP-IDF v5.4-552-ge37d33cc1c-dirty
```

log
```
boot: ESP-IDF v5.4-552-ge37d33cc1c-dirty 2nd stage bootloader
I (26) boot: compile time Mar  4 2025 15:36:54
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (30) boot: efuse block revision: v0.0
I (34) boot.esp32p4: SPI Speed      : 80MHz
I (38) boot.esp32p4: SPI Mode       : DIO
I (41) boot.esp32p4: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=40060020 size=23cd0h (146640) map
I (114) esp_image: segment 1: paddr=00033cf8 vaddr=30100000 size=0000ch (    12) load
I (116) esp_image: segment 2: paddr=00033d0c vaddr=3010000c size=00038h (    56) load
I (120) esp_image: segment 3: paddr=00033d4c vaddr=4ff00000 size=0c2cch ( 49868) load
I (137) esp_image: segment 4: paddr=00040020 vaddr=40000020 size=5de9ch (384668) map
I (202) esp_image: segment 5: paddr=0009dec4 vaddr=4ff0c2cc size=03d30h ( 15664) load
I (207) esp_image: segment 6: paddr=000a1bfc vaddr=4ff10000 size=024cch (  9420) load
I (213) boot: Loaded app from partition at offset 0x10000
I (214) boot: Disabling RNG early entropy source...
I (227) cpu_start: Multicore app
I (237) cpu_start: Pro cpu start user code
I (237) cpu_start: cpu freq: 360000000 Hz
I (237) app_init: Application information:
I (238) app_init: Project name:     bleprph_host_only_vhci
I (242) app_init: App version:      1
I (246) app_init: Compile time:     Mar  4 2025 16:21:28
I (251) app_init: ELF file SHA256:  a2d1f0432...
--- Warning: Checksum mismatch between flashed and built applications. Checksum of built application is 2ecbad2d0f801faf149529a6f2398c78f5d9882f81d177f741eacb17f627f4b6
I (255) app_init: ESP-IDF:          v5.4-553-gcaaab2c1a0
I (260) efuse_init: Min chip rev:     v0.1
I (264) efuse_init: Max chip rev:     v1.99
I (268) efuse_init: Chip rev:         v0.1
I (272) heap_init: Initializing. RAM available for dynamic allocation:
I (278) heap_init: At 4FF157C0 len 00025800 (150 KiB): RAM
I (283) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (288) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (294) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (299) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (305) spi_flash: detected chip: generic
I (308) spi_flash: flash io: dio
W (311) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (323) host_init: ESP Hosted : Host chip_ip[18]
I (354) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:183836 min-free:183836 lfb-def:139264 lfb-8bit:139264

I (359) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (367) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (375) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[... 83 lines cut ...]
I (5618) NimBLE_CENT: adv found
I (5618) NimBLE_CENT: adv found
E (5618) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5628) NimBLE_CENT: adv found
E (5628) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5628) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5628) NimBLE_CENT: adv found
E (5638) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5648) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5648) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5658) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5658) NimBLE_CENT: adv found
E (5658) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5668) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5668) NimBLE_CENT: adv found
E (5668) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5678) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5688) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5688) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5688) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5698) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5698) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5708) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5708) vhci_drv: Rx: failed transport_alloc_evt(1)
```

mantriyogesh, 2025-03-05:
@volca can you please capture close photos of the device?

volca, 2025-03-05:
Here is the photo. I ordered the board from espressif's taobao store.

[image]

SohKamYung-Espressif, 2025-03-05:
@volca Can you try this. Start `idf.py menuconfig`, navigate to `Component config` -> `Bluetooth` -> `NimBLE Options` -> `Memory Settings` -> `Discardable Transport Event Buffer Count` and increase the value. Try `10` or a higher number.

The `failed transport_alloc_evt(1)` means it failed to allocate a buffer for a event. The number of buffers is controlled by that config setting.

From the help message for that config setting:

```
Name: BT_NIMBLE_TRANSPORT_EVT_DISCARD_COUNT
Prompt: Discardable Transport Event Buffer count
Type: int
Value: '8'

Help:

  This is the low priority HCI events' buffer size. Low-priority event
  buffers are only used for advertising reports. If there are no free
  low-priority event buffers, then an incoming advertising report will
  get dropped
```

volca, 2025-03-05:
> Try `10` or a higher number.

I tried numbers `10`, `12`, `32`, `64`.  It seems works much longer with higher number. 
But it still went to error finally.

```
I (5589) NimBLE_CENT: adv found
I (5589) NimBLE_CENT: adv found
I (5589) NimBLE_CENT: adv found
I (5599) NimBLE_CENT: adv found
I (5599) NimBLE_CENT: adv found
I (5599) NimBLE_CENT: adv found
I (5599) NimBLE_CENT: adv found
I (5599) NimBLE_CENT: adv found
I (5609) NimBLE_CENT: adv found
I (5609) NimBLE_CENT: adv found
I (5609) NimBLE_CENT: adv found
I (5609) NimBLE_CENT: adv found
I (5619) NimBLE_CENT: adv found
I (5619) NimBLE_CENT: adv found
I (5619) NimBLE_CENT: adv found
I (5629) NimBLE_CENT: adv found
I (5629) NimBLE_CENT: adv found
I (5629) NimBLE_CENT: adv found
I (5629) NimBLE_CENT: adv found
I (5639) NimBLE_CENT: adv found
I (5639) NimBLE_CENT: adv found
I (5639) NimBLE_CENT: adv found
I (5649) NimBLE_CENT: adv found
I (5649) NimBLE_CENT: adv found
I (5649) NimBLE_CENT: adv found
I (5649) NimBLE_CENT: adv found
I (5659) NimBLE_CENT: adv found
I (5659) NimBLE_CENT: adv found
I (5659) NimBLE_CENT: adv found
I (5669) NimBLE_CENT: adv found
I (5669) NimBLE_CENT: adv found
I (5669) NimBLE_CENT: adv found
I (5669) NimBLE_CENT: adv found
I (5679) NimBLE_CENT: adv found
I (5679) NimBLE_CENT: adv found
I (5679) NimBLE_CENT: adv found
I (5689) NimBLE_CENT: adv found
I (5689) NimBLE_CENT: adv found
I (5689) NimBLE_CENT: adv found
I (5689) NimBLE_CENT: adv found
I (5699) NimBLE_CENT: adv found
I (5699) NimBLE_CENT: adv found
I (5699) NimBLE_CENT: adv found
I (5709) NimBLE_CENT: adv found
I (5709) NimBLE_CENT: adv found
I (5709) NimBLE_CENT: adv found
I (5709) NimBLE_CENT: adv found
I (5719) NimBLE_CENT: adv found
I (5719) NimBLE_CENT: adv found
I (5719) NimBLE_CENT: adv found
I (5729) NimBLE_CENT: adv found
I (5729) NimBLE_CENT: adv found
I (5729) NimBLE_CENT: adv found
I (5729) NimBLE_CENT: adv found
I (5739) NimBLE_CENT: adv found
I (5739) NimBLE_CENT: adv found
I (5739) NimBLE_CENT: adv found
I (5749) NimBLE_CENT: adv found
I (5749) NimBLE_CENT: adv found
I (5749) NimBLE_CENT: adv found
I (5749) NimBLE_CENT: adv found
I (5759) NimBLE_CENT: adv found
I (5759) NimBLE_CENT: adv found
I (5759) NimBLE_CENT: adv found
I (5769) NimBLE_CENT: adv found
I (5769) NimBLE_CENT: adv found
I (5769) NimBLE_CENT: adv found
I (5769) NimBLE_CENT: adv found
I (5779) NimBLE_CENT: adv found
I (5779) NimBLE_CENT: adv found
I (5779) NimBLE_CENT: adv found
I (5789) NimBLE_CENT: adv found
I (5789) NimBLE_CENT: adv found
I (5789) NimBLE_CENT: adv found
I (5789) NimBLE_CENT: adv found
I (5799) NimBLE_CENT: adv found
I (5799) NimBLE_CENT: adv found
I (5799) NimBLE_CENT: adv found
I (5809) NimBLE_CENT: adv found
I (5809) NimBLE_CENT: adv found
I (5809) NimBLE_CENT: adv found
I (5809) NimBLE_CENT: adv found
I (5819) NimBLE_CENT: adv found
I (5819) NimBLE_CENT: adv found
I (5819) NimBLE_CENT: adv found
I (5829) NimBLE_CENT: adv found
I (5829) NimBLE_CENT: adv found
I (5829) NimBLE_CENT: adv found
I (5829) NimBLE_CENT: adv found
I (5839) NimBLE_CENT: adv found
I (5839) NimBLE_CENT: adv found
I (5839) NimBLE_CENT: adv found
I (5849) NimBLE_CENT: adv found
I (5849) NimBLE_CENT: adv found
I (5849) NimBLE_CENT: adv found
I (5849) NimBLE_CENT: adv found
I (5859) NimBLE_CENT: adv found
I (5859) NimBLE_CENT: adv found
I (5859) NimBLE_CENT: adv found
I (5869) NimBLE_CENT: adv found
I (5869) NimBLE_CENT: adv found
I (5869) NimBLE_CENT: adv found
I (5869) NimBLE_CENT: adv found
I (5879) NimBLE_CENT: adv found
I (5879) NimBLE_CENT: adv found
I (5879) NimBLE_CENT: adv found
I (5889) NimBLE_CENT: adv found
I (5889) NimBLE_CENT: adv found
I (5889) NimBLE_CENT: adv found
I (5889) NimBLE_CENT: adv found
I (5899) NimBLE_CENT: adv found
I (5899) NimBLE_CENT: adv found
I (5899) NimBLE_CENT: adv found
I (5909) NimBLE_CENT: adv found
I (5909) NimBLE_CENT: adv found
I (5909) NimBLE_CENT: adv found
I (5909) NimBLE_CENT: adv found
I (5919) NimBLE_CENT: adv found
I (5919) NimBLE_CENT: adv found
I (5919) NimBLE_CENT: adv found
I (5929) NimBLE_CENT: adv found
I (5929) NimBLE_CENT: adv found
I (5929) NimBLE_CENT: adv found
I (5929) NimBLE_CENT: adv found
I (5939) NimBLE_CENT: adv found
I (5939) NimBLE_CENT: adv found
I (5939) NimBLE_CENT: adv found
I (5949) NimBLE_CENT: adv found
I (5949) NimBLE_CENT: adv found
I (5949) NimBLE_CENT: adv found
I (5949) NimBLE_CENT: adv found
I (5959) NimBLE_CENT: adv found
I (5959) NimBLE_CENT: adv found
I (5959) NimBLE_CENT: adv found
I (5969) NimBLE_CENT: adv found
I (5969) NimBLE_CENT: adv found
I (5969) NimBLE_CENT: adv found
I (5969) NimBLE_CENT: adv found
I (5979) NimBLE_CENT: adv found
I (5979) NimBLE_CENT: adv found
I (5979) NimBLE_CENT: adv found
E (5979) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5989) NimBLE_CENT: adv found
E (5989) vhci_drv: Rx: failed transport_alloc_evt(1)
E (5999) vhci_drv: Rx: failed transport_alloc_evt(1)
I (5999) NimBLE_CENT: adv found
E (5999) vhci_drv: Rx: failed transport_alloc_evt(1)
E (6009) vhci_drv: Rx: failed transport_alloc_evt(1)
E (6019) vhci_drv: Rx: failed transport_alloc_evt(1)
I (6019) NimBLE_CENT: adv found
E (6019) vhci_drv: Rx: failed transport_alloc_evt(1)
E (6029) vhci_drv: Rx: failed transport_alloc_evt(1)
I (6029) NimBLE_CENT: adv found
E (6029) vhci_drv: Rx: failed transport_alloc_evt(1)
```

mantriyogesh, 2025-03-05:
Hello @volca,

As you would agree, This is not exactly ESP-Hosted-MCU issue but higher level Bluetooth stack.
So we referred to our internal Bluetooth team for this issue. We have got below inputs from them:

We have gone through the post and this is very common & typical observation.  
Here, as confirmed by user also, there are too many BLE devices (noisy environment) in the vicinity.  
 
The issue here is the rate at which adv reports are being received is much higher than the rate at which we are processing the events. Overall, we have below suggestions.
 

> 
> ${\color{yellow}\textsf{We\ have\ gone\ through\ the\ post\ and\ this\ is\ a\ very\ common\ and\ typical\ observation.}}$
> 
> ${\color{yellow}\textsf{Here,\ as\ confirmed\ by\ the\ user,\ there\ are\ too\ many\ BLE\ devices\ (noisy\ environment)\ in\ the\ vicinity.}}$
> 
> ${\color{yellow}\textsf{The\ issue\ here\ is\ the\ rate\ at\ which\ adv\ reports\ are\ being\ received\ is\ much\ higher\ than\ the\ rate\ at\ which\ we\ are\ processing\ the\ events.}}$
> 
> ${\color{yellow}\textsf{Overall,\ we\ have\ below\ suggestions.}}$
> 
> ${\color{yellow}\textsf{1.\ This\ print\ is\ not\ an\ error,\ but\ just\ a\ debug\ print.\ It\ just\ means,\ that\ we\ couldn't\ allocate\ memory\ at\ this\ point\ of\ time\ for\ the\ received\ packet.}}$
> 
> ${\color{yellow}\textsf{Since\ advertisement\ reports\ keep\ coming,\ it\ is\ OK\ to\ drop\ this\ one\ and\ try\ to\ catch\ the\ next\ one.}}$
> 
> ${\color{yellow}\textsf{2.\ Increase\ discard\ count.\ Depending\ upon\ use\ case\ and\ available\ memory,\ this\ needs\ to\ be\ adjusted.\ For\ user,\ we\ can\ suggest\ 300–400.}}$
> 
> ${\color{yellow}\textsf{Though,\ if\ the\ environment\ is\ too\ noisy,\ this\ too\ may\ show\ up\ the\ print.}}$
> 
> ${\color{yellow}\textsf{3.\ We\ modify\ code\ to\ remove\ the\ print,\ though\ we\ don't\ think\ this\ is\ correct,\ since\ we\ tend\ to\ debug\ sometimes\ based\ on\ the\ debug\ prints.}}$
> 
> ${\color{yellow}\textsf{Hence,\ they\ are\ helpful.}}$
> 
> ${\color{yellow}\textsf{4.\ In\ an\ embedded\ system,\ we\ tend\ to\ give\ a\ solution\ which\ works\ for\ the\ majority\ of\ people.}}$
> 
> ${\color{yellow}\textsf{But\ we\ make\ certain\ options\ configurable\ to\ give\ customers\ the\ flexibility\ to\ adjust\ as\ per\ their\ use\ case.}}$
> 
> ${\color{yellow}\textsf{If\ this\ customer\ wants\ to\ capture\ each\ and\ every\ adv\ packet\ in\ a\ noisy\ environment,\ then\ the\ memory\ buffers\ that\ are\ used\ to\ hold\ the\ received\ report\ need\ to\ be\ increased.}}$
> 
> ${\color{yellow}\textsf{To\ ensure\ no\ packet\ is\ missed\ (only\ if\ the\ use\ case\ really\ demands\ so).}}$
> 
> ${\color{yellow}\textsf{Adv\ packets\ are\ continuous\ in\ nature,\ that\ is\ why\ the\ system\ treats\ it\ with\ low\ priority.}}$
> 
> ${\color{orange}\textsf{It's\ OK\ to\ miss\ a\ few advertisements,\ without\ causing\ issue\ to\ the\ functionality.}}$
>

mantriyogesh, 2025-03-05:
Having said that, please feel free to close this issue, if you think, there is nothing else to do from our side !
You can always re-open this issue later, if need be.

volca, 2025-03-05:
Thank you for your patience and assistance. As you mentioned, let's leave this issue to @sign25893

mantriyogesh, 2025-03-05:
Anyway the issue description is just the same and the explaination is the same, along with same remedies.

I will close the issue @sign25893 for now. Let us know if you still encounter any issues ahead.

mantriyogesh, 2025-03-06:
We still stand to fix the trace to lower severity. We can move this trace to warning, instead of 'error'.

closed 2025-03-06 by espressif-bot