espressif/esp-hosted-mcu #33: esp32p4 via spi connect to esp32c5  can't not work when enable esp_psram (IDFGH-14249)

skylin008, 2025-02-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.5-dev-847-gcb3ac7429c

### Espressif SoC revision.

esp32p4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32p4 evb

### Power Supply used.

External 5V

### What is the expected behavior?

I expected to esp32p4 via spi connect to esp32c5  wifi  can  work when  enable esp_psram

### What is the actual behavior?

1、when disable the esp_psram at menuconfig ,esp32p4 via spi connect to esp32c5 wifi  connect function can work;
2、when enable the esp_psram at menuconfig, esp32p4 via spi connect to esp32c5 wifi  connect function can't work;

### Steps to reproduce.

1、power on ;
2、from debug info can show the error information.

### Debug Logs.

```plain
��.5-dev-847-gcb3ac7429c
I (266) efuse_init: Min chE (10588) transport: Not able to connect with ESP-Hosted slave device
I (10588) transport: Reset slave using GPIO[17]
I (10588) os_wrapper_esp: GPIO [17] configured
I (10588) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

assert failed: verify_host_config_for_slave transport_drv.c:493 (0!=0)
Core  0 register dump:
MEPC    : 0x4ff00dda  RA      : 0x4ff07988  SP      : 0x4ff1b700  GP      : 0x4ff12480  
TP      : 0x4ff1b910  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000081  S1      : 0x00000001  A0      : 0x4ff1b73c  A1      : 0x4ff138f1  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff16000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4ff1b856  
S4      : 0x4ff138f0  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

[... 103 lines cut ...]
I (409) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (409) main_task: Started on CPU0
I (419) os_wrapper_esp: GPIO [2] configuring as Interrupt
I (429) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
E (439) gpio: gpio_install_isr_service(502): GPIO isr service already installed
I (449) main_task: Calling app_main()
I (459) wifi station: ESP_WIFI_MODE_STA
I (459) transport: Attempt connection with slave: retry[0]
I (459) transport: Reset slave using GPIO[17]
I (469) os_wrapper_esp: GPIO [17] configured
I (469) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (10579) transport: Not able to connect with ESP-Hosted slave device
I (10579) transport: Reset slave using GPIO[17]
I (10579) os_wrapper_esp: GPIO [17] configured
I (10579) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (21689) transport: Not able to connect with ESP-Hosted slave device
I (21689) transport: Reset slave using GPIO[17]
I (21689) os_wrapper_esp: GPIO [17] configured
I (21689) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (32799) transport: Not able to connect with ESP-Hosted slave device
I (32799) transport: Reset slave using GPIO[17]
I (32799) os_wrapper_esp: GPIO [17] configured
I (32799) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (43909) transport: Not able to connect with ESP-Hosted slave device
```

mantriyogesh, 2025-02-12:
Hello @skylin008 ,

Do you still face this issue?

mantriyogesh, 2025-03-04:
@skylin008 any updates to this?

mantriyogesh, 2025-04-01:
Closing due to inactivity. Please feel free to re-open.

closed 2025-04-01 by mantriyogesh