espressif/esp-hosted-mcu #35: Messages sent to SDIO even when log level set to  ESP_LOG_ERROR

mpolitowicz, 2025-02-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When log level is set to  ESP_LOG_ERROR code execution should not sent info messages to stdio.

### Actual behavior (suspected bug)

`sdio_wrapper.c` calls `sdmmc_card_print_info(stdout, card);` at line 322 and then `sdmmc_io_print_cis_info(temp_buf, cis_data_len, NULL);` at line 69 and `sdmmc_io_print_cis_info(cis_buffer, cis_data_len, NULL); ` at line 73. Since sdio or NULL is passed to functions internally fprintf() outputs information to sdio. 

The code in `sdio_wrapper.c` could be changed to (or similar)
```
	if (esp_log_level_get(TAG) >= ESP_LOG_INFO) {
		// output CIS info from the slave
		sdmmc_card_print_info(stdout, card);

		if (hosted_sdio_print_cis_information(card) != ESP_OK) {
			ESP_LOGW(TAG, "failed to print card info");
		}
	}
```
to suppress information.

### Error logs or terminal output

```shell
Name: 
Type: SDIO
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 0MB
CSD: ver=1, sector_size=0, capacity=0 read_bl_len=0
SCR: sd_spec=0, bus_width=0
TUPLE: DEVICE, size: 3: D9 01 FF 
TUPLE: MANFID, size: 4
  MANF: 0092, CARD: 6666
TUPLE: FUNCID, size: 2: 0C 00 
TUPLE: FUNCE, size: 4: 00 00 02 32 
TUPLE: CONFIG, size: 5: 01 01 00 02 07 
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
```

### Steps to reproduce the behavior

Use esp_hosted with SDIO slave

### Project release version

1.2.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Debian

### Shell

Bash

mantriyogesh, 2025-02-24:
Hello @mpolitowicz , we somehow missed to get notification for this issue.

Anyway, we will fix this issue in main in couple of days (non-urgent)

mpolitowicz, 2025-02-24:
@mantriyogesh no worries at all. We were cleaning logging in our project and it just pop up. So it is more like cosmetics and I already put some temporary solution to suppress it. Anyways on the occasion it would be good to put at INFO level.

closed 2025-03-06 by espressif-bot