espressif/esp-hosted-mcu #36: When running 'esp_wifi_scan_start' with option 'SPIRAM_XIP_FROM_PSRAM'  getting 'FreeRTOS: Fatal error: TLSP deletion callback at index 0 overwritten with non-excutable pointer'

mpolitowicz, 2025-03-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Starting and finishing scan of AP

### Actual behavior (suspected bug)

Om ESP32P4, when trying to invoke 'esp_wifi_scan_start' after setting in options 'SPIRAM_XIP_FROM_PSRAM' getting FreeRTOS error and ESP crashes. 

When 'SPIRAM_XIP_FROM_PSRAM' is off everything works without issues 

### Error logs or terminal output

```shell
I (18606) rpc_req: Scan start Req

E (19565) FreeRTOS: Fatal error: TLSP deletion callback at index 0 overwritten with non-excutable pointer 0x480071a4
--- 0x480071a4: pthread_cleanup_thread_specific_data_callback at /home/poli/esp/v5.4/esp-idf/components/pthread/pthread_local_storage.c:126

abort() was called at PC 0x4ff196a7 on core 0
--- 0x4ff196a7: vPortTLSPointersDelCb at /home/poli/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:717
 (inlined by) vPortTCBPreDeleteHook at /home/poli/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:746
```

### Steps to reproduce the behavior

1. Set 'SPIRAM_XIP_FROM_PSRAM' option 
2. Invoke 'esp_wifi_scan_start'

### Project release version

master

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Debian

### Shell

Bash

### Additional context

The project uses downscalled ESP32P4 dev board: ESP32P4 with ESP32C6 connected via SDIO. In additon we are using LCD screen based on MPI-DSI and LVGL 9.2. 

We were playing with optimization and tried to move code execution to PSRAM. 

Interesting is that when  credentials are set wifi connects and works without issues with flag set so it seems that scan is causing the problem....

mantriyogesh, 2025-03-19:
Hello @mpolitowicz ,

Apologise I missed to notice this issue. Is this still a problem or you found out a way ahead?

mpolitowicz, 2025-03-21:
No worries at all. We will most probably not use 'SPIRAM_XIP_FROM_PSRAM' as this it seems it lowers the performance. 
Any way, this one is little bit strange as all esp-hosted-mcu functionalities seem to work in our case except for esp_wifi_scan_start (and I remember you were adding support for that one recently). So it is not urgent or critical for us, but you may want to take a look at it as it is one of those 'weird mem / thread problems' that can cause someone big trouble (wi discovered it by accident and correlated with SPIRAM_XIP_FROM_PSRAM but it was not obvious at all ).

mantriyogesh, 2025-03-21:
Understood. Thanks for reporting it. We would definitely have a look.

Before moving out of the focus, Would it be possible to share your current :
- esp_hosted_mcu version number or commit?
- idf commit used
- are submodules updated at idf? alternatively, output of `git status` in idf
- sdkconfig at host

I believe slave has yet nothing to do with this problem. still, if you happen to have logs at host and slave, would be helpful to cross check log for deviation from our test logs.

@SohKamYung-Espressif  ++

SohKamYung-Espressif, 2025-03-21:
@mpolitowicz I tested the ESP-IDF `examples/wifi/scan` example, modified to support P4 and ESP-Hosted, and did not encounter the problem you described.

Console output from my P4 dev board:

```
I (25) boot: ESP-IDF v5.4-727-g5cbd2a3877-dirty 2nd stage bootloader
I (26) boot: compile time Mar 21 2025 15:51:35
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v1.0
I (30) boot: efuse block revision: v0.1
I (34) boot.esp32p4: SPI Speed      : 80MHz
I (38) boot.esp32p4: SPI Mode       : DIO
I (42) boot.esp32p4: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=48060020 size=24840h (149568) map
I (115) esp_image: segment 1: paddr=00034868 vaddr=30100000 size=0002ch (    44) load
I (117) esp_image: segment 2: paddr=0003489c vaddr=3010002c size=0003ch (    60) load
I (120) esp_image: segment 3: paddr=000348e0 vaddr=4ff00000 size=0b738h ( 46904) load
I (137) esp_image: segment 4: paddr=00040020 vaddr=48000020 size=5d014h (380948) map
I (202) esp_image: segment 5: paddr=0009d03c vaddr=4ff0b738 size=0719ch ( 29084) load
I (209) esp_image: segment 6: paddr=000a41e0 vaddr=4ff12900 size=03038h ( 12344) load
I (217) boot: Loaded app from partition at offset 0x10000
I (217) boot: Disabling RNG early entropy source...
I (227) hex_psram: vendor id    : 0x0d (AP)
I (227) hex_psram: Latency      : 0x01 (Fixed)
I (227) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (228) hex_psram: dev id       : 0x03 (generation 4)
I (233) hex_psram: density      : 0x07 (256 Mbit)
I (237) hex_psram: good-die     : 0x06 (Pass)
I (241) hex_psram: SRF          : 0x02 (Slow Refresh)
I (246) hex_psram: BurstType    : 0x00 ( Wrap)
I (250) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (255) hex_psram: BitMode      : 0x01 (X16 Mode)
I (259) hex_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (264) hex_psram: DriveStrength: 0x00 (1/1)
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 20MHz
I (296) mmu_psram: .rodata xip on psram
I (339) mmu_psram: .text xip on psram
I (339) hex_psram: psram CS IO is dedicated
I (340) cpu_start: Multicore app
I (4045) esp_psram: SPI SRAM memory test OK
I (4056) cpu_start: Pro cpu start user code
I (4056) cpu_start: cpu freq: 360000000 Hz
I (4056) app_init: Application information:
I (4056) app_init: Project name:     scan
I (4060) app_init: App version:      v5.4-727-g5cbd2a3877-dirty
I (4066) app_init: Compile time:     Mar 21 2025 15:51:28
I (4071) app_init: ELF file SHA256:  29140011c...
I (4075) app_init: ESP-IDF:          v5.4-727-g5cbd2a3877-dirty
I (4081) efuse_init: Min chip rev:     v0.1
I (4085) efuse_init: Max chip rev:     v1.99
I (4089) efuse_init: Chip rev:         v1.0
I (4093) heap_init: Initializing. RAM available for dynamic allocation:
I (4099) heap_init: At 4FF17FA0 len 00023020 (140 KiB): RAM
I (4104) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (4110) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (4115) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (4120) heap_init: At 30100068 len 00001F98 (7 KiB): TCM
I (4126) esp_psram: Adding pool of 32000K of PSRAM memory to heap allocator
I (4133) spi_flash: detected chip: generic
I (4136) spi_flash: flash io: dio
W (4139) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (4151) host_init: ESP Hosted : Host chip_ip[18]
[... 84 lines cut ...]
I (9171) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (9171) scan: Channel          10
I (9171) scan: SSID             ESP_Guest
I (9181) scan: RSSI             -46
I (9181) scan: Authmode         WIFI_AUTH_WPA2_PSK
I (9181) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (9191) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (9191) scan: Channel          11
[...]
``` 

Attached is my sdkconfig for reference:

[sdkconfig.txt](https://github.com/user-attachments/files/19384424/sdkconfig.txt)

`CONFIG_SPIRAM_XIP_FROM_PSRAM=y` is set.

For reference, here is my ESP-IDF version

```bash
> idf.py --version
ESP-IDF v5.4-727-g5cbd2a3877-dirty
```

Can you provide the info requested in https://github.com/espressif/esp-hosted-mcu/issues/36#issuecomment-2742560256 so we can take a look at your configuration?

mpolitowicz, 2025-03-21:
Ok. I will set it up and try to gather the information as soon as possible.

mpolitowicz, 2025-03-25:
Ok, I believe I know where is the catch (and sorry for the delay in answer) - it looked like the scan is causing the trouble but I overlooked that prior to this I invoke stop/deintialize for provisioning (apologies for the confusion).

When I took a deeper look at the code it appeared that most probably the crash happens on `wifi_prov_mgr_stop_provisioning` and more precisely on some kind delayed action that happens asynchronously when stop provisioning is called. And that was source of confusion: asynchronous action on provision stopping, as I wait on WIFI_PROV_END event and that wait is completed. The execution moved forward to network scan and just after reaching it (I reported in logs 'rpc_req: Scan start Req') system crashed. So it looked like the scan code is the cause of trouble, but now when I remove scanning it still crashes, and when I remove provisioning it doesn't. 

And still the crash happens only if SPIRAM_XIP_FROM_PSRAM is enabled. 

I also checked slave and in case of crash it does not get any requests after soft ap, while without crash it gets 0x11d, 0x104, 0x11c, 0x118, 0x11e and reports 'Wi-Fi sta scan done'. 

Hope that helps to frame the source of issue or redirect it to provisioning part.

IDF version 5.4.0, esp_hosted version 1.3.4

SohKamYung-Espressif, 2025-03-26:
@mpolitowicz We are still looking this. At the moment, it looks more like an ESP-IDF issue than a ESP-Hosted issue.

We are checking with the ESP-IDF team on this and will get back to you.

johnboiles, 2025-06-28:
I believe this is fixed by https://github.com/espressif/esp-idf/pull/15999