espressif/esp-hosted-mcu #39: When using esp_wifi_remote, the compilation reports an error: undefined reference to "esp_netif_create_default_wifi_ap_collect".

hbs2001, 2025-03-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Successfully built the project and ran it on the development board.

### Actual behavior (suspected bug)

I am currently porting a project from ESP32-S4 to ESP32-P4. Since my code uses `esp_netif`, I added the `esp_wifi_remote` component version 0.8.1 using `idf_component.yml`. However, when I attempted to build, the build failed. Upon checking the logs, I found that the API could not be located.

This issue occurs in both ESP-IDF versions 5.4.0 and 5.3.2.

[build_log.txt](https://github.com/user-attachments/files/19220359/build_log.txt)
[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Introducing esp_wifi_remote to use the esp_netif_create_default_wifi_ap API.

### Project release version

esp_wifi_remote （0.8.1）    esp_hosted (0.0.27)

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

ZSH

mantriyogesh, 2025-03-13:
@SohKamYung-Espressif 
Please update your observations on transferred issue (here)..

mantriyogesh, 2025-03-13:
@SohKamYung-Espressif ,
Also, if this is issue, any workaround can be given to customer, so that they would be unblocked by the time actual fix doesn't come?

SohKamYung-Espressif, 2025-03-13:
@hbs2001 Thank you for the report. A fix is being looked at.

As a work-around, can you change wifi-remote to this:

```yml
espressif/esp_wifi_remote:
  version: "~0.7.2"
```

hbs2001, 2025-03-13:
> [@hbs2001](https://github.com/hbs2001) Thank you for the report. A fix is being looked at.感谢您的报告。正在考虑修复方法。
> 
> As a work-around, can you change wifi-remote to this:作为解决方法，您能否将 wifi-remote 更改为：
> 
> espressif/esp_wifi_remote:
>   version: "~0.7.2"

OK, I'll try it

mantriyogesh, 2025-03-20:
Hello @hbs2001 ,  after @david-cermak fixes added to wifi-remote, the build issues should be fixed now.

Can you use
```
espressif/esp_wifi_remote:
  version: "*"
```

And 
```
idf.py fullclean
rm -rf sdkconfig build 
idf.py set-target <host_name>
idf.py -p <host_port> flash monitor
```

host name could be like `esp32p4` etc.

closed 2025-04-02 by hbs2001