espressif/esp-hosted-mcu #41: Hosted can be support BCM4334

skylin008, 2025-04-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Thank you shared this project.ESP32C5 wifi download speed only 20Mbps, it can not be suit ours' project.This hoted can be support BCM4334 WIFI chip?

### Use cases

Thank you shared this project.ESP32C5 wifi download speed only 20Mbps, it can not be suit ours' project.This hoted can be support BCM4334 WIFI chip?

mantriyogesh, 2025-04-01:
The throughput depend on the communication bus i.e. hosted transport used in between host and slave. If you need higher throughput, consider using SDIO. The ESP32-C5 is going to support SDIO in mass production in couple of months. The experience of SDIO for ESP32-C5 would be much similar to that of **ESP32-C6** SDIO.

You can refer to the throughput table [here](https://github.com/espressif/esp-hosted-mcu?tab=readme-ov-file#6-decide-the-communication-bus-in-between-host-and-slave), if it suits your needs.

For example, one entry in table for SDIO:

[image]

If you decide to use SDIO, You can proceed to check the hardware design required [here](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md)

Please check the documentation of this MCU only repo at https://github.com/espressif/esp-hosted-mcu for complete details.

skylin008, 2025-04-01:
Thanks @mantriyogesh. With ESP32P4 + ESP32C6 used the espresso esp32p4 eval demo, used iperf testing with tcp/ip protocol , Rx data rate is only 19 Mpbs, it used SDIO communication with esp32p4. BCM4334 has SDIO  V2.0, up to 50Mhz clock rate.

mantriyogesh, 2025-04-01:
Have you used latest slave version and lastest host?
Can you share their text logs from bootup both sides?

skylin008, 2025-04-01:
[iperf_p4_c6.log](https://github.com/user-attachments/files/19551678/iperf_p4_c6.log)

[iperf_p4_c5_spi_full.log](https://github.com/user-attachments/files/19551680/iperf_p4_c5_spi_full.log)

Attach esp32c5 and esp32c6 logs, please resume what is issue. Thanks!

mantriyogesh, 2025-04-02:
> [iperf_p4_c6.log](https://github.com/user-attachments/files/19551678/iperf_p4_c6.log)

1. You have shared only P4 log. Where iis C6 log?
2. Which ESP-Hosted component version you have used?
3. Same version should be used at slave and host for high throughput compatibility.

> [iperf_p4_c5_spi_full.log](https://github.com/user-attachments/files/19551680/iperf_p4_c5_spi_full.log)

This is Full duplex. You might have already reached max throughput limit. Use QSPI if you wish to bump the throughput ahead.
Once SDIO is available on C5, you can switch to C5, if C6 throughput is sufficient to you.

I want to also know what the exact throughput requirement for you?
If you can spot some light on the use case, would be better to suggest something.

skylin008, 2025-04-02:
The project is mainly applied to in-vehicle MP3 players. An Android host is used to download MP3 files via the WiFi connection mode of ESP32C5/ESP32C6 and store them in the EMMC node side. The node side includes ESP32P4, ESP32C5/ESP32C6 and EMMC. There is still a very important problem now: when multiple C5s (such as 5 node devices) are connected to the Android host, the download rate changes from more than 1MB/s for a single device to 200KB/s. We suspect that it is a problem of the bandwidth of the Android host. We have tested the situation of connecting a PC to the Android host, and the actual bandwidth is more than 10MB/s. How can we solve the above problem?

mantriyogesh, 2025-04-02:
As i understand, Use case: 
[image]

Please correct me if wrong in above image. Is my understanding correct?

I am doubting sta+softap mode at Android Host is experiencing less than optimal performance as output.

Are you using STA+SoftAP mode on ESP32-C6 of Android Host? What is the max throughput even when only one node is connected?

skylin008, 2025-04-02:
Android Host not include ESP32C5 or ESP32C6, attach system figure and esp32c6 log

[image]

[iperf_c6.log](https://github.com/user-attachments/files/19564106/iperf_c6.log)

mantriyogesh, 2025-04-02:
As stated earlier, C5 was running on Full duplex SPI, with 20Mbps, which almost reaching limit . Still short by 3-4Mbps, but still okay.

If you run SDIO or QSPI then only further higher throughput could be seen. QSPI supported with almost all esp chips. SDIO only supported on C6 or ESP32.

slave in 
> [iperf_c6.log](https://github.com/user-attachments/files/19564106/iperf_c6.log)

Slave looks freshly flashed. Did you flash latest version now, or it was same when you ran the iperf test on P4?

You can alternatively send throughput from one P4-C6 to another P4-C6 altogether.
This way the android host would go out of picture, to localise the problem.