espressif/esp-hosted-mcu #44: Setting of reset active high/low does is always low active

TiKoESoPe, 2025-04-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Rebarding "RESET GPIO config" in the menuconfig. 

- I expect the assigned GPIO to be low and pulled high shortly for a reset. 
- When i set it to "Active Low i expect the assigned GPIO to be high and pulled low shortly for a reset.

[image]

### Actual behavior (suspected bug)

Regardless of the setting, the GPIO is always high and shortly pulled to low for the reset. I measured this, since our PCB needs the inversion.

### Error logs or terminal output

```shell
I (1965) transport: Attempt connection with slave: retry[0]
I (1965) transport: Reset slave using GPIO[46]
I (1965) os_wrapper_esp: GPIO [46] configured
I (1965) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3085) sdio_wrapper: SDIO master: Data-Lines: 4-bit Freq(KHz)[40000 KHz]
I (3085) sdio_wrapper: GPIOs: CLK[43] CMD[44] D0[39] D1[40] D2[41] D3[42] Slave_Reset[46]
I (3085) sdio_wrapper: Queues: Tx[20] Rx[20] SDIO-Rx-Mode[1]
I (3085) H_SDIO_DRV: Starting SDIO process rx task
E (3105) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (3105) sdio_wrapper: sdmmc_card_init failed
I (3105) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3105) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3115) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3125) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3125) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3125) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3125) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3125) gpio: GPIO[43]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3135) gpio: GPIO[44]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3135) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3135) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3135) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3135) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3145) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3145) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3145) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3145) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (3145) H_SDIO_DRV: sdio card init failed

FreeRTOS: FreeRTOS Task "sdio_read" should not return, Aborting now!
Core  1 register dump:
MEPC    : 0x4ff00cf8  RA      : 0x4ff0b6c6  SP      : 0x4ff210a0  GP      : 0x4ff15180  
TP      : 0x4ff210d0  T0      : 0x4fc0a9f8  T1      : 0x4000f3d2  T2      : 0xffffffff  
S0/FP   : 0x4ff16000  S1      : 0x00000000  A0      : 0x4ff16134  A1      : 0x40142e7b  
A2      : 0x00000023  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4ff1b000  
A6      : 0x4000e8b2  A7      : 0x0000000a  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x4005f152  T5      : 0x4005d9ea  T6      : 0x4005ff08  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000001
```

### Steps to reproduce the behavior

1. Change the "RESET GPIO config"
2. Measure the assigned GPIO.

### Project release version

v1.4.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

The problem seems to lie in the following file:

https://github.com/espressif/esp-hosted-mcu/blob/main/host/api/include/esp_hosted_config.h

In Line 328:

```
#ifdef CONFIG_ESP_HOSTED_RESET_GPIO_ACTIVE_LOW
  #define H_RESET_ACTIVE_HIGH                        0
#else
  #define H_RESET_ACTIVE_HIGH                        1
#endif

#ifdef H_RESET_ACTIVE_HIGH
  #define H_RESET_VAL_ACTIVE                         H_GPIO_HIGH
  #define H_RESET_VAL_INACTIVE                       H_GPIO_LOW
#else
  #define H_RESET_VAL_ACTIVE                         H_GPIO_LOW
  #define H_RESET_VAL_INACTIVE                       H_GPIO_HIGH
#endif
```

Based on the sdkconfig value of CONFIG_ESP_HOSTED_RESET_GPIO_ACTIVE_LOW you define H_RESET_ACTIVE_HIGH as 1 or 0. But afterwards you check it with #ifdef and not with #if, so it always uses the values from the if and never from the else.

SohKamYung-Espressif, 2025-04-17:
@TiKoESoPe Thank you for reporting this bug and providing a fix.

mantriyogesh, 2025-04-24:
@TiKoESoPe We believe it is not very useful feature and plan to remove entirely. Do you still find it useful?

TiKoESoPe, 2025-04-24:
@mantriyogesh We designed a ESP32-P4 board connecting the ESP32-P4 as a host with an ESP32-C6.
To ensure there is no problem in reprogramming the ESP32-C6 using an ESP-PROG, we put a transistor at the output of the P4 to have an open connector on the reset of the C6. This way we don't have a problem when ESP32-P4 is running when we flash the ESP32-C6. Therefore our board only runs with the inversion.

I don't know if it would be good to have an output set to +3V3 and then pulling it to GND. If others also have this kind of inversion, they would need it too.