espressif/esp-hosted-mcu #47: Breaking change commit https://github.com/tasmota/esp-idf/commit/dd9f8bfcbcc14dd141f6b5ab26896b6319c66089 (IDFGH-15246)

Jason2866, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Brakes esp_hosted https://github.com/espressif/esp-hosted-mcu/blob/e6489fbbfe3b35e837b34e97cc34912c59e210a6/host/drivers/rpc/core/rpc_rsp.c#L268
rename from reserved to reserved1

mantriyogesh, 2025-05-07:
Hello @Jason2866 ,

We have this issue open.

There were breaking changes in Wi-Fi header file, due to which the issue is started to pop.
Rest assured, The fix is underway.

@SohKamYung-Espressif ++

closed 2025-05-07 by espressif-bot

mantriyogesh, 2025-05-07:
Still the https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/examples/wifi/iperf/main/idf_component.yml#L14-L15

needs to be manually changed to 
```
  espressif/esp_hosted:
    version: "~2"
```

by the time IDF MR doesn't get pushed to add this change (this might take some time as IDF change takes own sweet time)

mantriyogesh, 2025-05-07:
@Jason2866 Please cross check with above change and let us know if you still face issues.

${\color{yellow} \text{The latest master should build with }}$
```
  espressif/esp_hosted:
    version: "~2"
```

${\color{yellow} \text{However if (1) and (2) satisfied below, you still would need to use}}$

```
  espressif/esp_hosted:
    version: "~1"
```

(1) your idf master branch has version (5,5,0)
(2) commit is before espressif/esp-idf@dd9f8bfc

mantriyogesh, 2025-05-08:
Just to add up further,

${\color{yellow} \text{The latest master should build with }}$ `idf_component.yml` having:
```yaml
  espressif/esp_wifi_remote:
    version: "~0.9.2"
  espressif/esp_hosted:
    version: "~2"
```