espressif/esp-hosted-mcu #49: ESP-HOSTED: assert failed: xTaskCreateStaticPinnedToCore freertos_tasks_c_additions.h:299 (xPortcheckValidStackMem(puxStackBuffer))

sukesh-ak, 2025-05-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

```bash
# Device spec and version
- Device name: GUITION ESP32-P4 + ESP32-C6 with 10.1inch (800x1280) touch display (JC8012P4A1) 
- Device = ESP32-P4 + ESP32-C6
- ESP-IDF version: v5.4

# components
dependencies:
  lvgl/lvgl:
    version: 9.2.*
    public: true
  espressif/esp_hosted: ^2.0.1
  espressif/esp_wifi_remote: ^0.9.2
```
Expecting to use WiFi with LVGL to scan, connect to AP etc...
Device spec and details: https://github.com/sukesh-ak/JC8012P4A1-GUITION-ESP32-P4_ESP32-C6

### Actual behavior (suspected bug)

```bash
LVGL 9 itself works fine. WiFi using ESP-HOSTED itself works fine.
If LVGL is initialized and ESP-HOSTED is part of the solution, the following error happens.
Keep in mind, I am not invoking anything with WiFi yet in the code when the error happens.
```

### Error logs or terminal output

```shell
I (1071) host_init: ESP Hosted : Host chip_ip[18]
I (1076) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:32401624 min-free:32401624 lfb-def:31981568 lfb-8bit:31981568

I (1092) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1100) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1109) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1117) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1126) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1134) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1143) H_API: ** add_esp_wifi_remote_channels **
I (1147) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4800bee4] Rx[0x48019c28]
--- 0x4800bee4: transport_drv_sta_tx at D:/Dev/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:219
0x48019c28: esp_wifi_remote_channel_rx at D:/Dev/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (1155) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4800be2a] Rx[0x48019c28]
--- 0x4800be2a: transport_drv_ap_tx at D:/Dev/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:249
0x48019c28: esp_wifi_remote_channel_rx at D:/Dev/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

assert failed: xTaskCreateStaticPinnedToCore freertos_tasks_c_additions.h:299 (xPortcheckValidStackMem(puxStackBuffer))
Core  0 register dump:
MEPC    : 0x4ff00d1e  RA      : 0x4ff19056  SP      : 0x4ff3c9e0  GP      : 0x4ff23680
--- Stack dump detected
--- 0x4ff00d1e: panic_abort at D:/espressif/v5.4/esp-idf/components/esp_system/panic.c:468
0x4ff19056: flash_read at D:/espressif/v5.4/esp-idf/components/esp_system/port/image_process.c:77

TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000001  S1      : 0x00000050  A0      : 0x4ff3ca28  A1      : 0x4ff25a71
A2      : 0x00000001  A3      : 0x4ff2b000  A4      : 0x00000001  A5      : 0x4ff2b000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x4ff3ca28  S3      : 0x4ff3ca67
S4      : 0x4ff25a70  S5      : 0x4ff3ca28  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00010020  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000
```

### Steps to reproduce the behavior

This is the main function and it has no code to invoke WiFi.
WiFi invoking code is part of the file, looks like issue is happening when ESP-HOSTED + esp_wifi_remote is getting initialized.

```bash
extern "C" void app_main(void)
{
    lvgl_port_cfg_t lvgl_cfg = ESP_LVGL_PORT_INIT_CONFIG();
    // lvgl_cfg.task_affinity = 1;

    bsp_display_cfg_t cfg = {
        .lvgl_port_cfg = lvgl_cfg,
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE,
        .double_buffer = BSP_LCD_DRAW_BUFF_DOUBLE,
        .flags = {
            .buff_dma = true,
            .buff_spiram = true,
            .sw_rotate = true, // software rotate
        }};

    lv_display_t *disp = bsp_display_start_with_config(&cfg);
    bsp_display_backlight_on();

    bsp_display_lock(0);
    bsp_display_rotate(disp, LV_DISPLAY_ROTATION_270);

    lv_demo_widgets();

    bsp_display_unlock();
}
```

### Project release version

esp_hosted:2.0.1, esp_wifi_remote:0.9.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

ZSH

### Additional context

Looks like some buffer is null and triggering the assert.

mantriyogesh, 2025-05-09:
Would it be possible to push small code including idf_component.yml, using which we can reproduce the issue?

sukesh-ak, 2025-05-09:
> Would it be possible to push small code including idf_component.yml, using which we can reproduce the issue?

Please find repo to reproduce the issue. Like I mentioned earlier, no WiFi related code needs to be run since it's something to do with memory usage of ESP-Hosted + ESP_REMOTE_WiFi + LVGL (buffer in PSRAM) during the initialization.

https://github.com/sukesh-ak/Test-P410-Display-IDF-800x1280

sukesh-ak, 2025-05-17:
@mantriyogesh any update on this?

mantriyogesh, 2025-05-18:
@SohKamYung-Espressif , PTAL..

sukesh-ak, 2025-05-24:
> [@SohKamYung-Espressif](https://github.com/SohKamYung-Espressif) , PTAL..

What does `PTAL` mean? Is this going to fixed soon?
I bought a P4 device to build something and now stuck with this issue. Any timelines for the fix?

sukesh-ak, 2025-05-31:
@mantriyogesh  & @SohKamYung-Espressif 
Any update on this? Is there a workaround for this?

mantriyogesh, 2025-06-03:
Sorry, I will get this checked tomorrow, @sukesh-ak .

Can you please push your complete code @sukesh-ak ? It would be a lot easier, to reproduce.

mantriyogesh, 2025-06-29:
Sorry, I had missed this. Let me check in the coming week and come back on this.

mantriyogesh, 2025-07-07:
[changed_Test-P410-Display-IDF-800x1280.zip](https://github.com/user-attachments/files/21094818/changed_Test-P410-Display-IDF-800x1280.zip)

Hello @sukesh-ak ,

I did not observe any crashes. Please find above attached file for logs and changes used.

## What I changed?

[] Enabled the wifi scan task you intended to uncomment and test.
[] Added vTaskDelay(1) in the idf code, just not to watchdog at one place. This is not your code issue, but idf code issue; it was busy looping. Also added some debug logs, before the busyloop and after, so that I get to know where it is currently stuck.You can also remove the patch if you don't like it.
[] Changed the idf component versions to `*` for `esp_hosted` and `esp_wifi_remote`. I was getting a build issue in your default example for \*_weak.c file. Anyway, it's better to use the latest version to build.

## Observations

I had shared the code changes at your example and idf, what i used.
I have not used LVGL as such anytime. if you still feel there is any issues, please let me know.

## Things to note
- Attached sdkconfigs both sides for your reference.
- I used P4-Function-EV-Board, so I re-used the on-board SDIO lines for SPI. sdkconfig should include the pins used anyway.

mantriyogesh, 2025-07-07:
I apologise for the delay in the response. I was busy with some other customer issues in parallel.
Please let me know, if you can try the patches I used, and still reproduce the issues.

I am not doing anything on LVGL, etc, so if you see a crash, I think it would be better to simulate so that I can check it quickly and reproduce and come to RCA.

sukesh-ak, 2025-07-09:
hi @mantriyogesh Its not a crash but an `assert` due to heap corruption/poisoning from your code in `esp_hosted` or `esp_wifi_remote`.

Looking at the warning shown, I would guess someone working on this component already knows something about this issue.
```
I (1170) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4800b2d4] Rx[0x48018cd8]
--- 0x4800b2d4: transport_drv_sta_tx at D:/Dev/Test-P410-Display-IDF-800x1280/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:228
--- 0x48018cd8: esp_wifi_remote_channel_rx at D:/Dev/Test-P410-Display-IDF-800x1280/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (1177) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4800b21a] Rx[0x48018cd8]
--- 0x4800b21a: transport_drv_ap_tx at D:/Dev/Test-P410-Display-IDF-800x1280/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:259
--- 0x48018cd8: esp_wifi_remote_channel_rx at D:/Dev/Test-P410-Display-IDF-800x1280/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
```

If I enable either of these in `menuconfig`, the assert goes away.
```bash
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_POISONING_LIGHT=y
```

I have removed all the comments etc from the code to clean-it up and updated.

```
# Working output looks like this after enabling the above HEAP Poisoning settings.

Build:Jul 10 2024
rst:0x17 (CHIP_USB_UART_RESET),boot:0x20e (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4fc19ec6
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x177c
load:0x4ff2abd0,len:0xf20
load:0x4ff2cbd0,len:0x33fc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (25) boot: compile time Jul  9 2025 14:50:59
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v1.0
I (28) boot: efuse block revision: v0.3
I (32) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32p4: SPI Speed      : 80MHz
I (40) boot.esp32p4: SPI Mode       : QIO
I (44) boot.esp32p4: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00800000
I (81) boot:  3 storage          Unknown data     01 82 00810000 00700000
I (88) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=48080020 size=683f8h (427000) map
I (151) esp_image: segment 1: paddr=00078420 vaddr=30100000 size=0002ch (    44) load
I (153) esp_image: segment 2: paddr=00078454 vaddr=3010002c size=0003ch (    60) load
I (156) esp_image: segment 3: paddr=00078498 vaddr=4ff00000 size=07b80h ( 31616) load
I (168) esp_image: segment 4: paddr=00080020 vaddr=48000020 size=7ebf0h (519152) map
I (235) esp_image: segment 5: paddr=000fec18 vaddr=4ff07b80 size=1b304h (111364) load
I (253) esp_image: segment 6: paddr=00119f24 vaddr=4ff22f00 size=04240h ( 16960) load
I (257) esp_image: segment 7: paddr=0011e16c vaddr=50108080 size=0001ch (    28) load
I (265) boot: Loaded app from partition at offset 0x10000
I (265) boot: Disabling RNG early entropy source...
I (278) hex_psram: vendor id    : 0x0d (AP)
I (279) hex_psram: Latency      : 0x01 (Fixed)
I (279) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (280) hex_psram: dev id       : 0x03 (generation 4)
I (284) hex_psram: density      : 0x07 (256 Mbit)
I (289) hex_psram: good-die     : 0x06 (Pass)
I (293) hex_psram: SRF          : 0x02 (Slow Refresh)
I (298) hex_psram: BurstType    : 0x00 ( Wrap)
I (302) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (306) hex_psram: BitMode      : 0x01 (X16 Mode)
I (311) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
[... 112 lines cut ...]
I (3176) transport:        - HCI over SDIO
I (3179) transport:        - BLE only
I (3182) transport: EVENT: 13
I (3185) transport: ESP board type is : 13

I (3189) transport: Base transport is set-up

I (3194) transport: Slave chip Id[12]
I (3197) hci_stub_drv: Host BT Support: Disabled
I (3201) H_SDIO_DRV: Received INIT event
I (3230) rpc_wrap: --- ESP Event: Slave ESP Init ---
I (4240) P410: Starting WiFi scan...
I (4240) rpc_req: Scan start Req

I (5400) gsl3680: load fw success
I (5401) gsl3680: enter
I (5413) gsl_point_id: gsl_DataInit
I (5474) gsl3680: enter
I (5485) gsl_point_id: gsl_DataInit
I (5486) gsl3680: enter read_ram_fw
I (5517) gsl3680: gsl3680 startup_chip failed read 0xb0 = 5a,5a,5a,5a 
I (5519) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
I (5524) ESP32_P4_EV: Setting LCD backlight: 100%
I (6978) P410: Found 25 access point(s):
```

mantriyogesh, 2025-07-10:
## Channel Registration Logs
```
I (1170) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4800b2d4] Rx[0x48018cd8]
--- 0x4800b2d4: transport_drv_sta_tx
--- 0x48018cd8: esp_wifi_remote_channel_rx

I (1177) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4800b21a] Rx[0x48018cd8]
--- 0x4800b21a: transport_drv_ap_tx
--- 0x48018cd8: esp_wifi_remote_channel_rx
```
These logs are expected and just indicate successful channel registration—key for enabling Wi-Fi remote TX/RX communication.

### Code reference:

- Log source:
https://github.com/espressif/esp-hosted-mcu/blob/0c4aa5f40ce0feebb454c1d671c39329c795b037/host/api/src/esp_hosted_api.c#L61-L87

- esp_hosted_add_channel()
https://github.com/espressif/esp-hosted-mcu/blob/0c4aa5f40ce0feebb454c1d671c39329c795b037/host/drivers/transport/transport_drv.c#L326

## About the patch I shared in zip

I had the project the repo you shared on P4-C6 using the latest components. I made only minimal adjustments to resolve build errors—likely due to esp-hosted and wifi-remote version mismatches. I also shared the working logs from both sides here:

👉 https://github.com/espressif/esp-hosted-mcu/issues/49#issuecomment-3043369724

I did not observe any issues like:
```
gsl3680
```
It seems your actual project includes more layers not present in the shared repo.

### Request for Clarification

Could you please clarify the exact issue mentioned in this comment?
 https://github.com/espressif/esp-hosted-mcu/issues/49#issuecomment-3052300916
Is it memory low condition at p4? or something else?

With the current repo, I couldn’t reproduce any crash or corruption.(check the logs attached in zip) 

Can you please apply the patches I suggested on top of your shared code, remove the `sdkconfig` and `build/` both sides, and reflash both sides, if you can reproduce your stated issue.

Let me know if there’s a specific case I missed. I can have a second look.

sukesh-ak, 2025-07-10:
> It seems your actual project includes more layers not present in the shared repo.

@mantriyogesh 
I just cloned the same repo yesterday to repro the issue. 
I don't have another project since this was part of my basic testing with P4+P6 to get WiFi & LVGL working.
I have already updated the components like you mentioned to the latest.

I will check again about the patches though the above `workaround` of enabling heap poisoning detection did stop the issue.

sukesh-ak, 2025-07-10:
@mantriyogesh 
If you look at my first post here you will see the assert being triggered. 
I think its happening internally somewhere while handling memory for creating task (not in my code).

`assert failed: xTaskCreateStaticPinnedToCore freertos_tasks_c_additions.h:299 (xPortcheckValidStackMem(puxStackBuffer))`

sukesh-ak, 2025-07-10:
@mantriyogesh Your SDK config file shows IDF 6.x maybe that's why you are not seeing the issue.
`Espressif IoT Development Framework (ESP-IDF) 6.0.0 Project Configuration`

sukesh-ak, 2025-07-10:
When `CONFIG_HEAP_POISONING_LIGHT=y` is enabled the problem goes away, simply means the heap corruption is getting masked probably by the guard pages for heap protection or now slower operation due to overhead. 

Anyway the issue should be either Buffer Overflow/Underflow or Use After free. 
I did work on debugging memory dumps for a decade, and we used gflags to enable guard pages on Windows Servers. Not sure how to troubleshoot on microcontrollers but that's something you can look into.

There is a chance this bug is fixed on the `IDF` version you are using since it shows as `6.x` (maybe the main branch?).