espressif/esp-hosted-mcu #50: How is esp_hosted initialised when it is used as a managed component?

readysteadygo2006, 2025-05-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

I have build the getting started example from ESP-IDF, following the same instructions as are provided for building
the iperf example.

I am trying to understand how the init function for esp_hosted gets called (esp_hosted_host_init):
I can see that its run but I can't see how it is called?

I (344) host_init: ESP Hosted : Host chip_ip[0]

A trace is shown below, the line is above is from the function (esp_hosted_host_init):

I (31) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (31) boot: compile time May 22 2025 17:58:10
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v1.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1c8b4h (116916) map
I (130) esp_image: segment 1: paddr=0002c8dc vaddr=3ff80000 size=0001ch (    28) load
I (130) esp_image: segment 2: paddr=0002c900 vaddr=3ffb0000 size=026c8h (  9928) load
I (137) esp_image: segment 3: paddr=0002efd0 vaddr=40080000 size=01048h (  4168) load
I (143) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=2b478h (177272) map
I (209) esp_image: segment 5: paddr=0005b4a0 vaddr=40081048 size=0e608h ( 58888) load
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
I (250) cpu_start: Multicore app
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 160000000 Hz
I (259) app_init: Application information:
I (259) app_init: Project name:     host_hci_uart
I (263) app_init: App version:      478d946-dirty
I (268) app_init: Compile time:     May 22 2025 17:58:04
I (273) app_init: ELF file SHA256:  5c6ee3eac...
I (277) app_init: ESP-IDF:          v5.4-dirty
I (281) efuse_init: Min chip rev:     v0.0
I (285) efuse_init: Max chip rev:     v3.99 
I (289) efuse_init: Chip rev:         v1.0
I (293) heap_init: Initializing. RAM available for dynamic allocation:
I (299) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (304) heap_init: At 3FFB31B0 len 0002CE50 (179 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (315) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (320) heap_init: At 4008F650 len 000109B0 (66 KiB): IRAM
I (327) spi_flash: detected chip: generic
I (329) spi_flash: flash io: dio
W (332) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (344) host_init: ESP Hosted : Host chip_ip[0]
I (349) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
spi_mempool_create free:183484 min-free:183484 lfb-def:180224 lfb-8bit:180224

mantriyogesh, 2025-05-23:
I believe what you are asking is, How init happens?

We wanted to set-up ESP-Hosted-MCU when user wishes to do wifi init automatically, if not already init.

## Use latest components
Please use latest released [esp-wifi-remote](https://components.espressif.com/components/espressif/esp_wifi_remote) and [esp-hosted-mcu](https://components.espressif.com/components/espressif/esp_hosted) components

You can change it at https://github.com/espressif/esp-idf/blob/28ac0243bb65f2d3ad6fc4e90c54365b10d435b9/examples/wifi/iperf/main/idf_component.yml#L10-L17

to till date latest components as:
```
  espressif/esp_wifi_remote:
 --   version: "~0.5.3"
++    version: "~0"
    rules:
      - if: "target in [esp32p4, esp32h2]"
  espressif/esp_hosted:
--    version: "~1"
++    version: "~2"
    rules:
      - if: "target in [esp32p4, esp32h2]"
```
## Auto Init
- Constructor for idf component (in C)
  - https://github.com/espressif/esp-hosted-mcu/blob/22c924eeec99df03ba84430077c6a4585a32944f/host/esp_hosted_host_init.c#L16
- Wi-Fi Init hook
  - https://github.com/espressif/esp-hosted-mcu/blob/22c924eeec99df03ba84430077c6a4585a32944f/host/api/src/esp_hosted_api.c#L193

## Manual Init
You can always call these functions 
```c
ESP_ERROR_CHECK(esp_hosted_init()); /*Allowed before scheduler start */
ESP_ERROR_CHECK(esp_hosted_reconfigure()); /* Only after scheduler started */
```

Directly from your example, to make it manually.

## Configure

- ESP as Host (Easiest)
  - Change in [Kconfig](https://github.com/espressif/esp-hosted-mcu/blob/main/Kconfig) using `idf.py menuconfig`
  - Browse to `(Top) → Component config → Wi-Fi Remote`
  - Browse to `(Top) → Component config → ESP-Hosted config` and change ESP-Hosted config
- Non ESP Host
  - All above ESP config are hooked/mapped to new macros that used in host code : https://github.com/espressif/esp-hosted-mcu/blob/main/host/api/include/esp_hosted_config.h
- Alternative (As Function APIs)
  - You can also use APIs in https://github.com/espressif/esp-hosted-mcu/blob/main/host/api/include/esp_hosted_transport_config.h, **before** calling `esp_hosted_init()`

---

It looks from logs, that the host example is not yet set up ESP-Hosted-MCU connection with slave?

## Set-up ESP-Hosted-MCU in between host and slave
Please use ESP-IDF **release/5.5** example : https://github.com/espressif/esp-idf/tree/release/v5.5/examples/wifi/iperf#running-iperf-on-the-esp32-p4-function-ev-board

Follow the Set-Up [documentation](https://github.com/espressif/esp-hosted-mcu?tab=readme-ov-file#8-detailed-setup) to 
1. Choose which connection bus to use in between ESP and your host
2. Hardware connections (if different from ESP32-P4-Function-EV-Board)
3. Flash co-processor
4. Flash Host

Please attach the logs both sides in textual form for esp and host from bootup.

And let us know if any issues you face after using [IDF release/5.5](https://github.com/espressif/esp-idf/tree/release/v5.5).

readysteadygo2006, 2025-05-23:
Hi, thanks for your prompt response!

To explain, I am intending to port the esp_hosted host to another SOC, so I'm trying to 
understand how the package is initialised on an ESP host so that I'm clear what need to happen
on the SOC which I'm targeting. From your earlier post I understand that two function need to be 
called to initialise esp_hosted:

ESP_ERROR_CHECK(esp_hosted_init()); /*Allowed before scheduler start */
ESP_ERROR_CHECK(esp_hosted_reconfigure()); /* Only after scheduler started */

I can see that in esp_wfi_weak.c the esp_wifi_init function is defined, I assume this 
will get invoked when wifi is started.

But how does esp_hosted_init get called?