espressif/esp-hosted-mcu #51: MCU wire protocol: Use with any host

David-OConnor, 2025-05-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi! I'm looking for assistance or documentation on how to use ESP-Hosted-MCU on an arbitrary MCU. In this case, it's an STM-32, programmed using Rust, but the point of this inquiry is not specific to any host or language. I'm attempting to send a Ping request, and it's failing.

I would like to confirm that this is the frame structure for ESP-Hosted:
**Bytes 0..12**: Payload header
**Bytes 12..18**: Payload
**Bytes 18..18** + payload len: Payload
**Trailing 2 bytes**: CRC (frame; different from the payload header CRC)

Using this, I am sending this message: `[80, 0, 8, 0, 0, 0, 121, 1, 0, 0, 0, 51, 236, 1, 0, 0, 0, 1, 2, 0]`

I get this error from `idf.py monitor`:
```
E (2088870) UART_DRIVER: uart_rx_task: cal_chksum[8055] != exp_chksum[0], drop len[256] offset[2]
mem_dump free:149548 min-free:149472 lfb-dma:114688 lfb-def:114688 lfb-8bit:114688
E (2091370) UART_DRIVER: incoming data too big: 20492
E (2093870) UART_DRIVER: incoming data too big: 60479
E (2093871) UART_DRIVER: uart_rx_task: cal_chksum[416] != exp_chksum[1], drop len[0] offset[30976]
E (2096370) UART_DRIVER: uart_rx_task: cal_chksum[12] != exp_chksum[20480], drop len[0] offset[513]
E (2096371) UART_DRIVER: uart_rx_task: cal_chksum[124] != exp_chksum[60467], drop len[1] offset[0]
mem_dump free:149548 min-free:149472 lfb-dma:114688 lfb-def:114688 lfb-8bit:114688
E (2108870) UART_DRIVER: uart_rx_task: cal_chksum[2095] != exp_chksum[0], drop len[80] offset[8]
mem_dump free:149548 min-free:149472 lfb-dma:114688 lfb-def:114688 lfb-8bit:114688
```

Payload, broken down:
## Payload header: `[80, 0, 8, 0, 0, 0, 121, 1, 0, 0, 0, 51]
**80**: Interface type Priv and 0 Interface  number
**0**: flags
**[8, 0]:** len of RPC header, CRC, and payload (6 + 0 + 2 = 8)
**[0, 0]:** offset
**[121, 1]:** Checksum,
**[0, 0]:** Sequence number]
**51:** Packet type = Event

## RPC header: `[236, 1, 0, 0, 0, 1]`
**236**: Magic
**1**: Version, 
**[0, 0]**: length,
**0:** Module = Ctrl ,
**1:** Command = PingReq

## Frame CRC (Different from Payload CRC):  [2, 0]

How I'm computing CRC; note the description above the function:
```rust
/// Compute CRCs; for by the one embedded in the payload header, and at the end of the message.
/// The `buf` argument for this function must already by set to the correct range.
///
/// Payload header: Covers Entire ESP frame from byte 0 of the 12-byte payload header up to – but not including – the
/// two-byte trailer CRC that terminates the frame.
/// In other words: header + RPC header + payload.
///
/// Frame (trailing) CRC: Starts with the version field of the RPC header
/// (i.e. it skips the constant magic byte 0xEC) and continues up to the last byte of the
/// payload.
pub(crate) fn calc_crc(buf: &[u8]) -> u16 {
    buf.iter().fold(0u16, |acc, &b| acc.wrapping_add(b as u16))
}
```

What looks right here? What looks wrong? Is there documentation for this, so you can use ESP-Hosted without a library? Thank you! (For example, how you program a part from a datasheet's protocol section)

David-OConnor, 2025-06-07:
Stuck here. I guess, the disconnect is, I see esp-hosted-MCU as turning the ESP into a normal slave IC, but normal slave ICs have datasheets with a protocol description.

mantriyogesh, 2025-06-07:
I've been working on a document based on your suggestions, alongside other tasks. I can share the draft if you’d like understand (also review if you wish to).

The idea is to help new users understand the repo—from a high-level view down to the implementation details.

It’s still early. The diagrams are technically correct but need refinement and may need to be split based on different scenarios. The document mainly focuses on the RPC part, but it might feel overwhelming due to the level of detail. It’ll likely need a few review cycles.

That said, it does explain the framing and how RPC applies to different traffic types like ESP_SERIAL_IF.

[system_design_with_rpc_as_focus.md.zip](https://github.com/user-attachments/files/20639838/system_design_with_rpc_as_focus.md.zip)

David-OConnor, 2025-06-13:
Update. Here's my WIP document. I've gotten past the checksum errors, and believe I just need to fix the RPC payload itself; the final few bytes. Open to any ideas here. I've been modding the ESP-Hosted_MCU firmware eto add print statements:

```md
# ESP-Hosted-MCU: Protocol description

This document describes the wire protocol used by [ESP-Hosted-MCU](https://github.com/espressif/esp-hosted-mcu).
It is current as of version 2.0.8.

All byte encodings are little endian.

## Frame structure

### Bytes 0-11: Payload header
The first twelve bytes are a payload header common to all messages.

- **0, bits 0:4**: Interface type: An `8-bit enum` with values like Serial, AP, and HCI.
- **0, bits 4:8**: Interface number. 0 is a good default.
- **1:** flags
- **2, 3:** Payload length: `u16`. The size, in bytes, of everything in the frame following this header.
- **4, 5**: Offset: `u16`. **Always = [12, 0]** (This header's size). Indicates the byte index the payload starts.
- **6, 7:** Checksum, calculated over the entire frame; see below. Initialize it to 0, then compute at the end.
- **8, 9:** `u16`. Sequence number for tracking packets (Useful in debugging)
- **10, bits 0:2:** Throttle command. 0 is a good default.
- **11:** Packet type. Default = 0. For HCI and Private packet types, this takes the value  `0x33` and `0x22` respectively.

### Bytes 12-23: TLV header
The TLV (type, length, value) header further describes _endpoint_ and length data.

- **12:** `u8`. Always `1` to indicate the following bytes specify the Endpoint name.
- **13, 14:** Length of the endpoint name below. `u16.` **Always [6, 0]**.
- **15 - 20** Endpoint name. Hosts always writes."RPCRsp" as ASCII bytes. Slave may write that, or "RPCEvt".
- **21:** `u8.` Always `2` to indicate the following bytes are data.
- **22, 23:** Payload len: `u16`. This is the length, in bytes, of all remaining data after this TLV header. (Bytes 24-)

### Remaining bytes: RPC data
The remaining data is specific to the request or response type, and is structured according to the RPC protocol.
It uses variable-length integers (_varints_), so we will no longer use fixed indices to describe structure. The data is organized
as follows, with no spacing between items. 

The `Tag` type is used several types. It's a `u8`-encoded enum of two values: (field <<3) | (wire_type as u8). 
Field is a `u8`starting at 1. Wire type is an enum as follows:

```rust
 enum WireType {
    /// 0: int32, uint32, bool, enum, variant
    Basic = 0,
    /// 64-bit fixed
    SixtyFourBit = 1,
    /// Len-determined (string, bytes, sub-message)
    LenDetermined = 2,
    /// 32-bit fixed (fixed32, float)
    Fixed32Bit = 5,
}
```

[//]: # ( todo: QC these wire types)

The frame layout, starting at byte 24:
### RPC header len
- **RPC header len tag** `Tag`. **Always `0x08`**. (Field = 1, wire type = 0).
- **RPC header len**: `variant` of the total size of the RPC header. (Generally 5-7 for Esp-Hosted-MCU)

#### RPC header
- **RPC tag**: `Tag` (for Rpc ID). **Always `0x08`**. (Field = 1, wire type = 0).
- **Rpc ID.** A **varint-encoded enum, always len 2** that describes the operation performed. For example, if this is requesting to set a power level,
configure something, data in regards to a request etc. These are currently all 2 bits long for ESP-Hosted-MCU
operations.
- **Payload len tag**: `Tag`. **Always `0x10`**. (Field = 2, wire type = 0).
- **Payload len**: Variant of the payload following this. The payload is in a format determined by Rpd ID.

[... 62 lines cut ...]

[//]: # (**8:**: Tag; always this.)
[//]: # (**5:**: RPC header len = 5, encoded as varint.)
**8:**: Tag; always this.
**183, 2**: RPC ID = 311, encoded as varint. (Part of RPC header)

[//]: # (**16**: Tag for data len; always this.  &#40;Part of RPC header&#41;)
[//]: # (**1**: Data len encoded as varint. &#40;Part of RPC header&#41;)

**16** Tag for data again; always this.
**1** Data len encoded as varint, again
**0** A RpcID-specific payload. In this case, it specifies the interface number.
```

I've tried a few combinations of RPC formats. Any idea what's going wrong? Here's the output from `idf.py monitor`.

```c
E (121870) UART_DRIVER: Scratch buf: 3 0 18 0 12 0 53 3
E (121871) UART_DRIVER: Hdr rcvd len: 18, offset: 12, ift: 3, ifn: 0, Chksum: 821
E (121872) UART_DRIVER: Total len: 30, len: 18, offset: 12
E (121877) slave_ctrl: Data transfer handler. inlen: 6, inbuf: 8 183 2 18 1 0
E (121884) slave_ctrl: Passed calloc Inlne; 6
E (121888) slave_ctrl: Unable to unpack config data
E (121892) protocomm_pserial: Error in handling protocomm request -1
```

mantriyogesh, 2025-06-16:
Sorry, let me cross-check and provide you with the actual RPC frame data in use for your reference.

mantriyogesh, 2025-06-20:
perfect. Nevertheless, manually handling each case is not advisable. It is preferable to serialize the messages using a protobuf decoder. All the messages without manual handling should start working automatically.

David-OConnor, 2025-06-20:
I have confirmed that the latest errors are due to device state (In this case, the Wi-Fi commands seem to all fail with that error if the Init command is not run prior). Closing as complete.