espressif/esp-hosted-mcu #53: rpc_core: Timeout

Big-mouth-c, 2025-05-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I configured a third-party ESP32-P4 and ESP32-C6 using the ESP-Hosted SPI FD (Full Duplex) Operation mode and used the iperf example for testing, but a timeout issue occurred. The ESP32-P4 and ESP32-C6 are two separate development boards connected with short jumper wires. I'm not sure where the problem lies. Do you have any suggestions? (It seems that pull-up resistors are not needed in SPI FD mode.)

ESP32P4
I (28) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (28) boot: compile time May 31 2025 11:03:20
I (29) boot: Multicore bootloader
I (30) boot: chip revision: v1.0
I (32) boot: efuse block revision: v0.3
I (35) boot.esp32p4: SPI Speed      : 40MHz
I (39) boot.esp32p4: SPI Mode       : DIO
I (43) boot.esp32p4: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=40070020 size=286ach (165548) map
I (123) esp_image: segment 1: paddr=000386d4 vaddr=30100000 size=0000ch (    12) load
I (124) esp_image: segment 2: paddr=000386e8 vaddr=3010000c size=00038h (    56) load
I (127) esp_image: segment 3: paddr=00038728 vaddr=4ff00000 size=078f0h ( 30960) load
I (142) esp_image: segment 4: paddr=00040020 vaddr=40000020 size=6c7f4h (444404) map
I (229) esp_image: segment 5: paddr=000ac81c vaddr=4ff078f0 size=0e040h ( 57408) load
I (243) esp_image: segment 6: paddr=000ba864 vaddr=4ff15980 size=0218ch (  8588) load
I (250) boot: Loaded app from partition at offset 0x10000
I (250) boot: Disabling RNG early entropy source...
I (260) cpu_start: Multicore app
I (270) cpu_start: Pro cpu start user code
I (270) cpu_start: cpu freq: 360000000 Hz
I (270) app_init: Application information:
I (271) app_init: Project name:     iperf
I (274) app_init: App version:      1
I (277) app_init: Compile time:     May 31 2025 11:22:29
I (282) app_init: ELF file SHA256:  cd9899512...
I (287) app_init: ESP-IDF:          v5.4.1-dirty
I (291) efuse_init: Min chip rev:     v0.1
I (295) efuse_init: Max chip rev:     v1.99
I (299) efuse_init: Chip rev:         v1.0
I (303) heap_init: Initializing. RAM available for dynamic allocation:
I (309) heap_init: At 4FF1A720 len 000208A0 (130 KiB): RAM
I (314) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (319) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (325) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (330) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (336) spi_flash: detected chip: generic
I (339) spi_flash: flash io: dio
I (342) host_init: ESP Hosted : Host chip_ip[18]
I (346) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
spi_mempool_create free:163516 min-free:163516 lfb-def:122880 lfb-8bit:122880

I (361) spi_wrapper: Transport: SPI, Mode:3 Freq:1MHz TxQ:20 RxQ:20
 GPIOs: MOSI:14 MISO:15 CLK:18 CS:19 HS:16 DR:17 SlaveReset:45
I (374) H_API: ** add_esp_wifi_remote_channels **
I (377) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000edc6] Rx[0x4001cbf0]
--- 0x4000edc6: transport_drv_sta_tx at D:/vscodeidf/iperf4/iperf/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:219
--- 0x4001cbf0: esp_wifi_remote_channel_rx at D:/vscodeidf/iperf4/iperf/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19        

I (385) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000ed0c] Rx[0x4001cbf0]
--- 0x4000ed0c: transport_drv_ap_tx at D:/vscodeidf/iperf4/iperf/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:249
--- 0x4001cbf0: esp_wifi_remote_channel_rx at D:/vscodeidf/iperf4/iperf/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19        

[... 134 lines cut ...]
I (505) SPI_DRIVER: Slave chip Id[12]
I (509) fg_mcu_slave: Initial set up done
I (512) slave_ctrl: event ESPInit
mem_dump free:335640 min-free:335616 lfb-dma:303104 lfb-def:303104 lfb-8bit:303104
--- Error: ClearCommError failed (PermissionError(13, '璁惧涓嶈瘑鍒鍛戒护銆?', None, 22))
--- Waiting for the device to reconnect
I (474) SPI_DRIVER: Using SPI interface
I (474) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (475) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (483) SPI_DRIVER: SPI Ctrl:1 mode: 3, Freq:ConfigAtHost
GPIOs: MOSI: 7, MISO: 2, CS: 10, CLK: 6 HS: 3 DR: 4

I (493) SPI_DRIVER: Hosted SPI queue size: Tx:20 Rx:20
I (498) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (506) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (517) stats:

***** Slave: Raw Throughput testing (Report per 10 sec)*****

I (522) SPI_DRIVER: Slave chip Id[12]
I (525) fg_mcu_slave: Initial set up done
I (529) slave_ctrl: event ESPInit
mem_dump free:335640 min-free:335616 lfb-dma:303104 lfb-def:303104 lfb-8bit:303104

ELF file SHA256: cd9899512

mantriyogesh, 2025-06-03:
> It seems that pull-up resistors are not needed in SPI FD mode

Yes thats right. You can use the jumper wires for Full-Duplex SPI with <=10cm.
In fact, the SPI pins need not be strict to specific GPIOs, as they can be even routed through GPIO matrix as well, on both sides.

Would it be possible to show up the connections you had used in connections: both (1) in camera picture and (2) the hardware connections used.

mantriyogesh, 2025-06-03:
Let me cross-check if same thing works at my place or not and get back to this..

Big-mouth-c, 2025-06-03:
ESP32P4  reste out 53

Big-mouth-c, 2025-06-14:
I found the problem. There is an issue with the output voltage of GPIO14 and GPIO15. The measured voltage is only 1.3V. After switching these two pins to pins 20 and 21, the communication works normally. I still don't know if it's a problem with the board itself or something else. Looking at the development board schematic, GPIO4 - GPIO15 all belong to the VDDPST_1 power domain, and GPIO16 - 23 belong to the VDDPST_2 power domain. The schematic clearly indicates that the IO voltage of VDDPST_2 is 3.3V. And VDDPST_1 is the ESP_LDO_VO4 with adjustable output voltage.