espressif/esp-hosted-mcu #54: Interfacing USB Printer with ESP32-P4 Dev Board

AvikalpF, 2025-05-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Hello Community,
I’m working with the ESP32-P4 Dev Board and would like to interface it with a USB printer.
Could you please guide me on:

1. Whether ESP32-P4 supports USB host mode for standard printers?
2. Any SDK, examples, or documentation available for USB printing?
3. Required drivers or setup for interfacing a USB printer with ESP32-P4?

Thank you!
— Avikalp Fuley

### Use cases

Embedded POS System:
A portable terminal using ESP32-P4 that connects to a USB  printer to print receipts, lists and other data on the go.

mantriyogesh, 2025-06-03:
Hello @AvikalpF ,

The USB driver is not yet supported in ESP-Hosted-MCU, as you have correctly checked.
I would leave it for community.

We would support i, but as if now it is not in the near roadmap. Apologies for that.
Once it is supported, we would keep this thread notified.

mantriyogesh, 2025-06-05:
Oh I see, thanks for the clarification!

You can always check open issues or raise a new one on the ESP-IDF repo here:
👉 https://github.com/espressif/esp-idf/issues

ESP-IDF does provide some USB Host examples. While I’m not very experienced with USB myself, from what I understand, you might find the following helpful:

ESP32-P4 USB Host API documentation:
👉 https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/usb_host.html#c.USB_CLASS_PRINTER

You may need to override the USB class (Printer class is 0x07) and use the right APIs for bulk transfers, etc.
A good starting point could be this example:
👉 ${IDF_PATH}/examples/peripherals/usb/host/msc (for general USB Host setup)

Also, I haven’t tested this myself, but found this GitHub repo which worked on USB printing with the classic ESP32. It might give you some ideas:
👉 https://github.com/touchgadget/esp32-usb-host-demos

Hope this helps at least as a starting point! All the best with your project 😊

closed 2025-07-04 by mantriyogesh