espressif/esp-hosted-mcu #66: ESP-Hosted on Olimex ESP32-P4 as SDIO Host Fails to Initialize ESP32 Auxiliary MCU

marcelthiele, 2025-07-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

<html><head></head><body>
<hr>
<h2>Description</h2>
<p>I’m attempting to use the Olimex ESP32-P4 module as SDIO host (master) and a standard ESP32 as auxiliary MCU (slave) with the official ESP-IDF esp-hosted-mcu firmware. Despite following the pin-out recommendations, the ESP32-P4 never recognizes the ESP32 slave and eventually crashes the SDIO task.

i've flashed the official slave software onto the ESP32 and the host_bluedroid_host_only onto the P4.
</p>
<hr>
<h2>Expected Behavior</h2>
<ul>
<li>
<p>ESP32-P4 (host) should detect and initialize the ESP32 (slave) over SDIO, then proceed to establish the esp-hosted transport channels.</p>
</li>
<li>
<p>No <code>sdio_card_init failed</code> errors; normal <code>H_API: ESP-Hosted starting</code> messages.</p>
</li>
</ul>

<hr>
<h2>Hardware</h2>
<ul>
<li>
<p><strong>Host (Master):</strong> Olimex ESP32-P4</p>
</li>
<li>
<p><strong>Auxiliary (Slave):</strong> ESP32 DevKitC</p>
</li>
<li>
<p><strong>Connections (4-bit SDIO + reset):</strong></p>
</li>
</ul>

Function | Wire Color | ESP32-P4 Pin | ESP32 Pin
-- | -- | -- | --
CLK | Red | 18 | 14
CMD | Bluee | 19 | 15
D0 | Lila | 14 | 2
D1 | Yellow | 15 | 4
D2 | Black | 16 | 12
D3 | Green | 17 | 13
RESET | — | 54 | EN

<p>According to documentation, these mappings are correct:</p>
<hr>
<h2>Software</h2>
<ul>
<li>
<p><strong>ESP-IDF version:</strong> v5.4.1-dirty</p>
</li>
<li>
<p><strong>esp-hosted-mcu firmware:</strong> latest official (esp-hosted-mcu master)</p>
</li>
<li>
<p><strong>Build target:</strong> esp32p4 (master) + esp32 (slave)</p>
</li>
</ul>
<hr>

[... 238 lines cut ...]
<li>
<p>Monitor host UART output.</p>
</li>
</ol>
<hr>

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

M3 Max, Sequoia

### Shell

ZSH

mantriyogesh, 2025-07-04:
As @SohKamYung-Espressif rightly pointed out, please avoid using SDIO with jumper cables due to potential signal integrity issues. This often leads to instability and intermittent failures.

### Key points to consider

ESP32 as SDIO slave requires special hardware handling, as outlined in the Hardware Considerations section above.

Refer to Espressif’s SDIO documentation for mandatory pull-up requirements:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

- Ensure:
  - Correct eFuse configuration (for ESP32 as SDIO slave case)
  - Proper external pull-ups on expected lines
  - 3.3V supply on all SDIO lines (voltage drop is a common issue if hardware is misconfigured)
  - Missing any of these steps can result in unreliable communication.

### Some observation for IDF
Apart from these, I could interlink with one probable issue from idf, https://github.com/espressif/esp-idf/issues/10531#issuecomment-1381042599. You can check if this is your case.

### Workaround?
- I just have suspect, if ESP32 slave taking a little longer time to boot in your case?

Try to increase the https://github.com/espressif/esp-hosted-mcu/blob/0c4aa5f40ce0feebb454c1d671c39329c795b037/host/drivers/transport/transport_drv.c#L82

SohKamYung-Espressif, 2025-07-08:
@marcelthiele Regarding this question in https://github.com/espressif/esp-hosted-mcu/issues/66#issuecomment-3036575558

> I am planning a pcb design with both chips (esp32 and esp32-p4) but now i am a bit worried that i will make mistakes in the layout.

These guidelines on SDIO hardware design on the P4 may help

https://documentation.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32p4/pcb-layout-design-esp32p4.html?title=ESP32-P4%20Hardware%20Design%20Guidelines#sdio

mantriyogesh, 2025-07-08:
ESP_ERR_INVALID_CRC ? May be of help: https://github.com/espressif/esp-idf/issues/7354#issuecomment-891500290

closed 2025-09-15 by marcelthiele

mantriyogesh, 2025-09-15:
I notice the issue is closed. Did it work for you @marcelthiele ?

marcelthiele, 2025-09-15:
We followed the SDIO Hardware design guidelines provided by @SohKamYung-Espressif on the PCB Design - in the hopes that it will work on the PCB directly, and it did!

Thank you guys!