espressif/esp-hosted-mcu #67: ESP32p4+ESP32c3 initialization error problem, please consult

flylink-code, 2025-07-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, the question is as follows:
ESP32-P4 prompts RPC timeout
ESP32-P4 Use SPI full mode
What could be the possible cause of this?
thank！
[image]
ESP32P4 LOG
```
ELF file SHA256: f00277583

Rebooting...
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0xc (SW_CPU_RESET),boot:0x11f (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4ff00d9e
Core1 Saved PC:0x4ff09e64
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x162c
load:0x4ff2abd0,len:0xd6c
load:0x4ff2cbd0,len:0x3308
entry 0x4ff2abda
I (33) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (33) boot: compile time Jul  4 2025 16:46:46
I (34) boot: Multicore bootloader
I (34) boot: chip revision: v1.3
I (37) boot: efuse block revision: v0.3
I (40) boot.esp32p4: SPI Speed      : 80MHz
I (44) boot.esp32p4: SPI Mode       : DIO
I (48) boot.esp32p4: SPI Flash Size : 32MB
I (52) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvsfactory       WiFi data        01 02 00009000 00032000
I (72) boot:  1 nvs              WiFi data        01 02 0003b000 000d2000
I (78) boot:  2 otadata          OTA data         01 00 0010d000 00002000
I (85) boot:  3 phy_init         RF data          01 01 0010f000 00001000
I (91) boot:  4 model            Unknown data     01 82 00110000 000f0000
I (98) boot:  5 ota_0            OTA app          00 10 00200000 00c00000
I (104) boot:  6 ota_1            OTA app          00 11 00e00000 00c00000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00200020 vaddr=40070020 size=2aa3ch (174652) map
I (147) esp_image: segment 1: paddr=0022aa64 vaddr=30100000 size=00044h (    68) load
I (148) esp_image: segment 2: paddr=0022aab0 vaddr=4ff00000 size=05568h ( 21864) load
I (155) esp_image: segment 3: paddr=00230020 vaddr=40000020 size=6984ch (432204) map
I (219) esp_image: segment 4: paddr=00299874 vaddr=4ff05568 size=10830h ( 67632) load
I (231) esp_image: segment 5: paddr=002aa0ac vaddr=4ff15e00 size=02304h (  8964) load
I (238) boot: Loaded app from partition at offset 0x200000
I (238) boot: Disabling RNG early entropy source...
V (249) mmap: after coalescing, 2 regions are left
I (249) cpu_start: Multicore app
D (249) cpu_start: Pro cpu up
D (250) cpu_start: Starting app cpu, entry point is 0x4ff00308
D (235) cpu_start: App cpu up
V CACHE_ERR: access error intr clr & ena mask is: 0x1f
D (262) clk: RTC_SLOW_CLK calibration value: 4315750
V CACHE_ERR: access error intr clr & ena mask is: 0x1f
D (277) cpu_start: calling init function: 0x400001f0 on core: 0
D (277) cpu_start: calling init function: 0x400003b8 on core: 0
I (282) cpu_start: Pro cpu start user code
I (286) cpu_start: cpu freq: 360000000 Hz
D (289) cpu_start: calling init function: 0x40000076 on core: 0
I (295) app_init: Application information:
[... 374 lines cut ...]
I (430) h_bt: - BT/BLE
I (432) h_bt:    - HCI Over SPI
I (435) h_bt:    - BLE only
I (437) fg_mcu_slave: capabilities: 0xe8
I (441) fg_mcu_slave: Supported extended features are:
I (446) h_bt: - BT/BLE (extended)
I (449) fg_mcu_slave: extended capabilities: 0x0
I (456) h_bt: ESP Bluetooth MAC addr: 94:a9:90:49:41:2a
I (458) BLE_INIT: BT controller compile version [d74042a]
I (464) BLE_INIT: Bluetooth MAC: 94:a9:90:49:41:2a
I (469) phy_init: phy_version 1200,2b7123f9,Feb 18 2025,15:22:21
I (506) SPI_DRIVER: Using SPI interface
I (507) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (508) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (516) SPI_DRIVER: SPI Ctrl:1 mode: 3, Freq:ConfigAtHost
GPIOs: CLK:6 MOSI:7 MISO:2 CS:10 HS:3 DR:4

I (525) SPI_DRIVER: Hosted SPI queue size: Tx:20 Rx:20
I (530) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (538) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (548) SPI_DRIVER: Slave chip Id[12]
I (550) fg_mcu_slave: Initial set up done
I (553) slave_ctrl: event ESPInit
mem_dump free:100452 min-free:100376 lfb-dma:40960 lfb-def:40960 lfb-8bit:40960
```

flylink-code, 2025-07-07:
power domain：The corresponding power supply for esp32p4 has been determined to be 3.3v.

SohKamYung-Espressif, 2025-07-08:
Hello @flylink-code 

From your ESP32-P4 log, I see this:

```
I (3275) transport: EVENT: 12
I (3286) transport: EVENT: 11
I (3289) transport: capabilities: 0xe8
I (3292) transport: Features supported are:
I (3296) transport: 	   - HCI over SPI
I (3300) transport: 	   - BLE only
I (3303) transport: EVENT: 13
```
This means that the host received data from the C3 via MISO, and that the CLK, CS, Handshake and Data Ready signals are probably fine.

The timeout sending a RPC command indicates there may be an issue with the MOSI signal.

Please check that the MOSI is operating at 3.3V by probing the pin. As @Big-mouth-c pointed out in https://github.com/espressif/esp-hosted-mcu/issues/53#issuecomment-2972783054 , the issue was the power supply for the SPI pins on the P4 was not at 3.3V.

Also, double-check that the MOSI pin is correctly assigned on both the P4 and C3.

flylink-code, 2025-07-09:
hello @SohKamYung-Espressif
I have noticed an issue. When I opened # define CONFIG_SPIRAM_TRY_LOCATETD_IFI_LWIP 1 configuration, this initialization problem occurred. After turning off this feature, the initialization was normal

[image]
# BUG

[image]

```
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x1 (POWERON),boot:0x11f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x177c
load:0x4ff2abd0,len:0xf20
load:0x4ff2cbd0,len:0x33fc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (26) boot: compile time Jul  9 2025 13:27:49
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v1.3
I (29) boot: efuse block revision: v0.3
I (33) qio_mode: Enabling QIO for flash chip WinBond
I (37) boot.esp32p4: SPI Speed      : 80MHz
I (41) boot.esp32p4: SPI Mode       : QIO
I (45) boot.esp32p4: SPI Flash Size : 32MB
I (49) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvsfactory       WiFi data        01 02 00009000 00032000
I (69) boot:  1 nvs              WiFi data        01 02 0003b000 000d2000
I (75) boot:  2 otadata          OTA data         01 00 0010d000 00002000
I (82) boot:  3 phy_init         RF data          01 01 0010f000 00001000
I (88) boot:  4 model            Unknown data     01 82 00110000 000f0000
I (95) boot:  5 ota_0            OTA app          00 10 00200000 00c00000
I (101) boot:  6 ota_1            OTA app          00 11 00e00000 00c00000
I (109) boot: End of partition table
I (112) boot: No factory image, trying OTA 0
I (115) esp_image: segment 0: paddr=00200020 vaddr=48050020 size=22150h (139600) map
I (145) esp_image: segment 1: paddr=00222178 vaddr=30100000 size=00068h (   104) load
I (147) esp_image: segment 2: paddr=002221e8 vaddr=4ff00000 size=0de30h ( 56880) load
I (161) esp_image: segment 3: paddr=00230020 vaddr=48000020 size=49698h (300696) map
I (208) esp_image: segment 4: paddr=002796c0 vaddr=4ff0de30 size=0a264h ( 41572) load
I (217) esp_image: segment 5: paddr=0028392c vaddr=4ff18100 size=02ed0h ( 11984) load
I (226) boot: Loaded app from partition at offset 0x200000
I (265) boot: Set actual ota_seq=1 in otadata[0]
I (265) boot: Disabling RNG early entropy source...
I (274) hex_psram: vendor id    : 0x0d (AP)
I (274) hex_psram: Latency      : 0x01 (Fixed)
I (275) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (275) hex_psram: dev id       : 0x03 (generation 4)
I (280) hex_psram: density      : 0x07 (256 Mbit)
I (284) hex_psram: good-die     : 0x06 (Pass)
I (288) hex_psram: SRF          : 0x02 (Slow Refresh)
I (293) hex_psram: BurstType    : 0x00 ( Wrap)
I (297) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (302) hex_psram: BitMode      : 0x01 (X16 Mode)
I (306) hex_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (311) hex_psram: DriveStrength: 0x00 (1/1)
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 20MHz
I (338) mmu_psram: .rodata xip on psram
I (366) mmu_psram: .text xip on psram
I (367) hex_psram: psram CS IO is dedicated
I (367) cpu_start: Multicore app
I (4126) esp_psram: SPI SRAM memory test OK
I (4135) cpu_start: Pro cpu start user code
I (4135) cpu_start: cpu freq: 360000000 Hz
I (4136) app_init: Application information:
I (4136) app_init: Project name:     iperf
[... 158 lines cut ...]
I (146) esp_image: segment 2: paddr=002221e8 vaddr=4ff00000 size=0de30h ( 56880) load
I (157) esp_image: segment 3: paddr=00230020 vaddr=48000020 size=49698h (300696) map
I (196) esp_image: segment 4: paddr=002796c0 vaddr=4ff0de30 size=0a264h ( 41572) load
I (203) esp_image: segment 5: paddr=0028392c vaddr=4ff18100 size=02ed0h ( 11984) load
I (211) boot: Loaded app from partition at offset 0x200000
I (211) boot: Disabling RNG early entropy source...
I (221) hex_psram: vendor id    : 0x0d (AP)
I (222) hex_psram: Latency      : 0x01 (Fixed)
I (222) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (223) hex_psram: dev id       : 0x03 (generation 4)
I (227) hex_psram: density      : 0x07 (256 Mbit)
I (232) hex_psram: good-die     : 0x06 (Pass)
I (236) hex_psram: SRF          : 0x02 (Slow Refresh)
I (241) hex_psram: BurstType    : 0x00 ( Wrap)
I (245) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (249) hex_psram: BitMode      : 0x01 (X16 Mode)
I (254) hex_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (259) hex_psram: DriveStrength: 0x00 (1/1)
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 20MHz
I (286) mmu_psram: .rodata xip on psram
I (313) mmu_psram: .text xip on psram
I (314) hex_psram: psram CS IO is dedicated
I (315) cpu_start: Multicore app
```

SohKamYung-Espressif, 2025-07-10:
Regarding https://github.com/espressif/esp-hosted-mcu/issues/67#issuecomment-3051537155

Can you also send the monitor output from the co-processor / slave when you encounter this problem after you enable `CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP`?