espressif/esp-hosted-mcu #69: Reboot cycle: "Stack dump detected" / "panic_abort"

David-OConnor, 2025-07-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

I regularly encounter a scenario where the ESP slave will enter a reset cycle about once a second. This is the output message, from `idf.py monitor`:

See the "Terminal output" section below.

How would you troubleshoot this? Thank you!

### Actual behavior (suspected bug)

See above

### Error logs or terminal output

```shell
mem_dump free:147728 min-free:147676 lfb-dma:114688 lfb-def:114688 lfb-8bit:114688
--- 0x40380946: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32c3/system_internal.c:112

I (1608) fg_mcu_slave: More segment: 0 curr seq: 1122 header seq: 1123

I (1609) slave_ctrl: Resp_MSGId for req[0x122] is [0x222], uid 0
I (1609) slave_ctrl: Received Req [0x122]
E (1613) slave_ctrl: req_wifi_clear_ap_list:1446 failed [esp_wifi_clear_ap_list()] = [12289]
I (1622) slave_ctrl: Resp_MSGId for req[0x11e] is [0x21e], uid 0

assert failed: uart_rx_task uart_slave_api.c:291 (buf)
Core  0 register dump:
MEPC    : 0x4038098e  RA      : 0x4038c6dc  SP      : 0x3fcb7240  GP      : 0x3fc9ac00
--- Stack dump detected
--- 0x4038098e: panic_abort at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/panic.c:468
--- 0x4038c6dc: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/ubsan.c:311

TP      : 0x3fcb73f0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000091  S1      : 0x00000001  A0      : 0x3fcb727c  A1      : 0x3fc9cfd5
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fca2000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcb7386
S4      : 0x3fc9cfd4  S5      : 0x00000483  S6      : 0x00000028  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x40380001: _vector_table at C:/Espressif/frameworks/esp-idf-v5.4.1/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000

--- To exit from IDF monitor please use "Ctrl+]". Alternatively, you can use Ctrl+T Ctrl+X to exit.
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380946
--- 0x40380946: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32c3/system_internal.c:112

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1548
load:0x403cc710,len:0xc8c
load:0x403ce710,len:0x3588
entry 0x403cc71a
```

### Steps to reproduce the behavior

- Enable, and initialize Wi-Fi
- Start a BLE scan

### Project release version

latest

### System architecture

other (details in Additional context)

### Operating system

Windows

### Operating system version

Host: bare metal. Slave: FreeRTOS

### Shell

ZSH

SohKamYung-Espressif, 2025-07-08:
Hello @David-OConnor 

It is not very clear what is happening to the slave that caused the crash. You are using the UART transport?

Is your host a ESP32?

Could you send us a simple test case that can reproduce the issue?

Please also provide the `sdkconfig` for both the host and co-processor.

SohKamYung-Espressif, 2025-07-08:
In the `sdkconfig.txt` I see this:

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
# CONFIG_BT_BLUEDROID_ENABLED is not set
CONFIG_BT_NIMBLE_ENABLED=y
# CONFIG_BT_CONTROLLER_ONLY is not set
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set
```

This does not look correct. By default, for the esp-hosted co-processor, only the BT controller should be enabled and NimBLE should be disabled. Your host STM32 should be running the BT stack.

On the co-processor, run `idf.py menuconfig` ---> `Component config` ---> `Bluetooth` and set `Host` to `Disabled`. Only the `Controller` should be `Enabled`.

A question on the STM32. How is the BT stack sending HCI data? Is it sending it directly to the UART port? If yes, then this will conflict with ESP-Hosted use of the UART port as well.

See the [ESP Hosted Bluetooth design](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md) for more information. The transport mentioned in the document is SPI/SDIO but it also applies to UART as a transport when used by ESP-Hosted.

For NimBLE and Bluedroid BT stacks, a set of ESP-Hosted APIs is available and should be used for HCI data. These APIs encapsulate HCI for transport over ESP-Hosted.

David-OConnor, 2025-07-08:
Thank you regarding the config change; I will try that!

I am sending the HCI over the same UART connection as the Wi-Fi/RPC. (Will switch to SPI on the next revision). Sending HCI commands over UART (Preceded by the 12-byte ESP-Hosted header), and receiving HCI back, as expected. From what I understand: I should use a second set of IO (e.g. a second UART) dedicated to BLE? It sounds like the problem is collision between BLE and Wi-Fi messages on UART?

For the next design, I will use SPI. Should I use two separate SPI busses? Or One with two CS pins? One for Wi-Fi, and the other for BLE? Thank you.

SohKamYung-Espressif, 2025-07-10:
> I am letting ESP-Hosted handle BT HCI, I believe. From the Hosted HCI section of that document, is this what I need to do to prevent the bus collisions?

Yes. If ESP-Hosted is handling BT HCI, there should be no bus collisions.

> In the next revision, I will include both UART and SPI, for flexibility, even if I end up using Hosted HCI with all data over SPI.

Sounds good. 

In checking your original report https://github.com/espressif/esp-hosted-mcu/issues/69#issue-3209873912 , I see this:

`assert failed: uart_rx_task uart_slave_api.c:291 (buf)`

One possibility is that the slave received too many packets and ran out of buffer space.

You can try increasing this to see if it helps. In the `slave` directory, run `menuconfig` ---> `Example configuration` ---> `UART Configuration` and set `Rx Queue Size` to a larger number.