espressif/esp-hosted-mcu #70: Unable to compile slave with BLE host disabled: Mesh error

David-OConnor, 2025-07-08:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Slave firmware compiles

### Actual behavior (suspected bug)

Compile error: `error: 'BLE_MESH_ADV_TASK_CORE' undeclared (first use in this function); did you mean 'BLE_MESH_ADV_TASK_NAME'?` See logs below for details.

### Error logs or terminal output

```shell
C:/Espressif/frameworks/esp-idf-v5.4.1/components/bt/esp_ble_mesh/core/adv.c: In function 'bt_mesh_adv_init':
C:/Espressif/frameworks/esp-idf-v5.4.1/components/bt/esp_ble_mesh/core/adv.c:669:81: error: 'BLE_MESH_ADV_TASK_CORE' undeclared (first use in this function); did you mean 'BLE_MESH_ADV_TASK_NAME'?
  669 |                                       BLE_MESH_ADV_TASK_PRIO, &adv_task.handle, BLE_MESH_ADV_TASK_CORE);
      |                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
      |                                                                                 BLE_MESH_ADV_TASK_NAME
C:/Espressif/frameworks/esp-idf-v5.4.1/components/bt/esp_ble_mesh/core/adv.c:669:81: note: each undeclared identifier is reported only once for each function it appears in
[939/958] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/slave_control.c.obj
```

### Steps to reproduce the behavior

**In menuconfig, disable BLE host.** Note that it's not obvious how to disable Mesh in menuconfig, but I tried setting flags manually in sdkconfig; same error regardless of if I deselected all Mesh items in its menu, or set flags to "n" in sdkconfig directly.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

[sdkconfig.txt](https://github.com/user-attachments/files/21125001/sdkconfig.txt)

SohKamYung-Espressif, 2025-07-09:
By default, the BLE host is disabled on the slave. The `slave/sdkconfig.defaults` already has these settings:

```
CONFIG_BT_ENABLED=y
CONFIG_BT_CONTROLLER_ONLY=y
```
That should enable the BT controller only and disable the BT host.

Can you:

1.  remove `sdkconfig` for the slave
2.  run `menuconfig`, check that `Component config` ---> `Bluetooth` says `Host (Disabled)` and  `Controller (Enabled)`

Now check that the generated `sdkconfig` has these lines:

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
# CONFIG_BT_BLUEDROID_ENABLED is not set
# CONFIG_BT_NIMBLE_ENABLED is not set
CONFIG_BT_CONTROLLER_ONLY=y
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set
```

David-OConnor, 2025-07-09:
You're correct! That works; error on my end. I think what happened is, I had manually selected some Mesh settings at some point. I ended up in a mutually incompatible state. Your solution resolves it, as does going through the `sdkconfig` file, and setting all "mesh" lines to "n".

closed 2025-07-09 by David-OConnor