espressif/esp-hosted-mcu #71: Issues updating the slave from HTTPS and query it's current firmware version

me-no-dev, 2025-07-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

- OTA should be able to update the slave firmware from HTTPS URL (and possibly from local file/byte array)
- `esp_hosted_get_coprocessor_fwversion()` should work after calling `esp_hosted_init()`
- There should be a way to query the Host version and compare against the Slave version
- Possibly there should be a way to query the slave chip model

### Actual behavior (suspected bug)

When attempting OTA over HTTPS the following logs are emitted and update fails:
```
E (8461) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (8461) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (8471) esp-tls: create_ssl_handle failed
E (8474) esp-tls: Failed to open new connection
E (8478) transport_base: Failed to open a new connection
E (8484) HTTP_CLIENT: Connection failed, sock < 0
E (8488) hosted_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (8494) hosted_ota: Check if URL is correct and connectable: https://github.com/.../network_adapter.bin
```

When calling `esp_hosted_get_coprocessor_fwversion()` right after `esp_hosted_init()` the following exception is raised:
```
E (3238) H_SDIO_DRV: tx fail: NULL buff, invalid len (20) or len > max len (1524), transport_up(0))
E (3238) serial: Failed to write data

E (3239) serial_if: Failed to write TX data

CORRUPT HEAP: Bad head at 0x4ff2997c. Expected 0xabba1234 got 0x4ff160d4

assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)
```

There is no way to query the Host version or the Slave chip model at the moment

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. Try OTA with HTTPS URL
2. Call `esp_hosted_get_coprocessor_fwversion()` right after `esp_hosted_init()`

### Project release version

2.0.13

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.5

### Shell

Bash

mantriyogesh, 2025-07-10:
@SohKamYung-Espressif, once free, can you please pick it?

In addition, we need to add some api to get the host version.

in parallel to, `esp_hosted_get_coprocessor_fwversion()`, something like `esp_hosted_get_host_fwversion()`

TD-er, 2025-07-10:
> In addition, we need to add some api to get the host version.
> 
> in parallel to, `esp_hosted_get_coprocessor_fwversion()`, something like `esp_hosted_get_host_fwversion()`

And also the MCU type running as coprocessor, so you know which firmware to use for OTA.
There are now ESP32-P4 boards with a c5 on board for example.

me-no-dev, 2025-07-10:
> Possibly there should be a way to query the slave chip model

@TD-er already in the list

SebastianOtt, 2025-07-14:
I have met this issue in a similar way. We are using an esp32 P4 with and esp32 C6 on a custom board with esp hosted.

First of all, trying to use the `esp_hosted_slave_ota` function will use a `esp_http_client_handle_t` client using a client configuration with HTTP, but not HTTPS. There is also no option to configure this client to use HTTPS. Thus, i had to modify esp-hosted functions to use a different client for HTTPS OTA update retrieval. A certificate is needed, if i am correct.

Current client configuration of hosted:
`	/* Initialize HTTP client configuration */
	esp_http_client_config_t config = {
		.url = image_url,
		.timeout_ms = 5000,
		.event_handler = http_client_event_handler,
	};

        esp_http_client_handle_t client = esp_http_client_init(&config);`

Adapted client configuration:
`	/* Initialize HTTPS client configuration */
	esp_http_client_config_t config = {
		.url = image_url,
		.timeout_ms = 5000,
		.event_handler = http_client_event_handler,
		.transport_type = HTTP_TRANSPORT_OVER_SSL,
		.skip_cert_common_name_check = false,
		.cert_pem = cert_pem,
	};

	esp_http_client_handle_t client = esp_http_client_init(&config);`

This worked in my case to retrieve a ESP32 C6 firmware per HTTPS from a server and should help regarding your first problem.
Nonetheless, chunked OTA failed AFTERWARDS in my case because of SDIO buffer problems if i am correct.

Here are some monitoring prints that indicate successful download, several OTA write responses and the occuring error:

`I (23755) hosted_ota: Starting OTA attempt 1/3
I (23759) hosted_ota: http_open
W (23801) mbedtls: ssl_tls13_generic.c:1552 Perform PSA-based ECDH/FFDH computation.

W (24443) mbedtls: ssl_tls13_client.c:1950 Switch to handshake keys for inbound traffic

W (24571) mbedtls: ssl_tls13_generic.c:403 mbedtls_ssl_tls13_process_certificate_verify() returned 0 (-0x0000)

W (24579) mbedtls: ssl_tls13_client.c:2639 Switch to handshake traffic keys for outbound traffic

W (24582) mbedtls: ssl_tls13_generic.c:1235 Switch to application keys for inbound traffic

W (24587) mbedtls: ssl_tls13_generic.c:1238 Switch to application keys for outbound traffic

I (24595) hosted_ota: HTTP_EVENT_ON_CONNECTED
I (24600) hosted_ota: HTTP_EVENT_HEADER_SENT
I (24603) hosted_ota: http_fetch_headers
I (24983) hosted_ota: HTTP_EVENT_ON_HEADER, key=Server, value=nginx
I (24983) hosted_ota: HTTP_EVENT_ON_HEADER, key=Date, value=Mon, 14 Jul 2025 10:01:57 GMT
I (24985) hosted_ota: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/octet-stream
I (24994) hosted_ota: HTTP_EVENT_ON_HEADER, key=Content-Length, value=862992
I (25000) hosted_ota: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
I (25007) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-Powered-By, value=Express
I (25014) hosted_ota: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
I (25021) hosted_ota: HTTP_EVENT_ON_HEADER, key=Content-Disposition, value=attachment; filename="1.0.1-dev-all-C6-Code.bin"
I (25032) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-Checksum, value=ca62a1642f5159254eb04d03e4d20a6865e1c3570a40002c2000a1f90b4b4f6d
I (25044) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-Frame-Options, value=SAMEORIGIN
I (25051) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-Content-Type-Options, value=nosniff
I (25058) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-XSS-Protection, value=1; mode=block
I (25066) hosted_ota: HTTP_EVENT_ON_HEADER, key=X-Permitted-Cross-Domain-Policies, value=master-only
I (25075) hosted_ota: HTTP_EVENT_ON_HEADER, key=Referrer-Policy, value=same-origin
I (25082) hosted_ota: HTTP_EVENT_ON_HEADER, key=alt-svc, value=h3=":443"; ma=86400
I (25089) hosted_ota: HTTP GET Status = 200, content_length = 862992
W (25095) hosted_ota: ********* Started Slave OTA *******************
I (25102) hosted_ota: *** Please wait for 5 mins to let slave OTA complete ***
I (25108) hosted_ota: Preparing OTA
I (25112) rpc_wrap: Starting OTA
I (25399) rpc_wrap: OTA begin response: 0
I (25399) hosted_ota: Starting OTA transfer
I (25450) rpc_wrap: Writing OTA data
I (25488) rpc_wrap: OTA write response: 0
I (25938) rpc_wrap: Writing OTA data
I (25974) rpc_wrap: OTA write response: 0
....
< a lot of prints....>
...
I (89563) rpc_wrap: Writing OTA data
I (89599) rpc_wrap: OTA write response: 0
I (90049) rpc_wrap: Writing OTA data
I (90085) rpc_wrap: OTA write response: 0
I (90535) rpc_wrap: Writing OTA data
I (90571) rpc_wrap: OTA write response: 0
I (91026) rpc_wrap: Writing OTA data

assert failed: sdio_rx_get_buffer sdio_drv.c:674 (*buf)
Core  0 register dump:
MEPC    : 0x4ff00d36  RA      : 0x4ff0a220  SP      : 0x4ff28810  GP      : 0x4ff1d600  
TP      : 0x4ff289b0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000090  S1      : 0x00000001  A0      : 0x4ff2884c  A1      : 0x4ff1f565  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff4c000  
A6      : 0x00000008  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4ff28957  
S4      : 0x4ff1f564  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000
`

I used the C6 firmware which is around 840Kb (network_adapter.bin) in size and i am curious if some else had this error trying to use OTA updates or if you will also meet it. The P4 crashes, the C6 does not.

TD-er, 2025-07-14:
> I used the C6 firmware which is around 840Kb (network_adapter.bin) in size and i am curious if some else had this error trying to use OTA updates or if you will also meet it. The P4 crashes, the C6 does not.

Maybe also important to know what the version was that was running on the C6.
I've had a discussion with @me-no-dev about this and he also tried to query the firmware version on a 0.0.6 build of the C6 firmware.
Back then this function was not yet implemented and he noticed crashes (of the P4) also.
I already had updated my C6 to a self built 2.0.13 (N.B. there have been commits since and it is still version 2.0.13 ????)
A friend of mine tried my test build too on his node with 0.0.6 and got random data returned.
After initializing the struct before calling he got '0.0.0' :
```c++
        esp_hosted_coprocessor_fwver_t ver_info{};
        esp_hosted_get_coprocessor_fwversion(&ver_info);
```
As said, he got no crashes, but I was thinking maybe it is literally random as it might be related to uninitialized data being fed to (not yet supported) `esp_hosted_xxx` calls and then interpreted by the P4?

SohKamYung-Espressif, 2025-07-15:
@SebastianOtt 

> My guess is now, that stopping other networking tasks before the update could resolve the issue i have with SDIO traffic/buffers.

In general, stopping networking tasks is recommended before doing the OTA update, since the slave is busy updating its code and will reboot at the end, breaking the network link.

Will take a look to see why this is happening.

> E (28759) H_SDIO_DRV: task still writing Rx data to queue!

SebastianOtt, 2025-07-15:
@me-no-dev Here is the output i received, querying the firmware version right after `esp_hosted_init()`. Here, `esp_hosted_get_coprocessor_fwversion()` is basically the first line of code in` app_main()`, since `esp_hosted_init()` is automatically called before ` app_main()`. 

<details><summary>OUTPUT</summary>
<p>

```c
 I (1563) host_init: ESP Hosted : Host chip_ip[18]
I (1568) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:33744632 min-free:33744632 lfb-def:33030144 lfb-8bit:33030144

I (1584) H_API: ** add_esp_wifi_remote_channels **
I (1588) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000b928] Rx[0x4001abd2]
--- Error: ClearCommError failed (PermissionError(13, 'Zugriff verweigert', None, 5))
--- Waiting for the device to reconnect--- 0x4000b928: transport_drv_sta_tx at C:/Users/info/Desktop/ESP32-P4-Project/P4-C6-Hosted-minimized-custom-example/components/CustomEspHosted/host/drivers/transport/transport_drv.c:216
--- 0x4001abd2: esp_wifi_remote_channel_rx at C:/Users/info/Desktop/ESP32-P4-Project/P4-C6-Hosted-minimized-custom-example/components/CustomWifiRemote/esp_wifi_remote_net.c:19

I (1596) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000b870] Rx[0x4001abd2]
--- 0x4000b870: transport_drv_ap_tx at C:/Users/info/Desktop/ESP32-P4-Project/P4-C6-Hosted-minimized-custom-example/components/CustomEspHosted/host/drivers/transport/transport_drv.c:246
--- 0x4001abd2: esp_wifi_remote_channel_rx at C:/Users/info/Desktop/ESP32-P4-Project/P4-C6-Hosted-minimized-custom-example/components/CustomWifiRemote/esp_wifi_remote_net.c:19

I (1604) main_task: Started on CPU0
I (1606) esp_psram: Reserving pool of 75K of internal memory for DMA/internal allocations
I (1614) main_task: Calling app_main()
E (1618) H_SDIO_DRV: tx fail: NULL buff, invalid len (20) or len > max len (1524), transport_up(0))
E (1627) serial: Failed to write data

E (1630) serial_if: Failed to write TX data

assert failed: tlsf_free tlsf.c:630 (!block_is_free(block) && "block already marked as free")
Core  0 register dump:
MEPC    : 0x4ff00d22  RA      : 0x4ff072a8  SP      : 0x4ff1e940  GP      : 0x4ff10b80
--- 0x4ff00d22: panic_abort at C:/Users/info/esp/master/esp-idf/components/esp_system/panic.c:483
--- 0x4ff072a8: __ubsan_include at C:/Users/info/esp/master/esp-idf/components/esp_system/ubsan.c:311

TP      : 0x4ff1eb80  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000006a  S1      : 0x00000001  A0      : 0x4ff1e97c  A1      : 0x4ff1220d
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff14000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4ff1eaad
S4      : 0x4ff1220c  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4ff1e940: 0xa5a5a5a5 0xa5a5a5a5 0x40052390 0x4ff0e80c 0x4ff11934 0x40052390 0x4ff11918 0x40040f9b
--- 0x4ff0e80c: esp_libc_include_assert_impl at C:/Users/info/esp/master/esp-idf/components/newlib/src/assert.c:96

4ff1e960: 0x4ff11944 0x4ff1e974 0x4ff11948 0x40041534 0x4ff1220c 0x00303336 0xa5a5a5a5 0x65737361
4ff1e980: 0x66207472 0x656c6961 0x74203a64 0x5f66736c 0x65657266 0x736c7420 0x3a632e66 0x20303336
4ff1e9a0: 0x6c622128 0x5f6b636f 0x665f7369 0x28656572 0x636f6c62 0x2620296b 0x62222026 0x6b636f6c
4ff1e9c0: 0x726c6120 0x79646165 0x72616d20 0x2064656b 0x66207361 0x22656572 0x4ff10029 0x4ff0733a
.--- 0x4ff10029: spi_flash_chip_winbond_erase_block at C:/Users/info/esp/master/esp-idf/components/spi_flash/spi_flash_chip_winbond.c:142 (discriminator 1)
--- 0x4ff0733a: prvCopyDataToQueue at C:/Users/info/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2470

4ff1e9e0: 0x00000000 0x00000000 0x4ff148ac 0x4ff0733a 0x00000000 0x00000000 0x4ff148ac 0x4ff0795c
--- 0x4ff0733a: prvCopyDataToQueue at C:/Users/info/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2470
--- 0x4ff0795c: xQueueGenericSend at C:/Users/info/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1057

4ff1ea00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000003 0x00000000 0x0000000a
4ff1ea20: 0x4ff3b4d0 0x4ff1eaac 0x40053d54 0x00000000 0x4ff3b4d0 0x4ff1eabc 0x40053e90 0x4ff0e594
--- 0x4ff0e594: esp_log_impl_unlock at C:/Users/info/esp/master/esp-idf/components/log/src/os/log_lock.c:51

4ff1ea40: 0x4ff3b4d0 0x00000000 0x00000000 0x40000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff1ea60: 0x4ff3b4d8 0x4ff3afd4 0x4ff3b4e0 0x4ff0cc26 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x4ff0cc26: block_next at C:/Users/info/esp/master/esp-idf/components/heap/tlsf/tlsf_block_functions.h:161 (discriminator 1)
---  (inlined by) block_link_next at C:/Users/info/esp/master/esp-idf/components/heap/tlsf/tlsf_block_functions.h:168 (discriminator 1)
---  (inlined by) block_mark_as_free at C:/Users/info/esp/master/esp-idf/components/heap/tlsf/tlsf_block_functions.h:176 (discriminator 1)
[... 26 lines cut ...]
4ff1ebc0: 0x4ff13088 0x4ff1ebb4 0x4ff13080 0x00000002 0x4ff1b66c 0x4ff1b66c 0x4ff1ebb4 0x00000000
[... cut ...]
4ff1ed20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

```
</p>
</details> 

Thus, i can confirm your issue. The only difference may be that i have heap assertions enabled, why i do not see this message:
> ```
>CORRUPT HEAP: Bad head at 0x4ff2997c. Expected 0xabba1234 got 0x4ff160d4
>
>assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)
>```

So, the P4 (Host) crashes after direct query of the C6 firmware version in  `app_main()` and i can confirm your issue.

me-no-dev, 2025-07-16:
> Firmware version query does work too without crashing for 2.0.14

but you also did not query right after hosted init 😉

SebastianOtt, 2025-07-16:
> I can see no crashes updating via http. Updated from the factory 0.0.6 to the self compiled 2.0.14. Firmware version query does work too without crashing for 2.0.14. Sadly did not try to query version for the stock firmware.
> 

I have now updated/flashed with latest master [hosted codebase](https://github.com/espressif/esp-hosted-mcu.git)

- C6 slave firmware version: 2.0.16 and latest hosted version on P4.
- I have modified `esp_hosted_ota` to use HTTPS with a certificate

```
/* Initialize HTTP client configuration */
	esp_http_client_config_t config = {
		.url = image_url,
		.timeout_ms = 5000,
		.event_handler = http_client_event_handler,
		.transport_type = HTTP_TRANSPORT_OVER_SSL, // HTTPS
		.cert_pem = cert_pem                       // Certificate
	};

```
I tried:

- No delay between OTA chunk writing (original `esp_hosted_ota`)
- Delay between OTA chunk writing (10ms, 20, 50,...)
- Reducing OTA chunk size (1400 --> 1024--> 512...)
- Stopping other tasks that rely on network (MQTT, Websockets, HTTP server) before OTA update

Unfortunately, in my case, considering our realistic project, which is not truncated like the repos i shared before, the OTA update does not work and the P4 always crashes, while the C6 does not. Every error/crash report i saw was related to SDIO/transport/buffers, as you can see here: `assert failed: sdio_rx_get_buffer sdio_drv.c:674 (*buf)`

I have added the Outputs of P4 and C6 below as an example:

<details><summary>OUTPUT P4</summary>
<p>

```c
D (78) qio_mode: Raw SPI flash chip id 0xef4019
D (82) qio_mode: Manufacturer ID 0xef chip ID 0x4019
I (87) qio_mode: Enabling QIO for flash chip WinBond
D (92) qio_mode: Initial flash chip status 0x200
D (96) qio_mode: QIO mode already enabled in flash
D (101) qio_mode: Enabling QIO mode...
D (104) boot.esp32p4: magic e9
D (107) boot.esp32p4: segments 03
D (110) boot.esp32p4: spi_mode 03
D (113) boot.esp32p4: spi_speed 0f
D (116) boot.esp32p4: spi_size 05
I (119) boot.esp32p4: SPI Speed      : 80MHz
I (123) boot.esp32p4: SPI Mode       : QIO
I (127) boot.esp32p4: SPI Flash Size : 32MB
D (131) boot: Enabling RTCWDT(9000 ms)
I (134) boot: Enabling RNG early entropy source...
D (139) bootloader_flash: rodata starts from paddr=0x00009000, size=0xc00, will be mapped to vaddr=0x40000000
D (148) boot: mapped partition table 0x9000 at 0x40009000
--- 0x40009000: register_select at C:/Users/info/esp/master/esp-idf/components/esp_driver_uart/src/uart_vfs.c:424
D (154) flash_parts: partition table verified, 8 entries
I (159) boot: Partition Table:
I (161) boot: ## Label            Usage          Type ST Offset   Length
D (168) boot: load partition table entry 0x40009000
--- 0x40009000: register_select at C:/Users/info/esp/master/esp-idf/components/esp_driver_uart/src/uart_vfs.c:424
D (172) boot: type=1 subtype=2
I (175) boot:  0 nvs              WiFi data        01 02 00010000 00040000
D (182) boot: load partition table entry 0x40009020
--- 0x40009020: uart_start_select at C:/Users/info/esp/master/esp-idf/components/esp_driver_uart/src/uart_vfs.c:494
D (186) boot: type=1 subtype=0
I (189) boot:  1 otadata          OTA data         01 00 00050000 00002000
D (196) boot: load partition table entry 0x40009040
--- 0x40009040: uart_start_select at C:/Users/info/esp/master/esp-idf/components/esp_driver_uart/src/uart_vfs.c:500
D (200) boot: type=1 subtype=1
I (203) boot:  2 phy_init         RF data          01 01 00052000 00001000
D (210) boot: load partition table entry 0x40009060
--- 0x40009060: uart_start_select at C:/Users/info/esp/master/esp-idf/components/esp_driver_uart/src/uart_vfs.c:499
[... 865 lines cut ...]
--- 0x4001323c: process_rpc_tx_msg at C:/Users/info/Desktop/ESP32-P4-Project/esp32P4-PsychicHttp/components/CustomEspHosted/host/drivers/rpc/core/rpc_core.c:220
4ff2f190: 0x40128c6c 0x00000000 0x00000000 0x00000000 0x0000015e 0x00000001 0x0000015e 0x4ff3b4c0
4ff2f1b0: 0x00000000 0x00000000 0x4ff1d000 0x4001363a 0x00000000 0x00000000 0x00000000 0x4ff3b314
--- 0x4ff1d000: spi_flash_chip_winbond_erase_block at C:/Users/info/esp/master/esp-idf/components/spi_flash/spi_flash_chip_winbond.c:168
--- 0x4001363a: rpc_tx_thread at C:/Users/info/Desktop/ESP32-P4-Project/esp32P4-PsychicHttp/components/CustomEspHosted/host/drivers/rpc/core/rpc_core.c:526
4ff2f1d0: 0x00000000 0x00000000 0x00000000 0x4ff0b4a2 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x4ff0b4a2: vPortTaskWrapper at C:/Users/info/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:255
4ff2f1f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff2f210: 0x00000000 0x4ff2de40 0x4ff2de40 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5
4ff2f230: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x4ff2eb60 0x00000000 0x4ff20938
4ff2f250: 0x4ff20938 0x4ff2f244 0x4ff20930 0x00000002 0x4ff2bcfc 0x4ff2bcfc 0x4ff2f244 0x00000000
4ff2f270: 0x00000017 0x4ff2de40 0x5f637072 0x00007874 0x00000000 0x00000000 0x4ff2f230 0x00000017
4ff2f290: 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff249a0 0x4ff24a08 0x4ff24a70 0x00000000
4ff2f2b0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4fc0af14 0x00000000 0x00000000
4ff2f2d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f2f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... cut ...]
4ff2f330: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f350: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f390: 0x00000000 0x00001000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff2f3b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
```
</p>
</details>

tream2001, 2025-09-18:
> When calling esp_hosted_get_coprocessor_fwversion() right after esp_hosted_init() the following exception is raised:

While I am not doing anything with OTA, but I ran into this same issue. Like others in this thread, I found that `esp_hosted_get_coprocessor_fwversion()` would succeed if I waited to call the function until after I finished configuring the ESP-Hosted slave's Wi-Fi interface.

However, I found that the issue specifically stopped after calling `esp_wifi_init()`. From there, I found that the issue was specifically fixed by the `esp_hosted_reconfigure()` in `esp_wifi_init()`. `esp_hosted_reconfigure()` is exposed in the ESP-Hosted API through `esp_hosted_slave_reset()`.

TLDR: The following should work.
```c
esp_hosted_coprocessor_fwver_t fw_version;
esp_hosted_init();
esp_hosted_slave_reset();
esp_hosted_get_coprocessor_fwversion(&esp_fw_version);
```

mantriyogesh, 2025-09-21:
@SebastianOtt ,

Thanks for confirming the issue in detail.

> Sidenote: we have a lot of external memory available (32MB SPIRAM). mempool does not return a pointer to free external memory (MALLOC_CAP_SPIRAM not used). Would this not be an option? I guess communication would be slower.

Yes, this is definitely do-able, but we need to potentially create a hybrid allocation strategy - use internal RAM for time-critical/small allocations and fall back to SPIRAM for larger buffers or non-critical data.

This way we use best of both worlds. However, this might need some time to implement.

Would you mind the HTTPS code (for hosted) to add as oull request? We will review and pull to master with credits.

mantriyogesh, 2025-10-15:
hello @SebastianOtt ,

## **TL;DR**
I want to draw your attention to newer easier slave OTA example in 2.6.0, which also includes the HTTPS based slave OTA method. Please also note the migration from older rigid API for Slave OTA.

## **Details**

### New example for slave OTA
Please check out the [Slave OTA example](https://components.espressif.com/components/espressif/esp_hosted//examples/host_performs_slave_ota) added to **ESP-Hosted-MCU**.

The example demonstrates three **Slave OTA methods**:

1. **Partition-based:** Firmware from the Host’s plain partition
2. **File system-based:** Firmware from the Host’s LittleFS partition
3. **Internet-based:** HTTPS OTA

###  **Base Slave OTA APIs**

The OTA process is now modular, allowing developers to implement their own mechanisms (e.g., from a local partition, filesystem, HTTPS, SPIFFS, SD card, etc.), instead of being restricted to HTTP-only updates. The modular APIs include:

1. **`esp_hosted_slave_ota_begin()`** – Initializes the OTA process, setting up buffers and internal resources.
2. **`esp_hosted_slave_ota_write()`** – Sends firmware data in chunks to the Slave, managing transfer and integrity checks.
3. **`esp_hosted_slave_ota_end()`** – Finalizes the OTA process, validates the image, and marks it ready for activation.
4. **`esp_hosted_slave_ota_activate()`** – Activates the newly flashed partition on the Slave and reboots to apply the update.

These APIs are demonstrated in the [Slave OTA example](https://components.espressif.com/components/espressif/esp_hosted//examples/host_performs_slave_ota)

### Why new APIs??
Previously, the OTA logic was contained inside the ESP-Hosted **port layer**, including HTTP client code even when not required. A single strict API, like:

```c
const char *image_url = "http://example.com/network_adapter.bin";
esp_err_t ret = esp_hosted_slave_ota(image_url);
if (ret == ESP_OK) {
    printf("OTA update failed[%d]\n", ret);
}
```

proved **inflexible** because any intermediate step could fail (e.g., connection loss, write failure, checksum mismatch). The older API also **supported only HTTP-based OTA updates**, making the port unnecessarily heavy.

### **OTA Decision Checks**

The example performs OTA based on two version checks:

1. **Host’s current Hosted version** vs. **Slave’s current version**
2. **OTA image header version** vs. **Slave’s current version**

For the second check, the header of the OTA image is read before starting the update ([source](https://github.com/espressif/esp-hosted-mcu/blob/cc57617e2c13d5e172b7683d3a7c65e516090f0c/examples/host_performs_slave_ota/components/ota_littlefs/ota_littlefs.c#L261)).

Additional OTA logic in the example includes:

* Automatically selecting the latest modified file as the OTA candidate
* Provided as a separate component for easy reuse and templating

### **Additional Links**

* [Change log](https://github.com/espressif/esp-hosted-mcu/blob/main/CHANGELOG.md#260)
* [Migration guide for old APIs](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/migration_guide.md#coloryellow-text260---esp-hosted-slave-ota)

SebastianOtt, 2025-10-16:
Hey, @mantriyogesh the new API and OTA logic seems great and more flexible. 
However, i have one concern regarding `establish_wifi_connection();` inside the OTA update, which is called without checking whether wifi (station) is already active. I have some previous experience with starting/switching wifi configurations, also in combination with ethernet. If i am not mistaken, a double initialization of `esp_netif_init();` by `wifi_init_sta` can cause some problems/crashes. Correct me if i am wrong.

Regardless, i am pleased with the helpful changes 👍

me-no-dev, 2025-10-16:
@mantriyogesh now everything looks great, the only missing thing is to be able to get the slave model. We need to support C6/C5 etc and be able to distinguish which one is attached

mantriyogesh, 2025-10-16:
@me-no-dev ,

I did not understand your question.

Which is acting slave is primary KConfig switch https://github.com/espressif/esp-hosted-mcu/blob/main/Kconfig#L10-L25

It is a build time switch. Meaning, you cannot just attach P4 configured with C5 slave and just re-use the P4 firmware to connect C6 as slave.

So, there is no runtime check, you can just fetch the compile flag in use to see which is the slave.

ESP-Hosted however, on reeiving slave init/bootup message, verifies if the 'declared' slave target is the 'real' slave target or not, here:
https://github.com/espressif/esp-hosted-mcu/blob/cc57617e2c13d5e172b7683d3a7c65e516090f0c/host/drivers/transport/transport_drv.c#L506-L543

me-no-dev, 2025-10-16:
@mantriyogesh oh this is not good... that means Arduino can ever work with only one type of slave. Isn't it possible one host to work with any similarly configured slave?

lyusupov, 2025-10-16:
@me-no-dev 

Yes. there is a verification procedure by host to check if the host MCU matches to the slave one  as @mantriyogesh explained.
I suppose that this check is excessive - P4 host can work with C5 slave as well as it does with C6 one.

The Arduino Core for ESP32 uses hardcoded SDKCONFIG assignment to use C6 as a slave only.
When the Core is connected to C5 slave - it reports the slave mismatch.

To pass through this issue, it was necessary to build a custom C5 slave firmware that 'mimics' to C6 slave by reporting ESP_PRIV_FIRMWARE_CHIP_ESP32C6 Id .
After that the Arduino Core works with C5 slave just like it does with C6 one.

Yes, this is a hack but there is no way right tow to use C5 slave on a P4 board together with Arduino Core.

I hope that you ( **Espressif** company ) guys will negotiate this issue together soon and will provide a solution that makes everyone happy !

As a temporary solution, the P4 host that built for C6 slave, when it receives C5 slave (or C61 slave) identifier during the handshake - it could admit it as a valid slave device.

mantriyogesh, 2025-10-16:
@SebastianOtt 

> establish_wifi_connection(); inside the OTA update, which is called without checking whether wifi (station) is already active

You are right,  Adding some minor fixes from patch, [134.patch.txt](https://github.com/user-attachments/files/22952795/134.patch.txt)

Will push once reviewed internally.

mantriyogesh, 2025-10-16:
@lyusupov

> To pass through this issue, it was necessary to build a custom C5 slave firmware that 'mimics' to C6 slave by reporting ESP_PRIV_FIRMWARE_CHIP_ESP32C6 Id .
After that the Arduino Core works with C5 slave just like it does with C6 one.

> Yes, this is a hack but there is no way right tow to use C5 slave on a P4 board together with Arduino Core.

Noted.

I am afraid to state that this is not entirely correct way to hack. The reason we ensure expected slave is correct, because the Wi-Fi parameters, structures, the features, and many lower level structures depend upon the slave selected.

So although the things seen as superficially working, you are actually using some of the incompatible Wi-Fi configuration underlying.

these are important things that esp_wifi_remote component facilitates to map the correct slave configurations automatically assigned depending upon slave'chosen' build time.

Well, I am not denying the problem of loosing flexibility as you stated, But we may need to find a better solution.

Simplest example, C5 supports 5G band, C6 doesn't.

mantriyogesh, 2025-10-17:
Thank you everyone, I respect everyone viewpoint here and highly appreciate the inputs and your valuable time invested to make the software better.

mantriyogesh, 2025-11-21:
@joaomarcelo-dev, there are multiple things being discussed in this issue, and it has become a bit complicated.

You can try first with the latest slave and host firmwares or code released by Arduino team and decide if the issue is fixed or not. We had released [2.6.6](https://components.espressif.com/components/espressif/esp_hosted/versions/2.6.6/changelog?language=en) yesterday. @me-no-dev, is this pushed to arduino yet?

joaomarcelo-dev, 2025-11-21:
Hello everyone, thank you for replying to me. I tried setting the pins manually, but I still had no success. The ESP32-P4 starts requesting the data stream from the ESP32-C6. The streaming begins, but in the middle of the process there is still a communication error between the host and the slave. Then the ESP32-P4 ends up rebooting every time… I noticed that there are ready-made APIs for ESP-IDF, but my project is already fully developed in the Arduino IDE.

@me-no-dev 
@mantriyogesh 

```
    void init() {
      if (initialized) return;

      // Pinos SDIO: clk, cmd, d0, d1, d2, d3, reset
      WiFi.setPins(18, 19, 14, 15, 16, 17, -1);
      WiFi.mode(WIFI_STA);
      WiFi.disconnect(false);
      WiFi.setAutoReconnect(true);

      WiFi.onEvent([this](WiFiEvent_t event, WiFiEventInfo_t info) {
        this->onWiFiEvent(event, info);
      });

      WiFi.begin();
      initialized = true;

      WIFI_CLIENT_DEBUG("Inicializado com eventos WiFi configurados");
    }
```

Error:
```

E (29029) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x109
E (29030) H_SDIO_DRV: sdio_read_task: Failed to read data - 265 4398 4398
🧠 [MRI API DEBUG] - 📥 Progresso: 8.9% (196608/2197088 bytes)
🧠 [MRI API DEBUG] - 📥 Progresso: 10.4% (229376/2197088 bytes)
🧠 [MRI API DEBUG] - 📥 Progresso: 11.9% (262144/2197088 bytes)
🧠 [MRI API DEBUG] - 📥 Progresso: 13.4% (294912/2197088 bytes)
E (37726) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x109
E (37726) H_SDIO_DRV: sdio_read_task: Failed to read data - 265 4398 4398
🧠 [MRI API DEBUG] - 📥 Progresso: 14.9% (327680/2197088 bytes)
🧠 [MRI API DEBUG] - 📥 Progresso: 16.4% (360448/2197088 bytes)
E (43704) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x109
E (43704) H_SDIO_DRV: sdio_read_task: Failed to read data - 265 5864 5864

assert failed: sdio_rx_get_buffer sdio_drv.c:830 (*buf)
Core  1 register dump:
MEPC    : 0x4ff0805e  RA      : 0x4ff08014  SP      : 0x4ff31820  GP      : 0x4ff13580  
TP      : 0x4ff31a10  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x4ff31987  A0      : 0x4ff31884  A1      : 0x4ff15761  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff1d000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000090  S3      : 0x4ff31878  
S4      : 0x4ff31878  S5      : 0x4ff31884  S6      : 0x4ff1d000  S7      : 0x000fffff  
S8      : 0x4ff1d000  S9      : 0x00100000  S10     : 0x4ff1d000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000001  

Stack memory:
4ff31820: 0x00000000 0x00000000 0x40224d90 0x4ff10a94 0x00001800 0x00000001 0x00000200 0x4ff15760
4ff31840: 0x4ff2f640 0x1fc2300c 0x0000000c 0x00303338 0x4ff1d000 0x4ff15d88 0x40224d90 0x4ff15d98
4ff31860: 0x40113764 0x4ff15d9c 0x4ff3184c 0x4ff15da0 0x40113938 0x4ff15760 0x00000000 0x00000000
4ff31880: 0x4f000000 0x65737361 0x66207472 0x656c6961 0x73203a64 0x5f6f6964 0x675f7872 0x625f7465
4ff318a0: 0x65666675 0x64732072 0x645f6f69 0x632e7672 0x3033383a 0x622a2820 0x00296675 0x00000000
4ff318c0: 0x4ff1d000 0x00100000 0x4ff1d000 0x000fffff 0x4ff1d000 0x00000000 0x4ff1d8c0 0x4ff0ef90
4ff318e0: 0x000cda0c 0x00000040 0x4ff40000 0x4ff0eefc 0x00000001 0x00000000 0x4ff1d8c0 0x4ff0ef90
4ff31900: 0x00000000 0x000cda00 0x4ff40000 0x4ff0f0d6 0x4ff34a2c 0x4ff1d000 0x4ff1d000 0x00000040
4ff31920: 0x4ff1d000 0x00000003 0x4ff1d8ac 0x4ff01428 0x4ff1d000 0x0000080c 0x000cda00 0x00000040
4ff31940: 0x4ff13c38 0x4ff1d000 0x4ff1d000 0xdf99f39e 0x0000080c 0x4ff1d000 0x4ff1d000 0x000cda00
4ff31960: 0x4ff13c38 0x000cd9c9 0x00000000 0x400a56fc 0x000016e8 0x00000000 0x00000000 0x4ff13580
4ff31980: 0x4ff31a10 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x400a5672 0x00f84899
4ff319a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff319c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff09512
[... 21 lines cut ...]
```

🧠 [MRI API DEBUG] - Código HTTP recebido: 200
🧠 [MRI API DEBUG] - Tamanho reportado: 2197088 bytes
E (13065) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x109
E (13065) H_SDIO_DRV: sdio_read_task: Failed to read data - 265 14660 14660
E (17374) sdmmc_req: sdmmc_host_wait_for_event returned 0x107

assert failed: sdio_rx_get_buffer sdio_drv.c:830 (*buf)
Core  0 register dump:
MEPC    : 0x4ff0805e  RA      : 0x4ff08014  SP      : 0x4ff31820  GP      : 0x4ff13580  
TP      : 0x4ff31a10  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x4ff31987  A0      : 0x4ff31884  A1      : 0x4ff15761  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff1d000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000090  S3      : 0x4ff31878  
S4      : 0x4ff31878  S5      : 0x4ff31884  S6      : 0x4ff1d000  S7      : 0x000fffff  
S8      : 0x4ff1d000  S9      : 0x00100000  S10     : 0x4ff1d000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
4ff31820: 0x00000002 0x00000000 0x40224d90 0

```

joaomarcelo-dev, 2025-11-21:
[image]

[image]

joaomarcelo-dev, 2025-11-21:
My board version is esp32:3.3.3

me-no-dev, 2025-12-03:
@joaomarcelo-dev the reset pin is required to make the host able to reset the C6 before it attempts connection. This is how esp-hosted works.