espressif/esp-hosted-mcu #90: bleprph_host_only_vhci build error on esp32p4

Y1hsiaochunnn, 2025-08-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The build is successful and running fine

### Actual behavior (suspected bug)

here https://github.com/espressif/esp-hosted/tree/main/examples/bleprph_host_only_vhci
Both the esp-idf release 5.5 and the master release fail

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

idf.py set-target esp32p4

idf.py build flash monitor

### Project release version

v5.5.0 and master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows 11

### Shell

PowerShell

Y1hsiaochunnn, 2025-08-15:
I think this can definitely be done, but I'm not sure there's a convenient program available, such as a rich example of ESP-ADF

mantriyogesh, 2025-08-15:
@Y1hsiaochunnn ,

This functionality is **already supported** through the existing ESP-Hosted configuration:

### Configuration Details
The [`host_bluedroid_host_only` example](https://components.espressif.com/components/espressif/esp_hosted/examples/host_bluedroid_host_only) includes the necessary configuration in [sdkconfig.defaults](https://github.com/espressif/esp-hosted-mcu/blob/main/examples/host_bluedroid_host_only/sdkconfig.defaults#L5-L10):

**Key Configuration:**
- **Disable ESP32-S3 Bluetooth controller**: [CONFIG_BT_CONTROLLER_ENABLED=n](https://github.com/espressif/esp-hosted-mcu/blob/ce2f16f069ba4ecde848f365aabe40edf6c32525/examples/host_bluedroid_host_only/sdkconfig.defaults#L6)
- **Enable Bluedroid stack only**: [CONFIG_BT_BLUEDROID_ENABLED=y](https://github.com/espressif/esp-hosted-mcu/blob/ce2f16f069ba4ecde848f365aabe40edf6c32525/examples/host_bluedroid_host_only/sdkconfig.defaults#L7)

### How It Works
1. **ESP32-S3 (Host)**: Runs only the Bluetooth stack (Bluedroid) without local controller
2. **ESP32 (Slave)**: Provides the Bluetooth controller with Classic BT support  
3. **Communication**: ESP32-S3 uses **Hosted HCI** to communicate with ESP32's controller
4. **Result**: ESP32-S3 gains access to Classic Bluetooth functionality through the ESP32

### Usage
You can use the existing `host_bluedroid_host_only` example as your starting point - it's specifically designed for this use case where the host device needs Classic Bluetooth capabilities through a remote controller.

## Reference material
To understand how the `Hosted-HCI` works, please refer [Bluetooth Design Doc](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md#bluetooth-implementation-in-esp-hosted) -> [5. BlueDroid Host Stack ](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md#5-bluedroid-host-stack)

Y1hsiaochunnn, 2025-08-16:
Ok, I'll try it

closed 2025-08-16 by Y1hsiaochunnn