espressif/esp-hosted-mcu #92: Error while trying to set up esp hosted using SDIO

azzanrauf, 2025-08-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I am using an esp32s3 chipset as a host and esp32 as slave. I previously used the http-req example to run over spi using esp-hosted and it was working fine. Now i am doing the same thing but instead of spi full duplex, i am using sdio as my transport layer.
I am using SDIO 1 bit
I have tried on multiple frequencies staring from 400hz
I am not using streaming mode
The exact firmwares i am using:
Host:
http-req example from esp idf framework
Slave:
Slave firmware from esp-hosted mcu

### Actual behavior (suspected bug)

The Sdio communication is failing, i will attach a snapshot of my logs. These "inside function" print statement i have added for debugging, so dont worry about them

### Error logs or terminal output

```shell
I (362) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (370) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (378) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (388) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (393) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (399) H_SDIO_DRV: sdio_data_to_rx_buf_task started
I (399) main_task: Started on CPU0
I (409) main_task: Calling app_main()
I (409) example: host ready, start initializing slave...
I (419) example: Probe using SD 1-bit...
I (419) sdmmc_periph: sdmmc_host_init: SDMMC host already initialized, skipping init flow
I (429) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (439) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (439) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (449) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (459) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (469) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
------insdide function
------insdide function 2
I (509) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
------insdide function 3
------insdide function 4
------insdide function 5
------insdide function 6
-----inside function 7
-----inside function 8
------insdide function 9
Name:
Type: SDIO
Speed: 400.00 kHz (limit: 400.00 kHz)
Size: 0MB
CSD: ver=1, sector_size=0, capacity=0 read_bl_len=0
SCR: sd_spec=0, bus_width=0
inside function sdmmc_io
inside function sdmmc_io
inside function sdmmc_io
TUPLE: DEVICE, size: 3: D9 01 FF
TUPLE: MANFID, size: 4
  MANF: 6666, CARD: 2222
TUPLE: FUNCID, size: 2: 0C 00
TUPLE: FUNCE, size: 4: 00 00 02 32
TUPLE: CONFIG, size: 5: 01 01 00 02 07
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
I (579) example: send reset to slave...
I (1079) example: slave io ready
I (1119) example_connect: Start example_connect.
I (1119) transport: Attempt connection with slave: retry[0]
W (1119) H_SDIO_DRV: Reset slave using GPIO[42]
I (1129) os_wrapper_esp: GPIO [42] configured
I (1129) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2339) sdio_wrapper: SDIO master: Slot 1, Data-Lines: 1-bit Freq(KHz)[400 KHz]
I (2339) sdio_wrapper: GPIOs: CLK[14] CMD[15] D0[2] D1[4] Slave_Reset[42]
I (2339) sdio_wrapper: Queues: Tx[20] Rx[20] SDIO-Rx-Mode[2]
------insdide function
------insdide function 2
------insdide function 3
E (3379) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (3379) sdio_wrapper: sdmmc_card_init failed
I (3379) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3389) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3399) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3409) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3419) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3419) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3429) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3439) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3449) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3459) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3469) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (3479) H_SDIO_DRV: card init failed
I (3479) transport: Waiting for esp_hosted slave to be ready
I (3499) H_SDIO_DRV: SDIO Host operating in PACKET MODE
I (3499) H_SDIO_DRV: Open data path at slave
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

### Steps to reproduce the behavior

- Use httq req firmware with added dependencies of esp-hosted and esp-wifi-remote as host side code- 
- Use slave firmware from esp-hosted mcu as your co-processor side code

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

11

### Shell

CMD

mantriyogesh, 2025-08-11:
@azzanrauf,  Actually, these big wires would not work. Please check the `Hardware Constraints` in the prior post.

azzanrauf, 2025-08-21:
@mantriyogesh Your response is highly appreciated.

However, I previously tested **SDIO communication independently** using the ESP-IDF `sdio` host and slave examples (without ESP-Hosted). With the same hardware configuration (as shown in the pictures above), I was able to send packets reliably at a lower frequency (\~5 MHz).

However, when I switched to **ESP-Hosted** and tried running the `http-req` example over SDIO, I encountered the error shown in my logs.

My question is: **Is there a difference between the SDIO communication used in the ESP-IDF `sdio` example and the one used in ESP-Hosted?**
Since SDIO works fine with the ESP-IDF examples on this hardware, I would expect it to work with ESP-Hosted as well.

mantriyogesh, 2025-08-21:
Have you been using both the slave and host from **ESP-Hosted** only?

If you are working with **SDIO**, please refer to the official guide here:
=> [SDIO Setup Guide](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md)
This provides instructions for flashing both the slave and host. Please note that we **do not recommend** using SDIO with manual wiring or a breadboard. As documented in the guide, a proper PCB layout is **mandatory** for reliable SDIO operation.

As an alternative, you may consider:

* **SDIO 1-bit mode**, or
* **SPI Full-duplex mode** [SPI Full-duplex Guide](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/spi_full_duplex.md)

Both documents cover flashing procedures for slave and host.

Lastly, please keep in mind that without a compatible **slave**, the **host** will not function correctly.

azzanrauf, 2025-08-21:
@mantriyogesh Thanks for your quick responses.

So i have been trying to run SDIO 1 bit mode only.
I have followed the guide [SDIO Setup Guide](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md) and accordingly i am using the slave firmware [Slave](https://github.com/espressif/esp-hosted-mcu/tree/main/slave) directly to flash my co-processor and for the host i am simply using http-request example [http-request](https://github.com/espressif/esp-idf/tree/release/v5.4/examples/protocols/http_request) with the dependencies of esp-hosted and esp-wifi-remote referring to this:

[image]

In the menuconfig:
- i used the transport setting "SDIO" and set the gpio pins accordingly.
- I disabled the streaming mode and set the mode to 1 bit mode.
- The frequency i have tried setting on very low settings even minimum i.e. 400hz

I would also like to add that i was able to run this http-requests example successfully using the spi-full duplex mode

mantriyogesh, 2025-08-22:
Understood. Thanks for confirming that SPI full-duplex is fine.
I still would suggest you to have a look at hardware requirements from https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md#3-hardware-considerations.

In the meantime, can you please attach full textual logs from both sides so that if something obvious can be ruled out..

azzanrauf, 2025-08-22:
@mantriyogesh These are my full textual logs:

Host side logs:

```
I (360) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (368) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (377) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (386) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (391) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (398) H_SDIO_DRV: sdio_data_to_rx_buf_task started
I (398) main_task: Started on CPU0
I (408) main_task: Calling app_main()
I (458) example_connect: Start example_connect.
I (458) transport: Attempt connection with slave: retry[0]
W (458) H_SDIO_DRV: Reset slave using GPIO[42]
I (458) os_wrapper_esp: GPIO [42] configured
I (468) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1668) sdio_wrapper: SDIO master: Slot 1, Data-Lines: 1-bit Freq(KHz)[1000 KHz]
I (1668) sdio_wrapper: GPIOs: CLK[19] CMD[47] D0[13] D1[35] Slave_Reset[42]
I (1668) sdio_wrapper: Queues: Tx[20] Rx[20] SDIO-Rx-Mode[2]
------insdide function
------insdide function 2
------insdide function 3
E (2708) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (2708) sdio_wrapper: sdmmc_card_init failed
I (2708) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2718) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2728) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2738) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2738) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2748) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2758) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2768) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (2778) H_SDIO_DRV: card init failed
I (2778) transport: Waiting for esp_hosted slave to be ready
I (2798) H_SDIO_DRV: SDIO Host operating in PACKET MODE
I (2798) H_SDIO_DRV: Open data path at slave
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420226b7  PS      : 0x00060e30  A0      : 0x82020c1c  A1      : 0x3fc9c230
--- 0x420226b7: sdmmc_send_cmd at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/sdmmc/sdmmc_cmd.c:15

A2      : 0x00000000  A3      : 0x3fc9c250  A4      : 0x0000003c  A5      : 0x3fc9c28c
A6      : 0x3fc9c180  A7      : 0x00000003  A8      : 0x80376e05  A9      : 0x3fc9bf20
A10     : 0x3fc9bf40  A11     : 0x00000000  A12     : 0x3fc9c1a4  A13     : 0x00000000
A14     : 0x3c05da44  A15     : 0x3fc9c16c  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000040  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x420226b4:0x3fc9c230 0x42020c19:0x3fc9c250 0x42020e9f:0x3fc9c2b0 0x420192d6:0x3fc9c2f0 0x42016f62:0x3fc9c310 0x42017649:0x3fc9c340 0x4037b4ad:0x3fc9c370
--- 0x420226b4: sdmmc_send_cmd at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/sdmmc/sdmmc_cmd.c:14
--- 0x42020c19: sdmmc_io_rw_direct at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/sdmmc/sdmmc_io.c:269
--- 0x42020e9f: sdmmc_io_write_byte at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/sdmmc/sdmmc_io.c:295
--- 0x420192d6: hosted_sdio_write_reg at C:/Users/hp/Desktop/http-req-spi/http_request/managed_components/espressif__esp_hosted/host/port/esp/freertos/src/sdio_wrapper.c:508
--- 0x42016f62: sdio_generate_slave_intr at C:/Users/hp/Desktop/http-req-spi/http_request/managed_components/espressif__esp_hosted/host/drivers/transport/sdio/sdio_drv.c:303
--- 0x42017649: sdio_read_task at C:/Users/hp/Desktop/http-req-spi/http_request/managed_components/espressif__esp_hosted/host/drivers/transport/sdio/sdio_drv.c:998
--- 0x4037b4ad: vPortTaskWrapper at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 25ef21d51

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a48
--- 0x40375a48: esp_restart_noos at C:/Espressif5.4.1/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160
[... 101 lines cut ...]
rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

mantriyogesh, 2025-08-25:
Sorry for the delayed response. I've analyzed your logs and identified two distinct issues:

**Issue 1: SDIO Card Initialization Failure (Primary)**
The SDIO card initialization is failing on the first attempt, which appears to be related to hardware and signal integrity problems. This is the root cause that needs to be addressed first.

**Issue 2: Reset Loop Handling (Secondary)**  
As a consequence of issue 1, the host attempts to reset the slave and retry SDIO initialization, but there's a problem in this reset/retry mechanism. I have a fix for this secondary issue: [mcu_MR_108.patch.zip](https://github.com/user-attachments/files/21962615/mcu_MR_108.patch.zip)

**Critical Hardware Check Required:**
> rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371
> ets Jun  8 2016 00:22:57 

This indicates a serious power/signal integrity issue. Please verify:

1. **Power supply voltage** - Must be stable 3.3V at both the slave device and SDIO GPIO pins (CLK, CMD, DAT0-DAT3)
2. **Current capacity** - Ensure your power supply can handle peak current during WiFi/Bluetooth operations
3. **Connection quality** - Check for loose connections, proper SDIO pull-up resistors, and minimize cable lengths if using external connections

Once the hardware issues in (1) are resolved, the SDIO initialization should succeed. Please apply the provided patch to address the secondary reset handling issue.