espressif/esp-hosted-mcu #95: Zephyr support

andrewleech, 2025-08-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Is there any known support / port of the esp-hosted "host" api to [zephyr rtos](https://github.com/zephyrproject-rtos/zephyr) ?

I already know and use esp-hosted to provide wifi and bluetooth to host MCU's running micropython and would like to use this same system for host MCU's running zephyr.

I'm not that familar with the zephyr networking implementations though I know it's got some overlay / inspiration from linux, so hoping there might be a known working implementation we could reference. It would be fantastic if this integration could eventually be included in [upstream zephyr wifi](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi) and [bluetooth](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/bluetooth/hci) so everyone can take advantage of this project.

### Use cases

I would like to use esp-hosted running on a esp32-s3 or similar to provide network and bluetooth for host mcu's running zephyr, most commonly nxp mimxrt or stm32 devices in our applications.

### Alternatives

Currently the only officially supported means I have to adding wifi and ble to both micropython and zephyr is based on infineon chipsets but espressif parts are far more readily available on pre-certified modules so I'd like to move to this platform.

mantriyogesh, 2025-08-20:
Copying my comment from #94 

I completely understand the challenges Zephyr developers are facing with this, and I see how valuable this would be for the ESP-hosted environment. I want to assure you that we're committed to this - we've actually been planning to add this support for a while now, just waiting for the right time to dedicate the resources.

For now, you might want to check out the ESP-hosted support that @iabdalkader added in https://github.com/zephyrproject-rtos/zephyr/pull/87685. We haven't tested it ourselves yet, but it should hopefully get you unblocked.

Thanks for your patience on this.

iabdalkader, 2025-12-07:
I just found this thread randomly, I never got a notification for the mention. Anyway, the Zephyr FG driver has been working great for us, with the Portenta C33. I tested all sorts of WiFI libraries and clients with TLS and whatnot. @andrewleech I don't know if you've tested it or not yet, but let me know if you find any issues.

That said, my only concern with it, is that it was designed against fg-0.0.5. There's been some changes to ABI/protocol since then, @mantriyogesh explained that they're backwards compatible, but I pointed out some changes that might not be [here ](https://github.com/zephyrproject-rtos/zephyr/pull/87685#issuecomment-2983045413). So I'm not 100% confident it will work with the latest FG `v1.0.0.0.0`, I never tested it myself, but it seems it [does](https://github.com/micropython/micropython/pull/12937#issuecomment-3620465264), sure hope so.

Anyway, I hope that someone (Arduino, Qcom or ESP) extends the existing Zephyr driver to support this new mcu firmware, if it's compatible enough, instead of adding a new one, via Kconfig options. We could probably also support newer FG firmware releases via Kconfig options, to fix any backward compatibility issues, so that we could keep the C33 working.