espressif/esp-hosted-mcu #97: Wifi provisioning using ble

azzanrauf, 2025-08-27:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I have the esp32-p4-module dev kit. I am curious if i can do wifi provisioning using ble normally using esp-hosted on my p4. What i am trying to achieve is to setup wifi provisioning manager component on my p4 and when the provisioning manager runs, i can provide wifi credentials from my phone app and it uses esp-hosted support to communicate with the c6 and connect to wifi.
The devkit module has onboard c6 as slave jbtw.

azzanrauf, 2025-08-29:
```
idf_build_get_property(target IDF_TARGET)

if(${target} STREQUAL "linux")
    return() # This component is not supported by the POSIX/Linux simulator
endif()

# if( NOT CONFIG_ESP_WIFI_ENABLED
#     AND NOT CONFIG_ESP_HOST_WIFI_ENABLED
#     AND NOT CMAKE_BUILD_EARLY_EXPANSION )
#     # This component provides only "esp_provisioning" headers if WiFi not enabled
#     # (implementation supported optionally in a managed component esp_wifi_remote)
#     idf_component_register(INCLUDE_DIRS include)
#     return()
# endif()

set(srcs "src/wifi_config.c"
        "src/wifi_scan.c"
        "src/wifi_ctrl.c"
        "src/manager.c"
        "src/handlers.c"
        "src/scheme_console.c"
        "proto-c/wifi_config.pb-c.c"
        "proto-c/wifi_scan.pb-c.c"
        "proto-c/wifi_ctrl.pb-c.c"
        "proto-c/wifi_constants.pb-c.c")

if(CONFIG_ESP_WIFI_SOFTAP_SUPPORT)
    list(APPEND srcs "src/scheme_softap.c")
endif()

if(CONFIG_BT_ENABLED)
    if(CONFIG_BT_BLUEDROID_ENABLED OR CONFIG_BT_NIMBLE_ENABLED)
        list(APPEND srcs
            "src/scheme_ble.c")
    endif()
endif()

idf_component_register(SRCS "${srcs}"
                    INCLUDE_DIRS include
                    PRIV_INCLUDE_DIRS src proto-c
                    REQUIRES lwip protocomm
                    PRIV_REQUIRES protobuf-c bt json esp_timer esp_wifi)
```

- This is the wifi provisioning cmake, i have bypassed this if check:
```
 if( NOT CONFIG_ESP_WIFI_ENABLED
    AND NOT CONFIG_ESP_HOST_WIFI_ENABLED
    AND NOT CMAKE_BUILD_EARLY_EXPANSION )
```
and my ble and softap provisioning is working on p4 devkit module

- Now i am curious if it was disabled deliberately to not provide the soruce files linkage for non-wifi host controllers as p4 in our case and later add support through esp-wifi-remote or the support already exists in the component of esp-wifi-remote

Note: The provisioning manager is working fine, but i am just curious about this system which i have bypassed. Thanks

mantriyogesh, 2025-09-01:
@azzanrauf 

Thank you for raising this question regarding the CMake configuration in the wifi_provisioning component.
I understand you're referring to this specific CMake code:
https://github.com/espressif/esp-idf/blob/2044fba6e71422446986f9ae0909b1ab67e57815/components/wifi_provisioning/CMakeLists.txt#L7-L14

You're correct. When `esp-wifi-remote` and `esp-hosted` components are loaded, they provide the standard Wi-Fi APIs for P4. The RPC mechanism transparently forwards P4 Wi-Fi calls to C6 and returns results as if P4 has native Wi-Fi.

The supported Wi-Fi APIs from ESP-Hosted perspective are listed in: https://github.com/espressif/esp-hosted-mcu/blob/main/host/api/src/esp_wifi_weak.c

Since wifi_provisioning relies on standard Wi-Fi APIs (now provided via esp-hosted), there should be no compatibility issues with the (your) modified CMake configuration.

Although, I will check if this block, https://github.com/espressif/esp-idf/blob/2044fba6e71422446986f9ae0909b1ab67e57815/components/wifi_provisioning/CMakeLists.txt#L7-L14 can be removed altogether.

azzanrauf, 2025-09-02:
@mantriyogesh Thankyou for your response.

I just have another query regarding ble. I have tested wifi provisioning using soft ap and ble both
For soft-ap, it is understood that the native wifi calls are translated to wifi remote calls
However, for ble, i checked the configurations and i seem to come across the ble transport using uart
But i just assumed off the shelf that the provisioning would work fine using ble aswell and it did, but how is the ble support catered using esp-hosted?

mantriyogesh, 2025-09-02:
So for BLE, it is little different. We support bluetooth using [standard HCI interface](https://software-dl.ti.com/lprf/simplelink_cc2640r2_latest/docs/blestack/ble_user_guide/html/ble-stack-3.x/hci.html). This enables the interface messages from stack like blueZ or NimBLE etc to exist on host. As bluetooth stacks they themselve expose this standard interface, there is no specific API requirement.

On high level, there are two ways to handle bluetooth:
1. Standard HCI
Dedicated UART bus. No multiplexing of other traffic on this dedicated bus.

2. Hosted HCI
Multiplex all hosted traffic types using '[interface type](https://github.com/espressif/esp-hosted-mcu/blob/main/common/esp_hosted_interface.h)' . So the standard HCI messages are wrapped with ESP-Hosted header and [ESP_HCI_IF](https://github.com/espressif/esp-hosted-mcu/blob/51d9ef40a375ec443d80c0b046d8d8ba924e90cd/common/esp_hosted_interface.h#L19) as traffic type. slave removes this header upon receiving, understands that it is ESP_HCI_IF, so treats the payload as bluetooth and ends in controller.

I think, instead of repeating everything here, Bluetooth design is explained in in much simplified manner ([here](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md)). I would request you to go through this and let us know if any questions come to your mind.

azzanrauf, 2025-09-03:
@mantriyogesh This clears out the doubt, Thankyou!

closed 2025-09-20 by mantriyogesh