espressif/esp-hosted-mcu #106: BLE custom gapp service for esp32p4

azzanrauf, 2025-09-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I am trying to send files from my app to my esp32p4. The files are music files and i have read from idf documents that a2dp support is only provided for esp32 because it uses classic bluetooth.
Now i am trying to send the file in chunk format to my p4 using ble. 
Do i need to start with nimble bleprph example or bluedriod. What is supported in this mcu and what will be the optimal approach?

SohKamYung-Espressif, 2025-09-26:
Hello @azzanrauf ,

Just to verify,
- you are trying to build the `bluedroid/ble/gatt_server_service_table` on the ESP32-P4?
- for the co-processor, you will be using a ESP32?

SohKamYung-Espressif, 2025-10-03:
Hello @azzanrauf 

Just a check: did the patch I provided work?