espressif/esp-hosted-mcu #107: idf.py set-target esp32p4 generates double quote inside double quoted set(compile definitions

robogo-pj, 2025-09-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

expect set-target to generate proper string ie. 
set(COMPILE_DEFINITIONS "_GLIBCXX_USE_POSIX_SEMAPHORE;_GLIBCXX_HAVE_POSIX_SEMAPHORE;_GNU_SOURCE;IDF_VER='v5.5.1'")
or
set(COMPILE_DEFINITIONS "_GLIBCXX_USE_POSIX_SEMAPHORE;_GLIBCXX_HAVE_POSIX_SEMAPHORE;_GNU_SOURCE;IDF_VER=v5.5.1")

### Actual behavior (suspected bug)

.. idf.py set-target generates double version inside set(COMPILE_DEFINITIONS string. the second double quote terminates the parser prematurely. It assume there are two compile definitions missing a space between them.
v5.5.1 v does work for esp-webrtc-solution doorbell_local project

this is in build_properties.temp.cmake file:

set(COMPILE_DEFINITIONS "_GLIBCXX_USE_POSIX_SEMAPHORE;_GLIBCXX_HAVE_POSIX_SEMAPHORE;_GNU_SOURCE;IDF_VER="v5.5.1"")

### Error logs or terminal output

```shell
Command: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/pj/.espressif/python_env/idf5.5_py3.12_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32p4 -DCCACHE_ENABLE=0 /home/pj/esp-webrtc-solution/solutions/doorbell_local
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

CMake Error at /home/pj/esp/esp-idf/tools/cmake/component.cmake:229 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/pj/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Deprecation Warning at
  /home/pj/esp-webrtc-solution/solutions/doorbell_local/main/CMakeLists.txt:3
  (cmake_minimum_required):

    Compatibility with CMake < 3.10 will be removed from a future version of
    CMake.

    Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
    to tell CMake that the project requires at least <min> but has been updated
    to work with policies introduced by <max> or earlier.

  Call Stack (most recent call first):

    /home/pj/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    /home/pj/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:160 (__component_get_requirements)

  CMake Error at
  /home/linuxbrew/.linuxbrew/share/cmake/Modules/ExternalProject.cmake:1307
  (define_property):

    define_property command is not scriptable

  Call Stack (most recent call first):

    /home/pj/esp/esp-idf/tools/cmake/kconfig.cmake:1 (include)
    /home/pj/esp/esp-idf/tools/cmake/idf.cmake:43 (include)
    /home/pj/esp/esp-idf/tools/cmake/project.cmake:29 (include)
    /home/pj/esp-webrtc-solution/solutions/doorbell_local/main/CMakeLists.txt:6 (include)
    /home/pj/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    /home/pj/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:160 (__component_get_requirements)

Call Stack (most recent call first):
  /home/pj/esp/esp-idf/tools/cmake/build.cmake:664 (__component_get_requirements)
  /home/pj/esp/esp-idf/tools/cmake/project.cmake:740 (idf_build_process)
  CMakeLists.txt:14 (project)
```

### Steps to reproduce the behavior

1. run idf.py set-target esp32p4

### Project release version

esp-idf 5.5.1  esp=webrtc-solution main/default branch

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 22.04

### Shell

Bash

mantriyogesh, 2025-09-29:
@robogo-pj, Can you please attach your host example snippet and skdconfig (if generated) to test this independent host example?

vikramdattu, 2025-09-29:
@robogo-pj can you please upgrade the cmake and check if build works fine?

cc @TempoTian